Dilbert66/esphome-dsckeybus #1: Pin code issue

slipx06, 2020-07-25:
Thanks for this work. My pin code is being treated as an octal constant by the compiler as it starts with a zero. This results in an error as it also contains a 9 which is not a valid octal character. How do I specify the pin code so that the compiler recognized leading zero as a base 10 number?

slipx06, 2020-07-26:
OK so it it probably easirt just to create a new code that does not start with a zero. I'm now getting the follownig error when trying to build

```
Compiling /data/alarm/.pioenvs/alarm/src/main.cpp.o
src/main.cpp:136:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "auto DSCkeybus = new DSCkeybushome();\n\nDSCkeybus->onSystemStatusChange([&](std::string\
   ^
In file included from src/main.cpp:37:0:
src/dscAlarm.h: In member function 'void DSCkeybushome::set_alarm_state(int, std::string, std::string)':
src/dscAlarm.h:176:22: error: expected primary-expression before '==' token
   if (code.length()) == 4) { // ensure we get a 4 digit code
                      ^
src/dscAlarm.h:176:26: error: expected ';' before ')' token
   if (code.length()) == 4) { // ensure we get a 4 digit code
                          ^
*** [/data/alarm/.pioenvs/alarm/src/main.cpp.o] Error 1
========================= [FAILED] Took 10.40 seconds =========================
```

Dilbert66, 2020-07-26:
I've corrected this in later versions. Everything is treated as a string (instead of an integer) when sent to the system.  You can use any leading zeros.

Dilbert66, 2020-07-26:
Ok, I see your issue. esphome won't allow me to set a string constant there so I was stuck to using an integer which is not optimal. This code is really not necessary anyhow since technically it should only be used when the dsc panel sends a prompt for an access code.  I have not seen it happen with my system but it's supported by the library I used so I included it.  I send the code from HA using the alarm panel card.  I will see if I can find a better workaround.

The extra parenthesis is corrected. Tks.

Dilbert66, 2020-07-26:
Ok, this is fixed. Using substitutions instead of global.  You can use any codes including all zero's if you like.

closed 2020-07-27 by Dilbert66