Dilbert66/esphome-dsckeybus #2: Erroneous panel status changes

slipx06, 2020-07-27:
I seem to have a problem when using the development version libraries from dscKeybusInterface. I get erroneous and very brief status changes being reported from my panel. I don't experience this using version 1.2  I should probably log this as an issue under that project but hoping you can help with debugging. See example below. During this time the alarm was in a disarmed state the entire time.

[image]

Dilbert66, 2020-07-27:
What panel are you using?

Dilbert66, 2020-07-27:
I've added a new branch called debug.  Use the dscalarm.h file from there and place it in the dsckeybussinterface dir instead of the other. This prints all commands received.  I need to see what is triggering those changes.  Your panel version might behave different than mine.

slipx06, 2020-07-27:
> What panel are you using?

I'm using a PC5005. I'll test with the debug branch and post some logs soon. Thanks.

Dilbert66, 2020-07-28:
One thing I do want to point out is that  the alarm panel does not recognize the "not_ready" status (when a door or window is open and the system can't be armed).  It only recognizes the arm,disarm statuses, triggered and unavailable.  If it gets anything else it shows "unknown".  I suppose I could have the esphome code send "unavailable" as a status  instead of sending "not_ready".   I chose to just simply add a value template in the alarm panel like so:
 dsc_panel:
        name: "Alarm Panel"
        value_template: "{% if is_state('sensor.partition_1_status','not_ready') %} unavailable {%else%} {{states('sensor.partition_1_status')}}{%endif%}"

You can change the status in dsalarm.h to suit your preference.

slipx06, 2020-07-28:
Uploaded some debug logs. (https://pastebin.com/kCQZPYbF) The alarm was in a disarmed state the whole time. You can see the odd behaviour at 08:55:25 when it changes state to armed_away and then back to disarmed

[08:55:22][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[08:55:22][D][Debug02:288]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[08:55:25][D][Debug01:187]: Received: 1,0,01,81,5,1,00,05,00,05,00,81,05,00,C7
[08:55:25][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_away'
[08:55:25][D][Debug02:288]: Received: 0,0,01,81,5,1,00,05,00,05,00,81,05,00,C7
[08:55:25][D][Debug01:187]: Received: 1,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[08:55:25][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[08:55:25][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[08:55:25][D][Debug02:288]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7

Dilbert66, 2020-07-28:
Ok, I see what's happening.  One of the main changes from the master branch to the develop was the capture of armed and triggered status from the 05 command instead of just using the 27 command.  The reason for that was that on initial bootup the esp8266 doesnt know the system status until it gets a 05 or 27 command.  The 27 command has the same info as the 05 (plus zone data) but is normally only sent on a zone change.  The 05 is sent periodically.  So what happens if you only look at the 27 command for status, you will need to wait till something changes to get valid status (ie a door opening, motion sensor triggering, arming, etc).  By using the 05, the partition statuses get updated right away. The only statuses that won't be updated will be zones.

This works fine for most systems but in the case of your pc5005, it seems to send bogus arm and trigger commands periodically followed by the correct status.  Mine doesnt do that. I suspect it's a bug in the panel code.  Either way, most keypads probably ignore the 05 and depend on the 27 for status. 05 is probably used by older keypads.

So to fix this. I can either remove the 05 arm/triggered check code or add logic to check for another variable.
When my system is armed the lights set bit 1 so I can check for that to ensure that this is a valid status. I'll upload a new dsckeybusprocessdata.cpp file in the debug branch and you can try that.

08:55:22][D][Debug02:288]: Received: 0,0,00,81,1,0,00,05,00,[05],00,[81],[01],00,C7
cmd=05
lights=81 - ready lights only 
status=01 - disarmed and ready

[08:55:25][D][Debug01:187]: Received: 1,0,01,81,5,1,00,05,00,[05],00,[81],[05],00,C7
cmd=05
lights=81  ready light only
status=05 - armed in away mode

[08:55:25][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_away'
[08:55:25][D][Debug02:288]: Received: 0,0,01,81,5,1,00,05,00,05,00,81,05,00,C7
[08:55:25][D][Debug01:187]: Received: 1,0,00,81,1,0,00,05,00,[05],00,[81],[01],00,C7
cmd=05
lights=81
status=01 - disarmed and ready

triggered example:

[08:55:25][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[08:55:25][D][Debug02:288]: Received: 0,0,00,81,1,0,00,05,00,[05],00,[81],[01],00,C7
cmd=05
lights=81
status=01 - disarmed and ready

13:40:13][D][Debug01:187]: Received: 0,0,00,81,17,0,00,05,00,[05],00,[81],[11],00,C7
cmd=05
lights=81
status=11 - partition in alarm (triggered)

[13:40:13][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'not_ready'
[13:40:13][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'triggered'
[13:40:13][D][Debug02:288]: Received: 0,0,00,81,17,0,00,05,00,05,00,81,11,00,C7
[13:40:14][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
cmd=05
lights=81
status=01 - back to ready
[13:40:14][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'

Dilbert66, 2020-07-28:
If you could also post some sample debug messages with your system in actual armed mode, that would also be helpful to ensure the logic will work correctly on your system.

slipx06, 2020-07-28:
OK thanks. Let me try the new version. Here are some more logs for the current version showing the alarm being armed_home and then disarmed

[16:25:29][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[16:25:29][D][Debug02:288]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[16:25:29][D][Debug01:187]: Received: 0,0,00,81,1,0,00,0A,00,0A,00,80,01,00,00
[16:25:29][D][Debug01:187]: Received: 0,0,00,81,1,0,00,64,00,64,00,06,6A,00,00
[16:25:29][D][Debug01:187]: Received: 0,0,00,81,1,0,00,5D,00,5D,00,00,00,00,00
[16:25:29][D][Debug01:187]: Received: 0,0,00,81,1,0,00,E6,00,E6,00,18,01,00,00
[16:25:29][D][Debug01:187]: Received: 0,0,00,81,1,0,00,87,00,87,00,00,02,89,00
[16:25:29][D][Debug01:187]: Received: 0,0,00,81,1,0,00,BB,00,BB,00,00,00,BB,00
[16:25:30][D][Debug01:187]: Received: 0,0,00,81,1,0,00,E6,00,E6,00,1A,15,00,00
[16:25:30][D][Debug01:187]: Received: 0,0,00,81,1,0,00,75,00,75,00,00,75,00,00
[16:25:30][D][Debug01:187]: Received: 0,0,00,81,1,0,00,CE,00,CE,00,01,C0,00,00
[16:25:30][D][Debug01:187]: Received: 0,0,00,81,1,0,00,CE,00,CE,00,40,FF,FF,FF
[16:25:30][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,05,00,05,00,8B,08,00,C7
[16:25:30][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'pending'
[16:25:30][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,05,00,05,00,8B,08,00,C7
[16:25:30][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,75,00,75,00,00,75,00,00
[16:25:36][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:36][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:36][D][binary_sensor:036]: 'Passage': Sending state ON
[16:25:37][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:25:37][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:25:37][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:25:38][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:38][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:38][D][binary_sensor:036]: 'Passage': Sending state ON
[16:25:39][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:25:39][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:25:39][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:25:41][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,11,00,11,00,AA,AA,AA,AA
[16:25:48][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:48][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:48][D][binary_sensor:036]: 'Passage': Sending state ON
[16:25:50][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:25:50][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:25:50][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:25:59][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,E6,00,E6,00,2C,04,00,00
[16:25:59][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:59][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:25:59][D][binary_sensor:036]: 'Passage': Sending state ON
[16:26:01][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:01][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:01][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:26:08][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:08][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:08][D][binary_sensor:036]: 'Passage': Sending state ON
[16:26:09][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:09][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:09][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:26:17][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,11,00,11,00,AA,AA,AA,AA
[16:26:19][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:19][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:19][D][binary_sensor:036]: 'Passage': Sending state ON
[16:26:20][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:20][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:20][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:26:35][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,B1,00,B1,00,DF,0F,00,00
[16:26:37][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:37][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:37][D][binary_sensor:036]: 'Passage': Sending state ON
[16:26:38][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:38][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:38][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:26:45][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:45][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,10,27,00,8B,08,00,C7
[16:26:45][D][binary_sensor:036]: 'Passage': Sending state ON
[16:26:46][D][Debug01:187]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:46][D][Debug02:288]: Received: 0,1,00,8B,8,0,00,27,00,27,00,8B,08,00,C7
[16:26:46][D][binary_sensor:036]: 'Passage': Sending state OFF
[... 71 lines cut ...]
[16:28:19][D][Debug02:288]: Received: 0,0,01,8A,4,0,01,27,00,27,00,8A,04,00,C7
[16:28:19][D][binary_sensor:036]: 'Passage': Sending state OFF
[16:28:23][D][Debug01:187]: Received: 0,0,01,8A,4,0,01,E6,00,E6,00,0F,00,F5,00
[... cut ...]
[16:28:46][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[16:28:46][D][Debug02:288]: Received: 0,0,00,81,62,0,00,05,00,05,00,81,3E,00,C7
[16:28:46][D][Debug01:187]: Received: 0,0,00,81,62,0,00,EB,00,EB,00,01,20,15,51
[16:28:47][D][Debug01:187]: Received: 0,0,00,81,62,0,00,C3,00,C3,00,08,FF,CA,00
[16:28:51][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[16:28:51][D][Debug02:288]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[16:28:52][D][Debug01:187]: Received: 0,0,00,81,1,0,00,C3,00,C3,00,08,FF,CA,00

Dilbert66, 2020-07-28:
Perfect. I see that bit 1 is getting set in the lights status on armed:
05,00,[8A],[04],00
lights status = 8A (bit 1 set in lower byte)
arm status = 04 (armed home)

slipx06, 2020-07-28:
Thank you. Running the new version and will do some testing. Still getting these fire and trouble status changes. Any idea what could be causing that?

[17:17:49][D][Debug01:187]: Received: 0,0,00,C1,1,0,00,05,00,05,00,C1,01,00,C7
[17:17:49][D][binary_sensor:036]: 'Fire Status': Sending state ON

[17:17:49][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[17:17:49][D][binary_sensor:036]: 'Fire Status': Sending state OFF

[17:07:20][D][Debug01:187]: Received: 0,0,00,91,1,0,00,05,00,05,00,91,01,00,C7
[17:07:20][D][binary_sensor:036]: 'Trouble Status': Sending state ON

[17:07:21][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[17:07:21][D][binary_sensor:036]: 'Trouble Status': Sending state OFF

Dilbert66, 2020-07-28:
Same issue with the bogus status messages  except with the other flags. I have to see what I can do with this since the trouble and fire bits can be set no matter what the state of the alarm so I can't use the armed bit for these.  Might have to provide a customizable flag that you can set in the yaml to disable the trouble/fire check during the 05 cmd so that it can be set if the issue occurs on a panel.

[17:17:49][D][Debug01:187]: Received: 0,0,00,C1,1,0,00,05,00,05,00,[C1],01,00,C7
cmd: 05
status: c1 (fire bit on)

[17:17:49][D][binary_sensor:036]: 'Fire Status': Sending state ON

[17:17:49][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[17:17:49][D][binary_sensor:036]: 'Fire Status': Sending state OFF

[17:07:20][D][Debug01:187]: Received: 0,0,00,91,1,0,00,05,00,05,00,[91],01,00,C7
cmd: 05
status: 91 (trouble bit on)
[17:07:20][D][binary_sensor:036]: 'Trouble Status': Sending state ON

[17:07:21][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[17:07:21][D][binary_sensor:036]: 'Trouble Status': Sending state OFF

Dilbert66, 2020-07-28:
If you can simulate a real trouble status, it would help to compare.  Perhaps a power failure on the panel, etc.  This way I can compare a valid sequence vs a bogus one.

Dilbert66, 2020-07-28:
On my panel , panelbyte 4 also hold the fire/trouble bits similar to the panelbyte 2 when valid.  With your dump, panelbyte 4 was zero.  Will be interesting what yours show for a valid trouble condition.

slipx06, 2020-07-28:
This is what a real trouble condition looks like. (Simulated power failure)

[18:08:31][D][Debug01:187]: Received: 0,0,00,91,1,0,00,05,00,05,00,91,01,10,C7
[18:08:31][D][binary_sensor:036]: 'Trouble Status': Sending state ON

[18:10:01][D][Debug01:187]: Received: 0,0,00,81,1,0,00,05,00,05,00,81,01,00,C7
[18:10:01][D][binary_sensor:036]: 'Trouble Status': Sending state OFF

Dilbert66, 2020-07-28:
Ok, that's perfect. panel byte 4 gets set correctly. I've pushed a new version of dsckeybusprocess.cpp in debug that should address this.

slipx06, 2020-07-28:
Thank you. The bogus trouble condition is gone. How do I test Fire Status. Trigger the fire alarm?

Dilbert66, 2020-07-28:
There should be a key on your keypad you can press (or have to press two keys at the same time for a few seconds). It should be in your user manual.  If your system is monitored centrally, you need to call them first to let them know you are doing a test!!

You can reset the alarm with *72 (on my system anyhow).

slipx06, 2020-07-28:
I don't have any fire zones. So might just ignore those messages.

Dilbert66, 2020-07-28:
I'll test on my end some time. Ok changes pushed to master. Tks for your help.

closed 2020-07-28 by Dilbert66

slipx06, 2020-07-29:
It looks like this is still not completely solved. The alarm was armed in armed_home last night and I noticed bogus armed_night and no_entry_delay messages being sent. Some samples below

[02:29:42][D][Debug01:187]: Received: 1,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[02:29:42][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'
[02:29:42][D][Debug02:289]: Received: 0,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[02:29:42][D][Debug01:187]: Received: 1,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[02:29:42][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'

[03:40:32][D][Debug01:187]: Received: 1,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[03:40:32][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'
[03:40:32][D][Debug02:289]: Received: 0,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[03:40:32][D][Debug01:187]: Received: 1,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[03:40:32][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'
[03:40:32][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_home'

[03:59:27][D][Debug01:187]: Received: 1,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[03:59:27][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'
[03:59:27][D][Debug02:289]: Received: 0,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[03:59:27][D][Debug01:187]: Received: 1,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[03:59:27][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'

[04:41:14][D][Debug01:187]: Received: 1,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[04:41:14][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'
[04:41:14][D][Debug02:289]: Received: 0,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[04:41:14][D][Debug01:187]: Received: 1,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[04:41:14][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'
[04:41:14][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_home'

[04:56:09][D][Debug01:187]: Received: 0,0,01,8A,6,0,01,05,00,05,00,8A,06,00,C7
[04:56:09][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_entry_delay'
[04:56:09][D][Debug02:289]: Received: 0,0,01,8A,6,0,01,05,00,05,00,8A,06,00,C7
[04:56:09][D][Debug01:187]: Received: 0,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[04:56:09][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'

[05:05:44][D][Debug01:187]: Received: 0,0,01,8A,6,0,01,05,00,05,00,8A,06,00,C7
[05:05:44][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_entry_delay'
[05:05:44][D][Debug02:289]: Received: 0,0,01,8A,6,0,01,05,00,05,00,8A,06,00,C7
[05:05:44][D][Debug01:187]: Received: 0,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[05:05:44][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'

[05:06:54][D][Debug01:187]: Received: 0,0,01,8A,6,0,01,05,00,05,00,8A,06,00,C7
[05:06:54][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_entry_delay'
[05:06:54][D][Debug02:289]: Received: 0,0,01,8A,6,0,01,05,00,05,00,8A,06,00,C7
[05:06:54][D][Debug01:187]: Received: 0,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[05:06:54][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'

[05:12:20][D][Debug01:187]: Received: 1,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[05:12:20][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'
[05:12:20][D][Debug02:289]: Received: 0,0,01,8A,5,1,00,05,00,05,00,8A,05,00,C7
[05:12:20][D][Debug01:187]: Received: 1,0,01,8A,4,0,01,05,00,05,00,8A,04,00,C7
[05:12:20][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'armed_home'
[05:12:20][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_home'

Dilbert66, 2020-07-29:
That is a chatty panel you have. Never seen this on mine.   ok, well I've added a new option to the yaml called "enable05ArmStatus".
Set it to false to prevent use of 05 for those statuses.   The only issue is that when on the initial start up of your esp it will not know what state your alarm is at that time (weither open, disarmed,armed, etc) until something happens like disarming, opening a door, motion sensor activating, etc.    
You'll need to update your yaml to get the new flags and keybus init change.  Most files were changed so you will need to update most.

slipx06, 2020-07-29:
Getting some compile errors for the new version

```
src/main.cpp:156:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "auto DSCkeybus = new DSCkeybushome();\n\nDSCkeybus->onSystemStatusChange([&](std::string\
   ^
src/main.cpp: In lambda function:
src/main.cpp:363:40: error: 'getKeybusConnectionStatus' was not declared in this scope
       return getKeybusConnectionStatus();
                                        ^
src/main.cpp: In lambda function:
src/main.cpp:372:42: error: no matching function for call to 'DSCkeybushome::DSCkeybushome()'
       auto DSCkeybus = new DSCkeybushome();
                                          ^
src/main.cpp:372:42: note: candidates are:
In file included from src/main.cpp:35:0:
src/dscAlarm.h:20:4: note: DSCkeybushome::DSCkeybushome(const char*, bool)
    DSCkeybushome( const char *accessCode, bool enable05ArmStatus)
    ^
src/dscAlarm.h:20:4: note:   candidate expects 2 arguments, 0 provided
src/dscAlarm.h:18:7: note: DSCkeybushome::DSCkeybushome(const DSCkeybushome&)
 class DSCkeybushome : public Component, public CustomAPIDevice {
       ^
src/dscAlarm.h:18:7: note:   candidate expects 1 argument, 0 provided
src/dscAlarm.h:18:7: note: DSCkeybushome::DSCkeybushome(DSCkeybushome&&)
src/dscAlarm.h:18:7: note:   candidate expects 1 argument, 0 provided
src/main.cpp:412:24: error: could not convert '{DSCkeybus}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
       return {DSCkeybus};
                        ^
src/main.cpp: In lambda function:
src/main.cpp:364:3: warning: control reaches end of non-void function [-Wreturn-type]
   });
   ^
src/main.cpp: In lambda function:
src/main.cpp:413:3: warning: control reaches end of non-void function [-Wreturn-type]
   });
   ^
*** [/data/alarm/.pioenvs/alarm/src/main.cpp.o] Error 1
```

Dilbert66, 2020-07-29:
You need to do a "Clean Build" from the ... menu on the esphome config. It has a cached yaml file.  Then you can recompile/upload.

Dilbert66, 2020-07-29:
[image]

slipx06, 2020-07-29:
Ok that worked. Let me test

slipx06, 2020-07-29:
The new version looks to be working fine. Also seeing the new Partition messages. Will monitor tonight. Thanks for helping so quickly

taligentx, 2020-10-02:
Hi @slipx06 - I've opened up an issue to take a look at the Keybus data from the PC5005 to see if this an isolated instance or if it's common to all PC5005 panels.  If you get a chance to capture some data, feel free to add it to https://github.com/taligentx/dscKeybusInterface/issues/158, thanks!

Dilbert66, 2020-10-02:
Hi Nikhil, please see my comment in thread 158. The issue was resolved with a debounce update.