Dilbert66/esphome-dsckeybus #3: Can't disarm panic alarm for partition 2

slipx06, 2020-07-29:
If I trigger the panic alarm in partition 1, either by calling `esphome.alarm_alarm_trigger_panic` or `esphome.alarm_set_alarm_state` and manually specifying the partition, state  I can successfully disarm by using `esphome.alarm_alarm_keypress` and sending /1D(code). See logs below

Alarm triggered

[17:25:10][D][Debug22:236]: Partitions changed data:  00,00,00,00,81,11,00,00,05,00,81,11,00,C7,00
[17:25:10][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '11: Alarm'

Keypress sent

[17:25:19][D][Debug:118]: Writing keys: /1D(code)
[17:25:20][D][Debug22:236]: Partitions changed data:  07,00,00,00,81,11,00,00,27,00,81,11,00,C7,00

Alarm disarmed
[17:25:21][D][Debug22:236]: Partitions changed data:  00,00,00,00,81,01,00,00,05,00,81,01,00,C7,00
[17:25:21][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'

However if I trigger the same panic alarm for partition 2 using esphome.alarm_set_alarm_state, "Partition": "2",  "State": "P",    "code": "", the only way to turn off the alarm is at the keypad. If I send /2D(code) using  `esphome.alarm_alarm_keypress` nothing happens. 

17:28:53][D][Debug22:236]: Partitions changed data:  00,00,00,00,80,11,00,00,05,00,80,11,00,C7,00
[17:28:53][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '11: Alarm'
[17:28:56][D][Debug:118]: Writing keys: /2D(code)
[17:29:01][D][Debug22:236]: Partitions changed data:  00,00,00,00,81,11,00,00,27,00,81,11,00,C7,04

The only partition 2 messages I can find in my logs are

17:50:47][D][Debug22:236]: Partitions changed data:  00,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  01,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'E7: Input: 3 digits'
[17:50:47][D][Debug22:236]: Partitions changed data:  02,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  03,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  04,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  05,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  06,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  07,00,00,00,81,01,00,00,05,00,81,01,00,E7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  00,00,00,00,81,01,00,00,05,00,81,01,00,C7,00
[17:50:47][D][Debug22:236]: Partitions changed data:  01,00,00,00,81,01,00,00,05,00,81,01,00,C7,00
[17:50:47][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'C7: Unknown'

Please can you test if this is unique to my panel or a problem

Dilbert66, 2020-07-29:
Try sending /2xxxx where xxxx is your access code

Dilbert66, 2020-07-29:
The D command is only used for send alarm state

slipx06, 2020-07-29:
Sending /2xxxx does not work nor does the setting state D. Sent a few times just to make sure

[18:35:45][D][Debug:118]: Writing keys: /2xxxx
[18:35:45][D][Debug:118]: Writing keys: /2xxxx
[18:35:45][D][Debug:118]: Writing keys: /2xxxx
[18:35:46][D][Debug:118]: Writing keys: /2xxxx
[18:35:47][D][Debug22:236]: Partitions changed data:  00,00,00,00,85,11,00,00,05,00,85,11,00,C7,00
[18:35:47][D][Debug22:236]: Partitions changed data:  01,00,00,00,85,11,00,00,05,00,85,11,00,C7,00
[18:35:47][D][Debug22:236]: Partitions changed data:  02,00,00,00,85,11,00,00,05,00,85,11,00,C7,00
[18:35:47][D][Debug22:236]: Partitions changed data:  03,00,00,00,85,11,00,00,05,00,85,11,00,C7,00

Dilbert66, 2020-07-29:
That's a tricky one. Can't see what it is at the moment. Will do some tests to see what I can find.

Dilbert66, 2020-07-29:
I can't duplicate on my end since I don't use partition 2 as an active one.  I just have some sensors on it and use it only for that.
I did find an issue when trying to send a command to an inactive partition, the esp would lockup because it would block.  I've added a wait timer to prevent that.   I've also change some debug info which might catch what is going on in your end.  I suspect the library is not setting the correct bits for partitions other than 1. I need to debug this further.  Try out the debug version and post some logs when you get a chance.   
It's great info that you are providing and helps in making this project better.

slipx06, 2020-07-29:
Thanks will give it a try. Is there any coding logic that prevents me from triggering a panic alarm for an already armed partition. What I'm trying to do is use one of my Zigbee sensors to trigger the alarm based on certain conditions. If the alarm is armed sending the panic command to the armed partition does nothing, but if send to an unused partition I can trigger it. The problem is I cant disarm it. I have not tried pushing my panic button while the alarm is armed, but I understood that the panic alarm takes priority and should sound. Will test tomorrow.

Dilbert66, 2020-07-29:
There is logic with the set alarm_state to only allow during not armed and not exit delay.  This was how the original library had it but really there should be no check for it.  There is no logic though for the alarm_keypress where you could just send a "p".
I will also remove the check for an alarmed panel and you can test with set_alarm_status as well.

Dilbert66, 2020-07-29:
Can you try also sending a /2*2 command.  This is the trouble status command.  You can also try /1*2 to compare. Use the keypress service.

slipx06, 2020-07-30:
Ok so I now I can  successfully  trigger the panic alarm using the `esphome.alarm_alarm_trigger_panic` service for an already armed partition. I think your last change may have introduced some timing issues as I now have to issue commands a few times for it to work.  This inlcudes all the esphome.alarm_alarm_* services

Here is the output of sending "/2*2" or "/1*1"

[09:08:18][D][Debug:119]: Writing keys: /1*2
[09:08:18][D][Debug22:235]: Partition1 changed data:  00,00,00,80,9E,00,00,05,00,80,9E,00,C7,00
[09:08:18][D][Debug33:241]: Partition data 00: 9E,80,00,00
[09:08:18][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '9E: Enter * code'
[09:08:18][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
[09:08:18][D][Debug33:241]: Partition data 01: C7,00,00,00
[09:08:18][D][Debug33:241]: Partition data 02: C7,00,00,00
[09:08:18][D][Debug33:241]: Partition data 03: C7,00,00,00
[09:08:18][D][Debug33:241]: Partition data 04: 00,00,00,00
[09:08:18][D][Debug33:241]: Partition data 05: 00,00,00,00
[09:08:18][D][Debug33:241]: Partition data 06: 00,00,00,00
[09:08:18][D][Debug33:241]: Partition data 07: 00,00,00,00
[09:08:19][D][Debug22:235]: Partition1 changed data:  00,00,00,80,A1,00,00,05,00,80,A1,00,C7,00
[09:08:19][D][Debug33:241]: Partition data 00: A1,80,00,00
[09:08:19][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'A1: Trouble menu'
[09:08:19][D][Debug33:241]: Partition data 01: C7,00,00,00
[09:08:19][D][Debug33:241]: Partition data 02: C7,00,00,00
[09:08:19][D][Debug33:241]: Partition data 03: C7,00,00,00
[09:08:19][D][Debug33:241]: Partition data 04: 00,00,00,00
[09:08:19][D][Debug33:241]: Partition data 05: 00,00,00,00
[09:08:19][D][Debug33:241]: Partition data 06: 00,00,00,00
[09:08:19][D][Debug33:241]: Partition data 07: 00,00,00,00

[09:07:00][D][Debug:119]: Writing keys: /2*2
[09:07:02][D][Debug22:235]: Partition1 changed data:  00,00,00,81,01,00,00,A5,00,20,15,83,30,00
[09:07:02][D][Debug33:241]: Partition data 00: 01,81,00,00
[09:07:02][D][Debug33:241]: Partition data 01: C7,00,00,00
[09:07:02][D][Debug33:241]: Partition data 02: C7,00,00,00
[09:07:02][D][Debug33:241]: Partition data 03: C7,00,00,00
[09:07:02][D][Debug33:241]: Partition data 04: 00,00,00,00
[09:07:02][D][Debug33:241]: Partition data 05: 00,00,00,00
[09:07:02][D][Debug33:241]: Partition data 06: 00,00,00,00
[09:07:02][D][Debug33:241]: Partition data 07: 00,00,00,00

Here is an example of the same command being sent and the panel not responding 

[09:10:21][D][Debug:119]: Writing keys: /1*2
[09:10:51][D][Debug22:235]: Partition1 changed data:  00,00,00,81,01,00,00,05,00,81,01,00,C7,00
[09:10:51][D][Debug33:241]: Partition data 00: 01,81,00,00
[09:10:51][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'
[09:10:51][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[09:10:51][D][Debug33:241]: Partition data 01: C7,00,00,00
[09:10:51][D][Debug33:241]: Partition data 02: C7,00,00,00
[09:10:51][D][Debug33:241]: Partition data 03: C7,00,00,00
[09:10:51][D][Debug33:241]: Partition data 04: 00,00,00,00
[09:10:51][D][Debug33:241]: Partition data 05: 00,00,00,00
[09:10:51][D][Debug33:241]: Partition data 06: 00,00,00,00
[09:10:51][D][Debug33:241]: Partition data 07: 00,00,00,00

Dilbert66, 2020-07-30:
Something is not right with your partition 2.  It shows c7 which means it's unavailable or not active which is why the command has no affect on it.  Also the panic alarm you show, is on partition1 not 2 .  (1st byte of the data is the status.  11 is partition alarm, c7 is unavailable.
That E7  (input 3 digits). That I have no idea what that's about. You might want to check the setup for partition 2 so see why it differs from partition1.

Dilbert66, 2020-07-30:
As to the delay with the * commands. Yes, I will fix that.

slipx06, 2020-07-30:
I don't think my partition 2 is setup correctly as you say. I'm going to abandon that option now that I can trigger the Panic in partition 1 whilst armed.

Dilbert66, 2020-07-30:
Ok, I've pushed a new version to master that should address the issue with having to send commands twice.  Also has the panic while armed changes.

Dilbert66, 2020-07-30:
That's strange though that it did not pick up your system being armed if it was already running since a 27 message would have been sent.  Normally only need the 05 statuses if you boot your esp module while your system is armed so it will know about it.

Dilbert66, 2020-07-30:
You can enable the debug messages again on your system by setting this flag in the custom component.

 DSCkeybus->debug=true;

slipx06, 2020-07-30:
Thanks for all your help. Running the new master. There must of been some other changes to the development libraries that cause some issues with my panel. I was running a version 1.2 sketch for many months without issues.

Dilbert66, 2020-07-30:
whups. I had forgotten to push dscalarm.h.  Redownload that one

slipx06, 2020-07-30:
First message on boot. At least debug is working

[22:47:42][D][api.connection:583]: Client 'Home Assistant 0.113.2 (192.168.1.103)' connected successfully!
[22:47:45][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,84,00,01,05,00,8A,84,00,C7,00
[22:47:45][D][Debug33:243]: Partition data 00: 84,8A,01,00
[22:47:45][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '84: Unknown'
[22:47:45][D][Debug33:243]: Partition data 01: C7,00,00,00
[22:47:45][D][Debug33:243]: Partition data 02: C7,00,00,00
[22:47:45][D][Debug33:243]: Partition data 03: C7,00,00,00
[22:47:45][D][Debug33:243]: Partition data 04: 00,00,00,00
[22:47:45][D][Debug33:243]: Partition data 05: 00,00,00,00
[22:47:45][D][Debug33:243]: Partition data 06: 00,00,00,00
[22:47:45][D][Debug33:243]: Partition data 07: 00,00,00,00

Dilbert66, 2020-07-30:
84 is not a known command.  What state is your panel in right now? I can add it if we can guess what it is.

slipx06, 2020-07-30:
Booted up in armed_home.  I have no idea what is causing all these changes.  Strange data below since last boot

[22:47:45][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,04,00,01,05,00,8A,04,00,C7,00
[22:47:45][D][Debug33:243]: Partition data 00: 04,8A,01,00
[22:47:45][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '04: Armed stay'

[22:48:29][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,84,00,01,05,00,8A,84,00,C7,00
[22:48:29][D][Debug33:243]: Partition data 00: 84,8A,01,00
[22:48:29][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '84: Unknown'

[22:48:29][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,04,00,01,05,00,8A,04,00,C7,00
[22:48:29][D][Debug33:243]: Partition data 00: 04,8A,01,00
[22:48:29][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '04: Armed stay'

[22:57:53][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,24,00,01,05,00,8A,24,00,C7,00
[22:57:53][D][Debug33:243]: Partition data 00: 24,8A,01,00
[22:57:53][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '24: Unknown'

[22:57:53][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,04,00,01,05,00,8A,04,00,C7,00
[22:57:53][D][Debug33:243]: Partition data 00: 04,8A,01,00
[22:57:53][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '04: Armed stay'

[23:06:16][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,44,00,01,05,00,8A,44,00,C7,00
[23:06:16][D][Debug33:243]: Partition data 00: 44,8A,01,00
[23:06:16][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '44: Unknown'

[23:10:30][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,0C,00,01,05,00,8A,0C,00,C7,00
[23:10:30][D][Debug33:243]: Partition data 00: 0C,8A,01,00
[23:10:30][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '0C: Entry delay'

[23:11:14][D][Debug22:237]: Partition1 changed data:  00,00,01,8A,06,00,01,05,00,8A,06,00,C7,00
[23:11:14][D][Debug33:243]: Partition data 00: 06,8A,01,00
[23:11:14][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '06: Unknown'

Dilbert66, 2020-07-30:
Well, that's an effect of the message code I added to capture the 05 messages.  Since your system likes to send out various 05 messages out for no reason, it's showing those.  You would not have seen those before even though they are always sent.  It won't affect the operation as those are display only since the true status is the partition status and not the partition msg state. 
I will add another flag that you can also prevent those from showing as well as they don't mean much in your case.

Dilbert66, 2020-07-30:
Ok, made the change on master.  Changed files are dscalarm.yaml and dscalarm.h. New flag added which gets passed to the class.

slipx06, 2020-07-30:
That will help. I do have an example below where the state is changed to armed_night, but it should not have

[23:15:10][D][Debug22:237]: Partition1 changed data:  01,00,01,8A,05,01,00,05,00,8A,05,00,C7,00
[23:15:10][D][Debug33:243]: Partition data 00: 05,8A,01,00
[23:15:10][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '05: Armed away'
[23:15:10][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'

Dilbert66, 2020-07-30:
That usually means it received a 0x16 or 0x06 message in the 27 command to indicate that it has been set to noentrydelay.  The 05 command also gets those but if you have the enable05armstatus = false, it won't affect anthing.  It's difficult to know unless we get more debug info as to what your panel is really doing.   If your system is also sending bogus 27 commands , then that would create a whole new level of complexity.

Dilbert66, 2020-07-30:
I've pushed a new update to master that adds a few more fields in the debug lines . Removed a couple things.  Let's see if that captures those night arm messsages and we can see what's going on.

slipx06, 2020-07-31:
Ok so things are looking better. I still get some intermittent "Unavailable" states being set. As seen below there were no sensors triggered to cause the panel not to be ready.  Can we filter those out?

[10:20:09][D][Debug22:241]: Partition 1 panel data: 05,00,81,81,00,C7,00
[10:20:09][D][Debug33:247]: Partition data 00: 81,81,00,00,00,00,00,00,00
[10:20:09][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_messages'
**[10:20:09][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'**
[10:20:09][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[10:20:09][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'no_messages'
[10:20:09][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[10:20:09][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[10:20:09][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[10:20:09][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[10:20:09][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[10:20:09][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[10:20:09][D][Debug22:241]: Partition 1 panel data: 05,00,81,01,00,C7,00
[10:20:09][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00
[10:20:09][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_messages'
[10:20:09][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[10:20:09][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[10:20:09][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'no_messages'
[10:20:09][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[10:20:10][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[10:20:10][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[10:20:10][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[10:20:10][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[10:20:10][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug22:241]: Partition 1 panel data: 05,00,81,81,00,C7,00
[10:21:22][D][Debug33:247]: Partition data 00: 81,81,00,00,00,00,00,00,00
[10:21:22][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_messages'
**[10:21:22][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'**
[10:21:22][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[10:21:22][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'no_messages'
[10:21:22][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug22:241]: Partition 1 panel data: 05,00,81,01,00,C7,00
[10:21:22][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00
[10:21:22][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_messages'
[10:21:22][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[10:21:22][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[10:21:22][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'no_messages'
[10:21:22][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[10:21:22][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00

slipx06, 2020-07-31:
Here are debug logs for the armed_night issue. The panel was armed_home 

[11:32:05][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_messages'
[11:32:05][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'
[11:32:05][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[11:32:05][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'no_messages'
[11:32:05][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[11:32:05][D][Debug22:241]: Partition 1 panel data: 05,00,8A,04,00,C7,00
[11:32:05][D][Debug33:247]: Partition data 00: 04,8A,01,00,01,01,00,01,00
[11:32:05][D][text_sensor:015]: 'Partition 1 Msg ': Sending state 'no_messages'
[11:32:05][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_night'
[11:32:05][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[11:32:05][D][text_sensor:015]: 'Partition 2 Msg ': Sending state 'no_messages'
[11:32:05][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[11:32:05][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[11:32:12][D][Debug22:241]: Partition 1 panel data: 05,00,8A,06,00,C7,00

Dilbert66, 2020-07-31:
The no_messages issue was caused because with the removal of the messages with the flag, the initial init was still being done. Ive fixed that

Dilbert66, 2020-07-31:
Ok, changes pushed to master. Let's see if that corrects all issues.

slipx06, 2020-07-31:
New version is not updating the partition status. Showing as unavailable on restart. I also dont have anything in the logs that shows the AC power message being sent

[image]

[13:40:11][D][api.connection:583]: Client 'Home Assistant 0.113.2 (192.168.1.103)' connected successfully!
[13:40:11][D][Debug22:241]: Partition 1 panel data: 05,00,80,03,00,C7,00
[13:40:11][D][Debug33:247]: Partition data 00: 03,80,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 02: D7,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[13:40:11][D][Debug22:241]: Partition 1 panel data: 05,00,80,03,00,C7,00
[13:40:11][D][Debug33:247]: Partition data 00: 03,80,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[13:40:11][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[13:40:12][D][Debug22:241]: Partition 1 panel data: 27,00,80,03,00,C7,04
[13:40:12][D][Debug33:247]: Partition data 00: 03,80,00,00,00,00,00,00,00
[13:40:12][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[13:40:12][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[13:40:12][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[13:40:12][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[13:40:12][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[13:40:12][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[13:40:12][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[13:40:12][D][binary_sensor:036]: 'Dining Room': Sending state ON
[13:40:15][D][Debug22:241]: Partition 1 panel data: 05,00,80,03,00,C7,00
[13:40:15][D][Debug33:247]: Partition data 00: 03,80,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 02: E7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug22:241]: Partition 1 panel data: 05,00,80,03,00,C7,00
[13:40:15][D][Debug33:247]: Partition data 00: 03,80,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug22:241]: Partition 1 panel data: 27,00,80,03,00,C7,14
[13:40:15][D][Debug33:247]: Partition data 00: 03,80,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 01: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 02: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 03: C7,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 04: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 05: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 06: 00,00,00,00,00,00,00,00,00
[13:40:15][D][Debug33:247]: Partition data 07: 00,00,00,00,00,00,00,00,00

Dilbert66, 2020-07-31:
I just realized I had left some test code for ac power logging. I'll remove that as it puts out a bogus message.

slipx06, 2020-07-31:
The new version wont compile

```
Compiling /data/alarm/.pioenvs/alarm/src/main.cpp.o
In file included from src/main.cpp:35:0:
src/dscAlarm.h: In member function 'virtual void DSCkeybushome::loop()':
src/dscAlarm.h:238:33: warning: the address of 'enable05Messages' will always evaluate as 'true' [-Waddress]
   if (dsc.keypadFireAlarm && && enable05Messages) partitionMsgChangeCallback(1,"Keypad Fire Alarm");
                                 ^
src/dscAlarm.h:238:33: error: label 'enable05Messages' used but not defined
Generating LD script /data/alarm/.pioenvs/alarm/ld/local.eagle.app.v6.common.ld
Compiling /data/alarm/.pioenvs/alarm/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
*** [/data/alarm/.pioenvs/alarm/src/main.cpp.o] Error 1
```

Dilbert66, 2020-07-31:
typo on my end. Good now

slipx06, 2020-07-31:
Looking good. Was testing with the alarm armed_home. Below is an example of an erroneous state change. In theory the panel should not be able to directly change state from armed_home to armed_away. Could this be noise on the keybus?

[14:38:48][D][Debug33:247]: Partition data 00: 04,8A,01,00,01,00,00,00,00
[14:40:43][D][Debug22:240]: Partition 1 panel data: 05,00,8A,05,00,C7,00
[14:40:43][D][Debug33:247]: Partition data 00: 05,8A,01,01,00,00,00,01,00
**[14:40:43][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_away'**
[14:40:43][D][Debug22:240]: Partition 1 panel data: 05,00,8A,04,00,C7,00
[14:40:43][D][Debug33:247]: Partition data 00: 04,8A,01,00,01,00,00,01,00
**[14:40:43][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_home'**
[14:40:48][D][Debug22:240]: Partition 1 panel data: 05,00,8A,04,00,C7,00
[14:40:48][D][Debug33:247]: Partition data 00: 04,8A,01,00,01,00,00,00,00
[14:40:48][D][Debug33:247]: Partition data 03: E7,00,00,00,00,00,00,00,00

Dilbert66, 2020-07-31:
Yes, running with enable05Armstatus will use the 05 command for arm statuses and in your case will cause all those switching states since your panel sends all the possible arm states on the 05 command instead of the actual one.  In my case, it sends arm_away only (even when I set to arm home), so every panel has some odd behaviors looks like.  Setting enable05armstatus to false stops responding to 05 status arm status changes and only uses the 27 command which is dependable but only sent when something changes (instead of the 05 which is sent periodically)

Dilbert66, 2020-07-31:
So the best approach in your case is to set both flags to false to prevent all those invalid messages and states. For my panel, I can use true.

slipx06, 2020-07-31:
Agreed. The only thing now is filtering out all these fake "unavailable" messages. Is there anything in the logs I posted to help with that

Dilbert66, 2020-07-31:
You should not get those unavalable messages with the flags set to false???

slipx06, 2020-07-31:
I have both flags set to false these are my logs

[16:01:05][D][Debug22:240]: Partition 1 panel data: 05,00,81,01,00,C7,00
[16:01:05][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00
[16:01:06][D][Debug22:240]: Partition 1 panel data: 05,00,81,81,00,C7,00
[16:01:06][D][Debug33:247]: Partition data 00: 81,81,00,00,00,00,00,00,00
[16:01:06][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
[16:01:06][D][Debug22:240]: Partition 1 panel data: 05,00,81,01,00,C7,00
[16:01:06][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00
[16:01:06][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[16:02:02][D][Debug22:240]: Partition 1 panel data: 05,00,81,81,00,C7,00
[16:02:02][D][Debug33:247]: Partition data 00: 81,81,00,00,00,00,00,00,00
[16:02:02][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
[16:02:02][D][Debug22:240]: Partition 1 panel data: 05,00,81,01,00,C7,00
[16:02:02][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00
[16:02:02][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'

Dilbert66, 2020-07-31:
And an 81. That 05 command on your panel is sure quirky. Ok I'll remove those with the flag as well.

Dilbert66, 2020-07-31:
Ok, pushed a new update that removes those statuses when enable05armstatus is false to prevent those bogus status. 
41 and 81 are unknown statuses so the system just defaulted to sending unknown

slipx06, 2020-07-31:
Thank you. I think you finally tamed my panel. I'll monitor overnight but it seems all the erroneous messages are gone now.

Dilbert66, 2020-07-31:
I do notice another difference with your panel data compared to mine.  paneldata[04] in my case is 81  in your case it's always 00.  Not sure what that byte represents but might in some way show if the command is valid or not.  I'll test on my end to see if I can figure out it's purpose.

mine - pc5010: Partition 1 panel data: 05,00,80,03,[81],C7,00

yours:  Partition 1 panel data: 05,00,81,01,[00],C7,00

slipx06, 2020-07-31:
Just did a quick test arming the panel and it sent a strange disarmed message after pending

[18:12:48][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
**[18:12:48][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'pending'**
[18:13:21][D][Debug22:240]: Partition 1 panel data: 05,00,8B,08,00,CF,00
[18:13:21][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
[18:13:21][D][Debug33:247]: Partition data 01: CF,00,00,00,00,00,00,00,00,00,00
[18:13:21][D][Debug22:240]: Partition 1 panel data: 05,00,8B,08,00,C7,00
[18:13:21][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
[18:13:40][D][Debug22:240]: Partition 1 panel data: 05,00,8B,08,40,C7,00
[18:13:40][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,01,00,01,00,01
[18:13:40][D][binary_sensor:036]: 'Fire Status': Sending state ON
[18:13:41][D][Debug22:240]: Partition 1 panel data: 05,00,8B,08,00,C7,00
[18:13:41][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
[18:13:41][D][binary_sensor:036]: 'Fire Status': Sending state OFF
[18:13:53][D][Debug22:240]: Partition 1 panel data: A5,00,20,15,A6,C0,00
[18:13:53][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
[18:14:09][D][Debug22:240]: Partition 1 panel data: 05,00,8B,08,00,C7,00
[18:14:09][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
[18:14:09][D][Debug33:247]: Partition data 02: D7,00,00,00,00,00,00,00,00,00,00
[18:14:09][D][Debug22:240]: Partition 1 panel data: 05,00,8B,08,00,C7,00
[18:14:09][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
[18:14:24][D][Debug22:240]: Partition 1 panel data: 05,00,8B,0C,00,C7,00
[18:14:24][D][Debug33:247]: Partition data 00: 0C,8B,00,00,00,00,00,00,01,01,00
[18:14:24][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
[18:14:24][D][Debug22:240]: Partition 1 panel data: 05,00,8B,08,00,C7,00
[18:14:24][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,01,01
**[18:14:24][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'**
[18:15:11][D][Debug22:240]: Partition 1 panel data: 27,00,8A,04,00,C7,00
[18:15:11][D][Debug33:247]: Partition data 00: 08,8B,01,00,01,00,00,01,00,01,00
**[18:15:11][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_home'**
[18:15:12][D][Debug22:240]: Partition 1 panel data: 05,00,8A,04,00,C7,00
[18:15:12][D][Debug33:247]: Partition data 00: 04,8A,01,00,01,00,00,00,00,00,00

Dilbert66, 2020-07-31:
Your system is sending a 0c (entry delay in progress) in between the pending wait for arming.  Very strange. I'll try something. If the enable05armstatus flag is false, I will have it ignore ALL 05 messages instead of just the arming and unknown ones.  
I've pushed the change in the "debug" branch for testing

[18:14:09][D][Debug22:240]: Partition 1 panel data: 05,00,8B,[08],00,C7,00
[18:14:09][D][Debug33:247]: Partition data 00: 08,8B,00,00,00,00,00,00,01,00,01
[18:14:24][D][Debug22:240]: Partition 1 panel data: 05,00,8B,[0C],00,C7,00

Dilbert66, 2020-07-31:
wait on this. It's not updating status correctly.

Dilbert66, 2020-07-31:
ok, pushed a fix.  The issue is that the 05 cmd is used for most statuses but unfortunately, your panel sends out bogus ones at odd times.   If there was a reliable way to identify a bogus message from a valid one i would do that.  I supposed I could add logic that says if the previous status is such and such, ignore the current status. I will see how I can work with that.  For now, I use the flag to disable certain known  status updates that have issues.

Dilbert66, 2020-07-31:
I'm playing with the logic and seeing that the 05 command is necessary for capturing states so we can't block all messages.  I will implement better logic to use rules to only apply certain states only if certain conditions match to filter out those bogus messages

Dilbert66, 2020-07-31:
I've created a new version in debug that removes the need for the enable flags.  To test, set all the flags to true.  As noted above, i've added some logic to prevent status changes on some bogus commands but I need to know if I got all the paths corrected.  I'd rather do it this way as you would not lose the messages which are handy to have if you they can be made accurate.
I was finding the flag approach could be problematic in some cases.

slipx06, 2020-07-31:
Thanks for all your effort on this. Will upload now and let it run over night

Dilbert66, 2020-07-31:
it's buggy. Dont use it.  armed statuses are not updating correctly.

Dilbert66, 2020-07-31:
Well, made a couple more changes and pushed to debug brance. It works well for me but the issue is will it filter out most of your false status messages.

slipx06, 2020-08-01:
So the version I run last night was stable under the armed condition. No status changes the entire night. In the disarmed state I'm still seeing lots of bogus messages. Going to try the new debug version.  Some logs below

[06:50:33][D][Debug22:240]: Panel command data: 05,00,81,11,00,C7,00
[06:50:33][D][Debug33:247]: Partition data 00: 11,81,00,00,00,00,00,00,00,01,00
[06:50:33][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '11: Alarm'
**[06:50:33][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
[06:50:33][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'triggered'**
[06:50:33][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[06:50:33][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,01,01
[06:50:33][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'
[06:50:33][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[06:51:02][D][Debug22:240]: Panel command data: 05,00,85,01,00,C7,00
[06:51:02][D][Debug33:247]: Partition data 00: 01,85,00,00,00,00,00,00,00,00,01
[06:51:02][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[06:51:02][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[06:51:32][D][Debug22:240]: Panel command data: 05,00,81,23,00,C7,00
[06:51:32][D][Debug33:247]: Partition data 00: 23,81,00,00,00,00,00,00,00,01,00
**[06:51:32][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '23: Unknown'
[06:51:32][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'**
[06:51:32][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[06:51:32][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,01,01
[06:51:32][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'
[06:51:32][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[06:52:23][D][Debug22:240]: Panel command data: 05,00,81,01,08,C7,00
[06:52:23][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[06:52:23][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[06:52:23][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[06:52:25][D][Debug22:240]: Panel command data: A5,00,20,15,B1,50,00
[06:52:25][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[06:52:53][D][Debug22:240]: Panel command data: 05,00,81,21,00,C7,00
[06:52:53][D][Debug33:247]: Partition data 00: 21,81,00,00,00,00,00,00,00,01,00
**[06:52:53][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '21: Unknown'
[06:52:53][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'**
[06:52:53][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00

[07:23:21][D][Debug33:247]: Partition data 00: 11,81,00,00,00,00,00,00,00,01,00
[07:23:21][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '11: Alarm'
[07:23:21][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
[07:23:21][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'triggered'
[07:23:21][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[07:23:21][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,01,01
[07:23:21][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'
[07:23:21][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'
[07:23:27][D][Debug22:240]: Panel command data: 27,00,81,01,00,C7,04
[07:23:27][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01

Dilbert66, 2020-08-01:
When armed the program will ignore any other messages except disarm  which eliminates those other bogus messages. When disarmed I have to put in  some logic to eliminate those other messages you get.  Those are some odd values

slipx06, 2020-08-01:
Running the latest version. One bogus "triggered" and a couple of unknown's (13,41,83,43,23,2B)

[12:12:15][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '09: No entry delay'
[12:12:15][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[12:12:15][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[12:12:15][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'
[12:12:19][D][Debug22:240]: Panel command data: 05,00,81,01,40,C7,00
[12:12:19][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[12:12:19][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[12:12:19][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[12:12:34][D][Debug22:240]: Panel command data: 05,00,81,11,00,C7,00
[12:12:34][D][Debug33:247]: Partition data 00: 11,81,00,00,00,00,00,00,00,01,00
[12:12:34][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '11: Alarm'
[12:12:34][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
**[12:12:34][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'triggered'**
[12:12:34][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[12:12:34][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,01,01
[12:12:34][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'
[12:12:34][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'

[12:14:53][D][Debug33:247]: Partition data 00: 41,81,00,00,00,00,00,00,00,00,01
[12:14:53][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '41: Unknown'

[12:30:31][D][Debug33:247]: Partition data 00: 13,80,00,00,00,00,00,00,00,00,00
[12:30:31][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '13: Unknown'

[12:32:49][D][Debug33:247]: Partition data 00: 83,80,00,00,00,00,00,00,00,00,00
[12:32:49][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '83: Unknown'

[12:53:51][D][Debug33:247]: Partition data 00: 43,80,00,00,00,00,00,00,00,00,00
[12:53:51][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '43: Unknown'

[13:05:21][D][Debug33:247]: Partition data 00: 23,80,00,00,00,00,00,00,00,00,00
[13:05:21][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '23: Unknown'

[13:55:54][D][Debug33:247]: Partition data 00: 2B,80,00,00,00,00,00,00,00,00,00
[13:55:54][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '2B: Unknown'

Dilbert66, 2020-08-01:
Some of those unknown can be programmed to be ignored but the 0x11 cannot. You need that one to indicate a trigger.  That is so odd that your system is sending a bogus 11. I'm sure your panel keypad is seeing those but since they happen so fast you won't know it.  Do you have a log of a valid alarm trigger so we can compare.

slipx06, 2020-08-01:
OK here is a full log for arming in away mode and triggering the alarm

[14:48:22][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[14:48:22][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[14:48:22][D][Debug22:240]: Panel command data: 05,00,81,01,00,C7,00
[14:48:22][D][Debug33:247]: Partition data 00: 01,81,00,00,00,00,00,00,00,00,01
[14:48:28][D][Debug22:240]: Panel command data: 05,00,83,08,00,C7,00
[14:48:28][D][Debug33:247]: Partition data 00: 08,83,00,00,00,00,00,00,01,00,01
[14:48:28][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '08: Exit delay'
[14:48:28][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'pending'
[14:50:10][D][Debug22:240]: Panel command data: 05,00,83,08,00,C7,04
[14:50:10][D][Debug33:247]: Partition data 00: 08,83,00,00,00,00,00,00,01,00,01
[14:50:10][D][Debug22:240]: Panel command data: 05,00,83,08,00,C7,00
[14:50:10][D][Debug33:247]: Partition data 00: 08,83,00,00,00,00,00,00,01,00,01
[14:50:27][D][Debug22:240]: Panel command data: 05,00,A3,08,00,C7,00
[14:50:27][D][Debug33:247]: Partition data 00: 08,A3,00,00,00,00,00,00,01,00,01
[14:50:27][D][Debug22:240]: Panel command data: 05,00,83,08,00,C7,00
[14:50:27][D][Debug33:247]: Partition data 00: 08,83,00,00,00,00,00,00,01,00,01
[14:50:43][D][Debug22:240]: Panel command data: 05,00,83,28,00,C7,00
[14:50:43][D][Debug33:247]: Partition data 00: 28,83,00,00,00,00,00,00,01,00,01
[14:50:43][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '28: Unknown'
[14:50:44][D][Debug22:240]: Panel command data: 05,00,83,08,00,C7,00
[14:50:44][D][Debug33:247]: Partition data 00: 08,83,00,00,00,00,00,00,01,00,01
[14:50:44][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '08: Exit delay'
[14:50:46][D][Debug22:240]: Panel command data: 05,00,83,08,00,C7,08
[14:50:46][D][Debug33:247]: Partition data 00: 08,83,00,00,00,00,00,00,01,00,01
[14:50:46][D][Debug22:240]: Panel command data: 05,00,83,08,00,C7,00
[14:50:46][D][Debug33:247]: Partition data 00: 08,83,00,00,00,00,00,00,01,00,01
[14:50:52][D][Debug22:240]: Panel command data: 27,00,82,05,00,C7,00
[14:50:52][D][Debug33:247]: Partition data 00: 08,83,01,01,00,00,00,01,00,01,00
[14:50:52][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'armed_away'
[14:50:52][D][Debug22:240]: Panel command data: 05,00,82,05,00,C7,00
[14:50:52][D][Debug33:247]: Partition data 00: 05,82,01,01,00,00,00,00,00,00,00
[14:50:52][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '05: Armed away'
[14:51:33][D][Debug22:240]: Panel command data: 05,00,82,05,00,C7,80
[14:51:33][D][Debug33:247]: Partition data 00: 05,82,01,01,00,00,00,00,00,00,00
[14:51:33][D][Debug22:240]: Panel command data: 05,00,82,05,00,C7,00
[14:51:33][D][Debug33:247]: Partition data 00: 05,82,01,01,00,00,00,00,00,00,00
[14:51:43][D][Debug22:240]: Panel command data: 27,00,82,05,00,C7,08
[14:51:43][D][Debug33:247]: Partition data 00: 05,82,01,01,00,00,00,00,00,00,00
[14:51:43][D][binary_sensor:036]: 'Study': Sending state ON
[14:51:44][D][Debug22:240]: Panel command data: 05,00,82,11,00,C7,00
[14:51:44][D][Debug33:247]: Partition data 00: 11,82,01,01,00,00,00,00,00,00,00
[14:51:44][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '11: Alarm'
[14:51:44][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'triggered'
[14:51:44][D][Debug22:240]: Panel command data: EB,00,01,20,15,B7,EC
[14:51:44][D][Debug33:247]: Partition data 00: 11,82,01,01,00,00,00,00,00,00,00
[14:51:45][D][Debug22:240]: Panel command data: 27,00,82,11,00,C7,00
[14:51:45][D][Debug33:247]: Partition data 00: 11,82,01,01,00,00,00,00,00,00,00
[14:51:45][D][binary_sensor:036]: 'Study': Sending state OFF
[14:51:45][D][Debug22:240]: Panel command data: EB,00,01,20,15,B7,EC
[14:51:45][D][Debug33:247]: Partition data 00: 11,82,01,01,00,00,00,00,00,00,00
[14:51:49][D][Debug22:240]: Panel command data: 05,00,80,3D,00,C7,00
[14:51:49][D][Debug33:247]: Partition data 00: 3D,80,00,00,00,00,00,01,00,00,00
[14:51:49][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '3D: Disarmed'
[14:51:49][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'
[14:51:53][D][Debug22:240]: Panel command data: 05,00,80,22,00,C7,00
[14:51:53][D][Debug33:247]: Partition data 00: 22,80,00,00,00,00,00,00,00,00,00
[14:51:53][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '22: Alarm memory'
[14:52:02][D][Debug22:240]: Panel command data: 05,00,80,62,00,C7,00
[14:52:02][D][Debug33:247]: Partition data 00: 62,80,00,00,00,00,00,00,00,00,00
[14:52:02][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '62: Unknown'
[14:52:02][D][Debug22:240]: Panel command data: 05,00,80,22,00,C7,00
[14:52:02][D][Debug33:247]: Partition data 00: 22,80,00,00,00,00,00,00,00,00,00
[14:52:02][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '22: Alarm memory'

Dilbert66, 2020-08-01:
Tks. I will try a few things and post a new version when ready for testing

Dilbert66, 2020-08-01:
Ok, I've made a change to ignore unknown commands when enable05armstatus is false and pushed to production.  Make sure both your enable flags are set to "false" . This should clean up the logs a bit.
I do not yet have a solution for the triggering since that can happen for many reasons.  1. manual panic/fire/etc , 2. door/window entry .  Since there is no preamble for a manually generate triggered alarm, there is nothing to look for since it can happen no matter what the alarm state.   For a door entry alarm, that can only be triggered if the alarm is on or after an entrydelay.

Dilbert66, 2020-08-01:
Actually I might have a potential fix for the 0x11 trigger.  I moved it to the 27 command instead of the 05.  I pushed the changes to debug.   Try a few triggers such as manual, door entry, arm, disarm, etc and see how it goes.

[58 more replies]

closed 2020-08-04 by Dilbert66