Dilbert66/esphome-dsckeybus #4: Feature Request: Sensor(s) for zone alarm status

slipx06, 2020-08-02:
Would it be possible to add sensors for each zone alarm status. This would allow you to see which zone triggered the alarm.  Something like below

```
 // Zone alarm status is stored in the alarmZones[] and alarmZonesChanged[] arrays using 1 bit per zone, up to 64 zones
    //   alarmZones[0] and alarmZonesChanged[0]: Bit 0 = Zone 1 ... Bit 7 = Zone 8
    //   alarmZones[1] and alarmZonesChanged[1]: Bit 0 = Zone 9 ... Bit 7 = Zone 16
    //   ...
    //   alarmZones[7] and alarmZonesChanged[7]: Bit 0 = Zone 57 ... Bit 7 = Zone 64	
    if (dsc.alarmZonesStatusChanged) {
      dsc.alarmZonesStatusChanged = false;                           // Resets the alarm zones status flag
      for (byte zoneGroup = 0; zoneGroup < dscZones; zoneGroup++) {
        for (byte zoneBit = 0; zoneBit < 8; zoneBit++) {
          if (bitRead(dsc.alarmZonesChanged[zoneGroup], zoneBit)) {  // Checks an individual alarm zone status flag
            bitWrite(dsc.alarmZonesChanged[zoneGroup], zoneBit, 0);  // Resets the individual alarm zone status flag
            // Appends the mqttZoneAlarmTopic with the zone number
            char zonePublishTopic[strlen(mqttZoneAlarmTopic) + 2];
            char zone[3];
            strcpy(zonePublishTopic, mqttZoneAlarmTopic);
            itoa(zoneBit + 1 + (zoneGroup * 8), zone, 10);
            strcat(zonePublishTopic, zone);
            if (bitRead(dsc.alarmZones[zoneGroup], zoneBit)) {
              mqtt.publish(zonePublishTopic, "1", true);            // Zone alarm
            }
            else mqtt.publish(zonePublishTopic, "0", true);         // Zone restored
          }
        }
      }
    }
```

Dilbert66, 2020-08-02:
Sure, not a problem

Dilbert66, 2020-08-03:
Push new version to branch alarmzone.  Also adds new debug line "verbose"

Dilbert66, 2020-08-04:
New version on master

closed 2020-08-04 by Dilbert66