Dilbert66/esphome-dsckeybus #5: Unavailable when transitioning from armed to disarmed

slipx06, 2020-08-10:
When disarming the panel it briefly shows an unavailable state before the disarmed state. Could you take a look.

[12:22:14][D][Debug22:251]: Panel command data: 05,00,8A,04,00,C7,00,C7,00,C7
[12:22:14][D][Debug33:258]: Partition data 00: 8A,04,01,00,01,00,00,00,00,00,00,00,00
**[12:22:14][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '04: Armed stay'**
[12:22:14][D][Debug11:205]: Panel data: EB,00,01,20,16,A9,DC,02,9A,00,43,00
[12:22:18][D][Debug11:205]: Panel data: C3,00,08,FF,CA,00,00,00,00,00,00,00
[12:22:19][D][Debug11:205]: Panel data: A5,00,20,16,A9,E0,00,00,64,00,00,00
[12:22:19][D][Debug22:251]: Panel command data: A5,00,20,16,A9,E0,00,00,64,00
[12:22:19][D][Debug33:258]: Partition data 00: 8A,04,01,00,01,00,00,00,00,00,00,00,00
[12:22:36][D][Debug11:205]: Panel data: 64,00,0C,70,00,00,00,00,00,00,00,00
[12:22:36][D][Debug11:205]: Panel data: 5D,00,00,00,00,00,00,5D,00,00,00,00
[12:22:36][D][Debug11:205]: Panel data: E6,00,18,01,00,00,00,00,00,FF,00,00
[12:22:36][D][Debug11:205]: Panel data: 87,00,00,00,87,00,00,00,00,00,00,00
[12:22:36][D][Debug11:205]: Panel data: BB,00,00,00,BB,00,00,00,00,00,00,00
[12:22:36][D][Debug11:205]: Panel data: A5,00,20,56,A9,E0,E2,FF,85,00,00,00
[12:22:37][D][Debug11:205]: Panel data: E6,00,1A,15,00,00,01,01,00,00,17,00
[12:22:37][D][Debug11:205]: Panel data: 75,00,00,75,00,00,00,00,00,00,00,00
[12:22:37][D][Debug11:205]: Panel data: CE,00,01,C0,00,00,00,8F,00,00,00,00
[12:22:37][D][Debug11:205]: Panel data: 11,00,AA,AA,AA,AA,AA,AA,AA,02,00,00
[12:22:37][D][Debug11:205]: Panel data: CE,00,40,FF,FF,FF,FF,0A,00,00,00,00
[12:22:37][D][Debug11:205]: Panel data: EB,00,01,20,16,A9,E0,00,E2,FF,8C,00
[12:22:37][D][Debug22:251]: Panel command data: EB,00,01,20,16,A9,E0,00,E2,FF
[12:22:37][D][Debug33:258]: Partition data 00: 8A,04,00,00,00,00,00,01,00,00,00,00,00
**[12:22:37][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'unavailable'**
[12:22:37][D][Debug11:203]: Panel data: 05,00,81,3E,00,C7,00,C7,00,C7,01,00
[12:22:37][D][Debug11:205]: Panel data: 05,00,81,3E,00,C7,00,C7,00,C7,01,00
[12:22:37][D][Debug22:251]: Panel command data: 05,00,81,3E,00,C7,00,C7,00,C7
[12:22:37][D][Debug33:258]: Partition data 00: 81,3E,00,00,00,00,00,00,00,01,01,00,00
[12:22:37][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '3E: Disarmed'
**[12:22:37][D][text_sensor:015]: 'Partition 1 Status ': Sending state 'disarmed'**
[12:22:41][D][Debug11:203]: Panel data: 05,00,81,01,00,C7,00,C7,00,C7,01,00
[12:22:41][D][Debug11:205]: Panel data: 05,00,81,01,00,C7,00,C7,00,C7,01,00
[12:22:41][D][Debug22:251]: Panel command data: 05,00,81,01,00,C7,00,C7,00,C7
[12:22:41][D][Debug33:258]: Partition data 00: 81,01,00,00,00,00,00,00,00,00,01,00,00
[12:22:41][D][text_sensor:015]: 'Partition 1 Msg ': Sending state '01: Ready'
[12:22:55][D][Debug11:205]: Panel data: E6,00,2C,80,00,00,00,00,92,00,00,00
[12:23:06][D][Debug11:205]: Panel data: E6,00,1B,00,00,01,00,00,00,00,00,00

slipx06, 2020-08-10:
I also cannot compile the latest  master version

`INFO Reading configuration /config/esphome/DscAlarm.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/alarm
Processing alarm (board: nodemcuv2; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20502.0 (2.5.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa 1.40802.0 (4.8.2)
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757 @ 1.2.2
ESPAsyncTCP-esphome @ 1.2.2 has been successfully installed!
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/alarm/.pioenvs/alarm/src/dscKeybusInterface.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/dscKeybusPrintData.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/dscKeybusProcessData.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/api_connection.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/api_server.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/list_entities.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/proto.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/user_services.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/api/util.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/logger/logger.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/restart/restart_switch.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/status_led/status_led.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/switch/automation.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/switch/switch.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/template/binary_sensor/template_binary_sensor.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/template/switch/template_switch.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/template/text_sensor/template_text_sensor.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/application.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/component.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/controller.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/esphal.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/helpers.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/log.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/preferences.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/scheduler.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/esphome/core/util.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/main.cpp.o
src/main.cpp:164:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "auto DSCkeybus = new DSCkeybushome();\n\nDSCkeybus->accessCode=\"\
   ^
Compiling /data/alarm/.pioenvs/alarm/src/working/dscKeybusInterface.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/working/dscKeybusPrintData.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/working/dscKeybusProcessData.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/working/working/dscKeybusInterface.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/working/working/dscKeybusPrintData.cpp.o
Compiling /data/alarm/.pioenvs/alarm/src/working/working/dscKeybusProcessData.cpp.o
[... 818 lines cut ...]
/data/alarm/.pioenvs/alarm/src/dscKeybusProcessData.cpp.o:dscKeybusProcessData.cpp:(.text._ZN18dscKeybusInterface17processPanel_0xA5Ev+0x10): first defined here
/data/alarm/.pioenvs/alarm/src/working/working/dscKeybusProcessData.cpp.o: In function `dscKeybusInterface::processPanelStatus4(unsigned char, unsigned char)':
dscKeybusProcessData.cpp:(.text._ZN18dscKeybusInterface19processPanelStatus4Ehh+0x4): multiple definition of `dscKeybusInterface::processPanelStatus4(unsigned char, unsigned char)'
/data/alarm/.pioenvs/alarm/src/dscKeybusProcessData.cpp.o:dscKeybusProcessData.cpp:(.text._ZN18dscKeybusInterface19processPanelStatus4Ehh+0x4): first defined here
/data/alarm/.pioenvs/alarm/src/working/working/dscKeybusProcessData.cpp.o: In function `dscKeybusInterface::processPanel_0xEB()':
[... cut ...]
/data/alarm/.pioenvs/alarm/src/working/working/dscKeybusProcessData.cpp.o: In function `dscKeybusInterface::processPanel_0xE6()':
dscKeybusProcessData.cpp:(.text._ZN18dscKeybusInterface17processPanel_0xE6Ev+0x18): multiple definition of `dscKeybusInterface::processPanel_0xE6()'
/data/alarm/.pioenvs/alarm/src/dscKeybusProcessData.cpp.o:dscKeybusProcessData.cpp:(.text._ZN18dscKeybusInterface17processPanel_0xE6Ev+0x18): first defined here
collect2: error: ld returned 1 exit status
*** [/data/alarm/.pioenvs/alarm/firmware.elf] Error 1
========================= [FAILED] Took 107.90 seconds =========================`

Dilbert66, 2020-08-10:
Do a clean build. This is most likeley due to me changing the defined pins to use numeric values instead of D1,D2,, D8 since some boards don't have those pin definitiions setup in the arduino lib.

Dilbert66, 2020-08-10:
Also the new version should correct the temporary unavailable status when disarming

slipx06, 2020-08-10:
I have tried a clean build and it is still failing with the message above

Dilbert66, 2020-08-10:
Try deleting the temporary compile directory "alarm" in your esphome directory then recompile.

slipx06, 2020-08-10:
That worked. Thanks

slipx06, 2020-08-10:
Ok running the new master and just noticed that my services have not updated to use the new naming convention esphome.dscalarm.*

[image]

slipx06, 2020-08-10:
Can confirm that the unavailable issue when transitioning from armed to disarmed is not present in the master version.

Dilbert66, 2020-08-10:
You can just use your previous names so that your services won't be impacted.  I had to change them on mine since it was conflicting with another test system I am running and causing issues.  Alternatively, you can remove the alarm integration and re-add it.
The yaml names are meant to be changed and personalized to each users system.

Dilbert66, 2020-08-10:
What I might do with the naming is just use a substitition so that it will match the system name for each zone/partition

slipx06, 2020-08-10:
Sorry just noticed my yaml was still named as "alarm". Will close the issue.

closed 2020-08-10 by slipx06