Dilbert66/esphome-dsckeybus #6: Compile error using ESPhome Addon HA

PipeDeveloper, 2020-09-14:
My config.yaml:
```
#for documentation see project at https://github.com/Dilbert66/esphome-dsckeybus
substitutions:
  systemName: "alarma"
  accessCode: "my_code" #Only comes into effect if a password prompt occurs when arming eg. night mode
  cmdWaitTime: "0" # milliseconds.  set how long to delay before accepting a new 05 cmd as valid to filter out intermittent short duration bogus commands that some panels send. default = 0

esphome:
  name: $systemName
  platform: ESP8266
  board: d1_mini
  
  includes:
    - dscKeybusInterface/

wifi:
  ssid: "my_wifi"
  password: "my_password"

  ap:
    ssid: "$systemName Fallback"
    password: "my_password"

web_server:
  port: 80
  auth:
    username: "my_user"
    password: "my_password"

logger:
  baud_rate: 0
  level: debug

api:
   password: "my_password"

ota:
   password: "my_password"
   safe_mode: True

status_led:
  pin:
    number: D4
    inverted: yes

custom_component:
- lambda: |-
    auto DSCkeybus = new DSCkeybushome();
    
    DSCkeybus->accessCode="$accessCode";
    DSCkeybus->cmdWaitTime=$cmdWaitTime;
    DSCkeybus->debug=0;
    
    DSCkeybus->onSystemStatusChange([&](std::string statusCode) {
       id(system_status).publish_state(statusCode);
    });
    DSCkeybus->onPartitionStatusChange([&](uint8_t partition,std::string statusCode) {
       switch(partition) {
         case 1: id(p1).publish_state(statusCode); break;
         case 2: id(p2).publish_state(statusCode); break;
      }
    });
    DSCkeybus->onPartitionMsgChange([&](uint8_t partition,std::string msg) {
       switch(partition) {
         case 1: id(m1).publish_state(msg); break;
         case 2: id(m2).publish_state(msg); break;
      }
    });
    DSCkeybus->onTroubleStatusChange([&](bool open) {
[... 310 lines cut ...]
                                        ^
src/dscAlarm.h:244:8: error: 'lastStatus' was not declared in this scope
    if (lastStatus[partition] != dsc.status[partition]  ) {
        ^
src/dscAlarm.h:301:41: error: 'dscZones' was not declared in this scope
    for (byte zoneGroup = 0; zoneGroup < dscZones; zoneGroup++) {
                                         ^
src/dscAlarm.h:322:41: error: 'dscZones' was not declared in this scope
    for (byte zoneGroup = 0; zoneGroup < dscZones; zoneGroup++) {
                                         ^
Compiling /data/alarma/.pioenvs/alarma/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:559:14: error: 'dsc' was not declared in this scope
       return dsc.keybusConnected;
              ^
Compiling /data/alarma/.pioenvs/alarma/lib4d9/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling /data/alarma/.pioenvs/alarma/lib4d9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling /data/alarma/.pioenvs/alarma/lib4d9/ESP8266WiFi/WiFiServer.cpp.o
src/main.cpp:560:3: warning: control reaches end of non-void function [-Wreturn-type]
   });
   ^
*** [/data/alarma/.pioenvs/alarma/src/main.cpp.o] Error 1
Compiling /data/alarma/.pioenvs/alarma/lib4d9/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
========================== [FAILED] Took 7.55 seconds ==========================
```

Dilbert66, 2020-09-14:
In file "dscalarm.h" , right below "include "esphome.h" add this line and try again.

#include "dscKeybusInterface.h"

PipeDeveloper, 2020-09-14:
Nice! it works
Thank you!

[image]

closed 2020-09-14 by PipeDeveloper