Dilbert66/esphome-dsckeybus #9: Emulated PC5108 zone adapaters

slipx06, 2020-10-15:
Hi There

I already have a PC5132 wireless zone expander working on the latest master version but thought I would try the new dev branch to get wireless battery status etc. I see that you have also added "virtual zones". My panel supports up to 32 zones using either PC5100 and/or PC5132 modules. Silly question but if I enable this does the panel require any programming to add these virtual zone types/assign partition etc or how exactly does it work? 

I currently use zigbee contact sensors  to trigger a panic alarm when the alarm is armed to emulate an open zone. With this new feature I can create two virtual zones and call the service to set the zone state as open. If the alarm is armed it should trigger, assuming the virtual zone has been setup correctly?

Dilbert66, 2020-10-15:
In the yaml, you would first select the emulater address to be a different one than you already have. So if you have a pc5132, which if you don't have any other extenders is most likely set to zones 9-16.  (address 9).  You would therefore set the emulated expander to address 10 which gives you zones 17-24.  Now on the dsc, you would as usual have to assign these zones to a parition if they are not already (I believe they default to partition 1). (program 204 (partition1) or 212 for partition2)  Then you need to setup the zone attributes in  programs 002 (zone definition). You also go to programs 117-1234 and setup the atttributes there.

slipx06, 2020-10-15:
Perfect. I'll give it a try

Dilbert66, 2020-10-15:
Let me know how it goes.  If it works fine. I will tag the master and merge this code to it under a new tag.

slipx06, 2020-10-16:
Running the new version. Seems ok so far but I have not programmed in the new virtual zone yet and it shows as closed in HA. I tried calling the new service but it has no affect. I assume that will only work was I have assigned and configured the zone definition on the panel. Simulating a power failure and AC Status is still showing as on. I thought it would update to off when I cut power to the panel?

[image]

Dilbert66, 2020-10-16:
Yes, it will not have any effect until you program the zones.  The panel will ignore the zone.   If you want to see the panel intereraction, you can set debug=3 in the lambda of the yaml and you will see the packets being sent back and forth between the module and panel cmd 0x33 for range 17-24.

What i do for testing is i enable the zone for another partition and can then see the status updates for that zone without affecting the main partition.  For example my test zones are on partition 2 but my main system is on partition 1.

I'll check into the ac status issue to see what's going on there.

Dilbert66, 2020-10-16:
If possible can you capture some packets with debug=3.  Specifically the A5  or EB cmds when you disconnect and reconnect AC power.  I'd like to see if your panel acts different.  I can't see anything off at this time.

slipx06, 2020-10-17:
Sure see below

INFO Starting log output from 192.168.1.112 using esphome API
INFO Connecting to 192.168.1.112:6053 (192.168.1.112)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.112: timed out). Re-Trying in 1 seconds
INFO Connecting to 192.168.1.112:6053 (192.168.1.112)
INFO Successfully connected to 192.168.1.112
[11:14:14][I][app:105]: ESPHome version 1.15.2 compiled on Oct 17 2020, 11:13:31
[11:14:23][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[11:14:23][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[11:14:39][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 08 79 00 00 00 00 
[11:14:39][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 18 89 00 00 00 00 
[11:14:41][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 10 81 00 00 00 00 
[11:14:41][I][ Paneldata::227]: E6: E6 00 2C 40 00 00 00 00 52 00 00 00 
[11:14:42][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 14 85 00 00 00 00 
[11:14:44][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 10 81 00 00 00 00 
[11:14:48][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 00 71 00 00 00 00 
[11:14:50][I][Moduledata::227]: 27: DF 01 FF FF FF FF FF FF FF FF 03 00 00 00 00 00 
[11:14:53][I][ Paneldata::227]: E6: E6 00 1A 15 00 00 11 01 00 00 27 00 
[11:14:53][I][ Paneldata::227]: 87: 87 00 01 00 88 00 00 00 00 00 00 00 
[11:14:53][I][ Paneldata::227]: 5D: 5D 00 10 00 00 00 00 6D 00 00 00 00 
[11:14:54][I][ Paneldata::227]: E6: E6 00 18 01 10 00 00 00 00 0F 00 00 
[11:14:54][I][ Paneldata::227]: 05: 05 00 00 17 10 C7 10 C7 10 C7 01 00 
[11:14:59][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[11:14:59][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[11:15:13][I][ Paneldata::227]: 16: 16 00 0E 32 F6 4C 00 00 00 00 00 00 
[11:15:21][I][ Paneldata::227]: E6: E6 00 1A 15 00 00 01 01 00 00 17 00 
[11:15:21][I][ Paneldata::227]: 87: 87 00 00 00 87 00 00 00 00 00 00 00 
[11:15:22][I][ Paneldata::227]: 05: 05 00 80 03 00 C7 00 C7 00 C7 01 00 
[11:15:24][I][ Paneldata::227]: 5D: 5D 00 00 00 00 00 00 5D 00 00 00 00 
[11:15:24][I][ Paneldata::227]: E6: E6 00 18 01 00 00 00 00 00 FF 00 00 
[11:15:27][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 10 81 00 00 00 00 
[11:15:30][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[11:15:30][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[11:15:33][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 14 85 00 00 00 00 
[11:15:35][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 10 81 00 00 00 00 
[11:15:35][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 18 89 00 00 00 00 
[11:15:38][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 08 79 00 00 00 00 
[11:15:39][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 18 89 00 00 00 00 
[11:15:40][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 10 81 00 00 00 00 
[11:15:48][I][ Paneldata::227]: E6: E6 00 2C 80 00 00 00 00 92 00 00 00 
[11:15:48][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 00 71 00 00 00 00 

Looks like the sensor changes to unavialable
[image]

slipx06, 2020-10-17:
Programmed the virtual zone and everything working as expected. So far so good

Dilbert66, 2020-10-17:
The logs you show dont match the time .  They are after, so not sure if it's the same event.   The Ac power status normally comes in an A5 or an EB cmd and I don't see those.

slipx06, 2020-10-17:
OK synchronized time to HA. Lets try this again

[image]

Power off 
[16:16:53][I][ Paneldata::227]: 87: 87 00 01 00 88 00 00 00 00 00 00 00 
[16:16:54][I][ Paneldata::227]: 05: 05 00 90 03 10 C7 10 C7 10 C7 01 00 
[16:17:01][I][ Paneldata::227]: 27: 27 00 90 03 10 C7 12 A3 00 00 00 00 
[16:17:03][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:17:03][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:17:05][I][ Paneldata::227]: 27: 27 00 90 03 10 C7 02 93 00 00 00 00 
[16:17:05][I][Moduledata::227]: 27: FF 01 FF FF DF FF FF FF FF FF 03 00 00 00 00 00 
[16:17:05][I][ Paneldata::227]: 33: 33 00 FF FF FF FF FF 00 00 00 00 00 
[16:17:05][I][Moduledata::227]: 33: FF 01 75 55 FF FF 6F 01 00 00 00 00 00 00 00 00 
[16:17:05][I][ Paneldata::227]: 34: 34 00 90 03 10 C7 04 A2 00 00 00 00 
[16:17:06][I][ Paneldata::227]: 27: 27 00 90 03 10 C7 12 A3 00 00 00 00 
[16:17:06][I][ Paneldata::227]: 27: 27 00 90 03 10 C7 16 A7 00 00 00 00 
[16:17:07][I][ Paneldata::227]: 27: 27 00 90 03 10 C7 12 A3 00 00 00 00 
**[16:17:08][I][ Paneldata::227]: 5D: 5D 00 10 00 00 00 00 6D 00 00 00 00** 
[16:17:08][I][ Paneldata::227]: E6: E6 00 18 01 10 00 00 00 00 0F 00 00 
[16:17:08][I][ Paneldata::227]: 05: 05 00 00 17 10 C7 10 C7 10 C7 01 00 
[16:17:09][I][ Paneldata::227]: 27: 27 00 00 17 10 C7 1A 2F 00 00 00 00 
[16:17:10][I][Moduledata::227]: 27: FF 01 FF FF DF FF FF FF FF FF 03 00 00 00 00 00 
[16:17:10][I][ Paneldata::227]: 33: 33 00 FF FF FF FF FF 00 00 00 00 00 
[16:17:10][I][Moduledata::227]: 33: FF 01 55 75 FF FF 6F 01 00 00 00 00 00 00 00 00 
[16:17:10][I][ Paneldata::227]: 34: 34 00 00 17 10 C7 00 22 00 00 00 00 
[16:17:11][I][ Paneldata::227]: 27: 27 00 00 17 10 C7 12 27 00 00 00 00 
[16:17:11][I][ Paneldata::227]: 27: 27 00 00 17 10 C7 02 17 00 00 00 00 
[16:17:18][I][ Paneldata::227]: 16: 16 00 0E 32 F6 4C 00 00 00 00 00 00 
[16:17:26][I][Moduledata::227]: 16: FF 01 FF FF FF FF FF FE FF FF 03 00 00 00 00 00 
[16:17:33][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:17:33][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:17:48][I][ Paneldata::227]: E6: E6 00 2C 04 00 00 00 00 16 00 00 00 
[16:18:03][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00

After AC failure communication delay
[16:25:48][I][ Paneldata::227]: E6: E6 00 2C 01 00 00 00 00 13 00 00 00 
[16:25:48][I][ Paneldata::227]: B1: B1 00 DF 0F 07 00 00 00 00 00 A6 00 
[16:26:03][I][ Paneldata::227]: 05: 05 00 00 17 10 C7 10 C7 10 C7 01 00 
[16:26:03][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:26:03][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:26:06][I][Moduledata::227]: 11: FF 01 FF FD FF FF FF FF FF FF 03 00 00 00 00 00 
[16:26:11][I][Moduledata::227]: 11: F7 01 FF FF FF FF FF FF FF FF 03 00 00 00 00 00 
[16:26:18][I][ Paneldata::227]: B1: B1 00 DF 0F 07 00 00 00 00 00 A6 00 
[16:26:30][I][Moduledata::227]: B1: FF 01 7F FF FF FF FF FF FF FF 03 00 00 00 00 00 
[16:26:33][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:26:33][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:26:45][I][ Paneldata::227]: 05: 05 00 00 17 10 C7 10 C7 10 C7 01 00 
[16:26:48][I][ Paneldata::227]: E6: E6 00 2C 02 00 00 00 00 14 00 00 00 
[16:26:48][I][ Paneldata::227]: B1: B1 00 DF 0F 07 00 00 00 00 00 A6 00 
[16:26:48][I][Moduledata::227]: B1: FF 01 FF FB FF FF FF FF FF FF 03 00 00 00 00 00 
[16:27:03][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:27:03][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 

Power Restore

[16:28:01][I][Moduledata::227]: 27: FF 01 FF FF FF F5 FF FF FF FF 03 00 00 00 00 00 
[16:28:03][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:28:03][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:28:07][I][ Paneldata::227]: E6: E6 00 1A 15 00 00 01 01 00 00 17 00 
[16:28:07][I][ Paneldata::227]: 87: 87 00 00 00 87 00 00 00 00 00 00 00 
[16:28:07][I][ Paneldata::227]: 05: 05 00 80 03 00 C7 00 C7 00 C7 01 00 
**[16:28:08][I][ Paneldata::227]: 5D: 5D 00 00 00 00 00 00 5D 00 00 00 00** 
[16:28:08][I][ Paneldata::227]: E6: E6 00 18 01 00 00 00 00 00 FF 00 00 
[16:28:15][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 12 83 00 00 00 00 
[16:28:18][I][Moduledata::227]: 27: FF 01 FF FF DF FF FF FF FF FF 03 00 00 00 00 00 
[16:28:18][I][ Paneldata::227]: 33: 33 00 FF FF FF FF FF 00 00 00 00 00 
[16:28:18][I][Moduledata::227]: 33: FF 01 75 55 FF FF 6F 01 00 00 00 00 00 00 00 00 
[16:28:18][I][ Paneldata::227]: 34: 34 00 80 03 00 C7 04 82 00 00 00 00 

After AC failure communication delay
[16:36:30][I][ Paneldata::227]: 27: 27 00 80 03 00 C7 02 73 00 00 00 00 
[16:36:44][I][ Paneldata::227]: 2D: 2D 00 80 03 00 C7 08 7F 00 00 00 00 
[16:37:02][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:37:02][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:37:20][I][ Paneldata::227]: E6: E6 00 0B 00 F1 00 00 00 00 00 00 00 
[16:37:38][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:37:38][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:37:56][I][ Paneldata::227]: A5: A5 00 20 1E 29 70 00 00 7C 00 00 00 
[16:38:14][I][ Paneldata::227]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 
[16:38:14][I][Moduledata::227]: 11: FF 01 3F FF FF F3 FF FF FF 07 00 00 00 00 00 00 
[16:38:31][I][ Paneldata::227]: E6: E6 00 2C 01 00 00 00 00 13 00 00 00 
[16:38:32][I][ Paneldata::227]: B1: B1 00 DF 0F 07 00 00 00 00 00 A6 00

slipx06, 2020-10-17:
BTW how do I program no delay for ac failure under section 377. Would it be

*8 <code> 377 003 003 003 000 **000** ##

The 5th variable controlling the delay

slipx06, 2020-10-17:
Interesting. I got it to work but had to set the ac failure communication delay to 0

[17:03:22][I][ Paneldata::227]: EB: EB 00 00 20 1E 29 C4 00 E8 FF FD 00

[17:04:58][I][ Paneldata::227]: EB: EB 00 00 20 1E 29 C8 00 F0 FF 09 00 

[image]

Dilbert66, 2020-10-17:
That's interesting. I was looking at your logs and without the 0xEB, the only notification was the 0x05 command which sends a power failure message. The only problem is that on restore, it just sent out the regular statuses and no specific power notification. 
There might be a bit I'm missing but it was different and I didnt want to add logic to mess up other panels.  It's interesting that you had to specifically turn of the delay on your panel to see the message.  I'll have to see how mine is set.  Now we know at least.

Dilbert66, 2020-10-17:
Those message restoral messages ? Are they on the physical panel?    The fact you see different messages means there are different commands used.

slipx06, 2020-10-17:
I have another module plugged into the Keybus that sends me SMS messages. They come from it.

Dilbert66, 2020-10-17:
Does it send that message this exact way "A/C mains power failure/restore" or does it send one for failure and one for restore?

slipx06, 2020-10-17:
No two separate messages

A/C mains power failure
A/C mains power restore

Dilbert66, 2020-10-17:
I think I found the command that works both ways.  
It's the E6 1A  . If you get a chance can you post logs again of power off and power on again. The flag works for my panel and I seen the command in one of your dumps so it might work.

Dilbert66, 2020-10-17:
I've pushed a new version that uses the e6 - 1a command to detect ac power failure.  Works well on my system.

slipx06, 2020-10-17:
I'll give it a try. BTW do you have an ESPHome sketch that outputs data similar to the taligentx dscKeybusInterface example sketch so that I can post more data for that project. I dont really have a test bench so this would make testing much easier if I can flash OTA.

Dilbert66, 2020-10-17:
The version you have already outputs like the example sketch if you set debug=3. It does that via the usb/serial at 115kb.   The logging on the esp console is only a simple version since the api can't handle too much traffic.   but if you connect via serial/usb you will get the exact same data as the example sketch.

slipx06, 2020-10-17:
OK. Is it safe to use the usb port while thye esp is powered through the 5v Pin from the panel at the same time?

Dilbert66, 2020-10-17:
I'm pretty sure it is. I do it all the time.   There is diode that prevents vin voltage from going back to the pc side.

slipx06, 2020-10-18:
> I've pushed a new version that uses the e6 - 1a command to detect ac power failure. Works well on my system.

Tested and working. Thanks

Dilbert66, 2020-10-18:
Awesome. This for testing it

slipx06, 2020-10-18:
> I'm pretty sure it is. I do it all the time. There is diode that prevents vin voltage from going back to the pc side.

I manged to capture some keybus data. Unfortunately I could not pick up any strange behavior even with cmdWaitTime set to 0.  Anyway attaching output here if it helps

[alarm.log](https://github.com/Dilbert66/esphome-dsckeybus/files/5397771/alarm.log)

Dilbert66, 2020-10-18:
Cool ! I'll have a look!

slipx06, 2020-10-18:
Ok. Had to delete that log as it contains some sensitive data. Here is a sanitized version 
[alarm.log](https://github.com/Dilbert66/esphome-dsckeybus/files/5397817/alarm.log)

wkchick, 2020-12-24:
@Dilbert66 I found your idea of emulated zone expander is great! However it seems I can't really follow the way to use it. I got a situation similar to slipx06 just much, much simpler. I use a few Zigbee motion and window sensors connected to HA through a dCONZ II gateway that I would like to integrate it to the alarm panel PC1832. I don't have any PC5108 so expanderAddr1: "9", expanderAddr2: "0". I have no idea how to feed status of a binary sensor, say binary_sensor.family_motion, from HA to dscKeybus. Do I need to, for each and every virtual sensor, repeat the following code segment listed in your yaml file.
on_press:
  - lambda: |-
      dsc.setZoneFault(9, 1)
on_release:
  - lambda: |-
      dsc.setZoneFault(9, 0)

esphome is new to me and this is my first project. Before this, I use code from https://github.com/taligentx/dscKeybusInterface on a Arduino UNO+Ethernet using MQTT with HA for about a year and am still using it.

Thanks.

Dilbert66, 2020-12-24:
You just set your event in your home assistant config to trigger the zone fault using the esphome dsc service that is created by the integration.  So in HA when your zigbee motion detector is triggered, you call the esphome.set_zone_fault service with the zone you want.  See docs below (from the dev branch)

A new service is also made available in home assistant called esphome._set_zone_fault with attributes zone and fault: zone is 2 digit zone number and fault is 0 or 1 where 1 sets a zone as open and 0 closes it. By calling this service from a home assistant function, you can enable any event, sensor, etc to trigger any one of your emulated zones in your alarm system.

wkchick, 2020-12-24:
@Dilbert66, Thanks so much for your quick response. Does that means there's nothing to do with the dscalarm.yaml file and all is accomplished inside the automation.yaml under HA? So we are not importing states through the home assistant binary sensor platform documented here: [https://esphome.io/components/binary_sensor/homeassistant.html]?

Perhaps I was misled by the approach described above. Can you please provide a short HA yaml code segment, let say binary_sensor.family_motion to zone 9 of the PC1832 through the emulated zone expander? Many thanks.

Dilbert66, 2020-12-24:
Yes , it's done from an HA automation.  You just setup an automation with a trigger from your binary sensor motion and use a call service for the action if that trigger runs.. ie for the action.  The service name will depend on what your dsc implementation is called.

service: esphome.dscalarm_set_zone_fault
data:
  zone: 9
  fault: true

Juanko84, 2021-01-05:
> Running the new version. Seems ok so far but I have not programmed in the new virtual zone yet and it shows as closed in HA. I tried calling the new service but it has no affect. I assume that will only work once I have assigned and configured the zone definition on the panel. Simulating a power failure and AC Status is still showing as on. I thought it would update to off when I cut power to the panel?
> 
> [image]

Hi,

Thank you for the great thread. I setup everything and all is working 100%.
How do you set the timestamp to show when last it was triggered?

slipx06, 2021-01-05:
You need to edit the entities card in HA and set the secondary info for each entity to last changed 

[image]

closed 2021-01-21 by Dilbert66