Dilbert66/esphome-dsckeybus #10: Schematic Wiring Discussion

PipeDeveloper, 2020-11-04:
This is a "rare" case that i detect on my system.

On this case im using a **DSC PC1832** alarm panel and **DSC UA521** RF receptor.
My system got 30 wireless zones, 27 RF reed switch, 3 wireless infra red sensors and the siren tamper. Total 31 zones. All seems to be detected by the ESP8266 fron nodeMCU and with my Home Assitstant.

NodeMCU is connected with this shield
https://es.aliexpress.com/item/32966490069.html?spm=a2g0o.productlist.0.0.481c2635liIuN4&algo_pvid=fea52fd7-22a1-4c99-9f26-2853faba24a9&algo_expid=fea52fd7-22a1-4c99-9f26-2853faba24a9-10&btsid=0bb0622f16045296078108429e0110&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

DC power jack is connected directly to RED and BLK terminal con my DSC PC1832 pannel (it provides 12v to power the NodeMCU correctly)

### This is the issue:
zone 1 and 2 respond inmidiatly after they open or close.

zones from 3 to 31 didn't respond, some of them respond random status open or close even hours after you touch it. They just keep the current status after NodeMCU boots at first.

### Some test i did.
This was a recomendation by a tecnitian from ADT. One of the thing is that the pull up resistors in some way are making some interference, lowering the digital signal voltage and the pannel doesnt recognice the changing status of the zone.

1. So the first thing i did was unplug the NodeMCU and check zone status only with the keypad. The result was all zones are working correctly, in live the keypad *beeps* each time i open or close windows and doors, so the issue is beween the **NodeMCU** and the **DSC UA521** RF receptor

2. So second test, "maybe the NodeMCU is taking a lot a power, so the DSC UA521 have less RF range and thats why the receptor can't recognize". The Keypad says the number "5" meaning that the RF receptor have a wiring issue.
Lets unplug the NodeMCU without touching anything, just the RED terminal. The result was that the numbre "5" dissapear and all zones start working properly again.

3. Third test, if the RED terminal provides power to the RF receptor, so lets find another power source. The tecnitian suggest the AUX terminal which provides 12v as well independet from RED terminal. So i plugged the NodeMUC to the AUX and seems to boot normally... BUT, the issue was excatly the same, zones 3 - 31 not updating on open/close status + the number "5" on the keypad.

**Temporal Solution** 🎉 
4. So the tecnitian suggest to using an external source of power, on this case was a little difficult because it needs the BLK terminal to make work the serial comunication. So i just unplug the the 12v input and just leave the BKL, YEL and GRN wires plugged. I take a phone charger 5v/1A (externally) and plugged into the microUSB port of the NodeMCU.
The result: all zones are working perfectly as expected, from 1 to 31, even the IR sensors. All the comunication is very fluid.

So in conclussion it's highly recomended to use an external power source to power the NodeMCU, Im not sure what is the exactly cause of this issue but i hope @Dilbert66 you can reproduce it and make your conclussions. 
The next step on my case is how organize the electrical wiring to plug the phone charger on the same box of the alarm pannel but as i see is not so easy 👀

Dilbert66, 2020-11-04:
Without seeing the actual voltage values and oscilloscope view of the voltage supplied by the shield, I would guess that that this shield is not providing clean power to the nodemcu.   Without clean voltage you will get erratic operations which is why when you use a separate source all is good.    I recommend an adjustable LM2596 or MP1584EN buck converter module to convert the 12volts to 5v or 3.3 volt. This is what I use and have never seen problems  You can power the node mcu using that instead of using the output from the shield.    

If you can measure the output voltage going to the nodemcu from the shield , this confirm the issue.  The problem is not the resistors or other wiring. A bad ground connection could also cause a problem.

PipeDeveloper, 2020-11-05:
I have double check all conection, twisted all wires, put them again on terminal, and secure them strongly.

I measure the ground resistance on many points.

Between the **pannel terminal (BLK screw)** and the **UA521 RF receptor terminal)** = 0.1 ohm
Between the **pannel terminal (BLK screw)** and the **NodeMCU GND pin** = 0.2ohm
Between the  **NodeMCU GND pin** and the **UA521 RF receptor terminal)** = 0.2ohm

When Imeasure the resistance only with the multimiter test points, it shows 0.1ohm so i think is fine.

I tried again powering the NodeMCU by the RED and BLK terminal, and seems to work with some zones, but with some exceptions.
Just to give examples, zone 17,26,27,28,29,31 are not working (open/close states not detected even by the pannel).

So i measure the voltage between RED and BLK terminals = 13.70v (all plugged)
13.17v with the AC power disconnected (also with the NodeMCU Plugged)

Te result was the same, the zones mentioned above not responding.

I unplugged the NodeMCU just the RED cable discarding it for the equation and measure the voltage again.
the result was 13.68v between RED and BLK terminals.
All zones working properly with the NodeMCU unplugged.

Now the curius thing is that the zones mentioned above have in common they are the most away from the receptor, so im thinking that when i plug the NodeMCU at the RED or AUX terminal, im pretty sure that it is affecting on the distance range of the receptor.

Dilbert66, 2020-11-05:
I am still suspecting an issue with shield not providing correct voltage to the node mcu.  When you say that you plug the nodemcu to the red terminal, are you plugging 12 volts to the vin of the nodemcu or are you plugging the red terminal to the shield which then sends power to the nodemcu? If so, what voltage is it sending to the nodemcu to power it?   You should not power the nodemcu directly from 12 volts or 13.7 in this case. (even though the specs say you can) since it will get very hot and need to dissipate about 10 volts of power differential in heat. (since it only needs 3.3 volts to operate).   It is an odd situtation. I have never encountered it and have 2 different systems with the same circuit.  I definitvely suspect the power adapter in the shield in this case as the culprit.  That's the only thing it can be.

PipeDeveloper, 2020-11-05:
RED terminal is connected to the shield (on the DC power jack) Like this picture.
[image]

Maybe you're right, the NodeMCU shield is doing something strange, not sure if it is affecting the digital signals, maybe power consumption, but definetly is doing something. If you wish i can donate you one to make some testing, they are pretty cheap. and thats atractive to those who whats to build this system with things that are handly.

The main thing on the issue is when NodeMCU (actually the shield) is connected on RED or AUX terminal. Some zones are not detected by the RF receptor and with the central pannel.

This is a real example.
All zones closed, the keypad indicates the green light ready to arm.
When nodeMCU is plugged, and i open zone 31, the green light remains on saying ready to arm, and the Home Assistant says: zone 31 closed. If i press *2## nothing happen, meaning that all zones are closed.
Same example if i open zone 15 (this one wokrs fluid), when i open green light turns off, Home Assistant says: zone 15 open.

What happen if i unplug the NodeMCU.
Same example, all zones closed with the green light on the keypad ready to arm.
if i open zone 31, green light inmidiatly turns off, pressing *2## it says that zone 31 is open. if i close it green light turns on, all very fluid.

As i did on the temporal solution, using an external power from a phone charger via microUSB on the NodeMCU seems to solve it by leaving the shield outside the equation.

Dilbert66, 2020-11-05:
I do believe the power supply of the shield board is introducing noise or an unstable voltage to the nodemcu.  When that's the case, you will get unstable operation of the controller.   It definitively sounds like noise is appearing on the keybus .  When you power externally , you don't get the noise since the shield power supply is out of the circuit.   This is why I wanted to see the actual voltage going to the nodemcu from the shield .  An oscilloscope of course would be needed for that task but most people have no need for one of those .. So in this case we guess.

You have to remember that a lot of offshore electronics to keep the cost down are built with substandard components . I suspect this is the case here.   Again, without concrete evidence, this is only an assumption.

PipeDeveloper, 2020-11-06:
I was guessing the same. Thank you very much.
Maybe in the future you may order one of this shields and make some testing with an osciloscope.

I'll use external power but temporally, my idea is to make a PCB to have all organized, with a friendly wiring, like plug and play. Not necesary for NodeMCU, could be for a Wemos D1 or wemos D1 mini also.

PipeDeveloper, 2020-12-22:
Hello @Dilbert66

Its time to do the things right, so im designing a PCB to make the final device without any issue.
I would like to have your opinion about the design, how can i improve, maybe add some pins for develpment purposes but basically a functional pcb with this library.

First about the module:
I dont know if you use LM2596 as the transistor only or as the module, but i choose the module (see picture) as it is ready, cheap and i skip to buy each component separately and solder it, just more simple, solder, plug, screw till your voltage is correct, plug and play.
[image]
[image]

Second about the input for nodeMCU, i designed two jumpers, one for power it directly to VIN pin, some people says that they dont recomend to power from it due the internal regulator of nodeMCU sucks, but others may want to power it from 5v. Im concious that LM2596 have an eficiency and internal reulator from nodeMCU too, so if you convert 12v to 5v and then 5v to 3.3 its not so eficient.
The second jumper powers the nodeMCU directly to 3.3v As i can see LM2596 can convert 12v to 3.3v. I dont know if this is the best, but i leave both options.

I made some holes to just place the board on the top of the other and just solder the pads, so through hole components such th capacitor and the potenciometer dont bulk between each board.

And third, after finish i would like to share the .sch and .brd files, im agree that all people can use it, but obviusly im not a pro on electronic, so i leave this to your expert hand, im sure that you may improve this a lot.

So the schematic.
[image]

And the board 2layers (just the picture)
top
[image]

bottom
[image]

I didnt ordered anything yet, but if you think that is fine, lets do this

Dilbert66, 2020-12-22:
Nice. Myself,  I use the module for my project and feed directly to 3.3 volts bypassing the 5 volts.  The nodemcu 5 volt regulator is fine though. It's the d1 wemos mini clones that have a crappy 5volt regulator.  Some suggestions. For jp2/jp4 I would recommend a 3 pin jumper with center being out+.  This way, the jumper can only be one of the two or none.  No mistakes can happen by someone having jp2 and jp4 in at the same time. Also saves having an extra jumper.  I don't understand jp3? Why would you want to short out+ and out-?

Dilbert66, 2020-12-22:
You might also want to label your lines for jp1 so no confusion occurs.

PipeDeveloper, 2020-12-22:
Nice! good recomendation. i'll merge the jp2 and jp4 in a 3pin jumper.

jp3 you're right, its a voltage out connector, but i designed with many purposes, more than power another devices.
it may be used in case that someone wants to power directly from a 5v or 3.3v without use the LM2596 such a phone charger or an external power supply, obviusly using the jumper in the right position, without using the uUSB
I dont know the correct label for that pins that can be used for input or output.
As i understand
VDD for me is only for output as voltage drain
VCC as voltage current collector as input

Dilbert66, 2020-12-22:
For the output connector, normally the designator is just J as in J1 (for jack) since it's not a jumper.

PipeDeveloper, 2020-12-22:
Done!
I decided to leave it as "Vc" or voltage common, i dont know if it is technically correct call it as this, but i cannot put an specific value because the voltage on that line may vary depending on who is using it.

Final result:
[image]

[image]

[image]

If you agree, i can create a pull request if you want to share and feel free to modify it

Dilbert66, 2020-12-22:
VCC is best to use I think.  Or Vout might be more descriptive

Dilbert66, 2020-12-22:
Looks good! Maybe another version that supports the D1 mini ?  Go ahead and create a pull request.

PipeDeveloper, 2020-12-22:
ok! last mod, changing to VCC and making PL

closed 2020-12-22 by PipeDeveloper

PipeDeveloper, 2020-12-23:
If i got some time i'll make a board to wemos D1 mini :smile:
Im not an expert on components but to make this work on a D1 mini, it needs a small voltage regulator to convert 12v to 5v or 3.3v and obviusly some capacitors, an inductor maybe (?) and the corresponding resistor as the main schematic.

like this one
https://www.aliexpress.com/item/33049921166.html?spm=a2g0o.productlist.0.0.1793136d0X621y&algo_pvid=c8bb7b53-14ba-43d3-bc01-8ab5bdd64704&algo_expid=c8bb7b53-14ba-43d3-bc01-8ab5bdd64704-30&btsid=0b0a555916086964576944896e4e90&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Im open to suggestions, i never tried this module and i dont know if it may have troubles like the NodeMCU board noise in the bus signal (suspected). But anyway build this is more complicated (searching for components, and then solder those thiny 0603?) than buy this thing separately and just design a board for yellow and green connector with just the resistors and transistor. Like make a pie with 3 boards (?)

Dilbert66, 2020-12-23:
Actually the wemos schematic is exactly the same as using the nodemcu.  I use it myself. I use it with an lm2596 module to feed 3.3 volts to it.  There is no need for capacitors,inductors, etc.   The only real difference is that it has a smaller footprint.

Dilbert66, 2020-12-23:
Just to clarify, I mean a board using the d1 mini with the lm2596 regulator instead of a nodemcu and lm2596.  The d1 mini is very popular because it is very inexpensive and works just as well and smaller.  As long as you don't use the 5v regulator and feed directly to the 3.3 volt input with the lm2596.  You would not need a voltage jumper for the 5 volt vin (it does have one but it is too weak)

wkchick, 2020-12-24:
@PipeDeveloper, May I ask what software did you use to prepare the PCB artwork? I just finished soldering the resistors and transistor to the Wemo D1 mini prototype shield this morning and it is now up and running. I would also like to make a PCB for it.

PipeDeveloper, 2020-12-24:
Eagle by Autodesk
https://www.autodesk.com/products/eagle/free-download

I could desing just a simple board for wemos d1 mini with some resistors, the transistor and the terminal block
3.3 or 5v input
YEL
GRN
BLK

So you need to find how to power it, by using an external power source or using the RED terminal + LM2596 module but not on the same wemos d1 mini board, it should be hanging on the air so you also need how to secure.

My idea was to have all components on a small single board with only four terminal block, just to have all things secured and elegant but sounds complicated, since it needs small components (and also difficult to find), and soldering skills.

Let me see what happens, i could make one in the few days

Dilbert66, 2020-12-24:
I think there is some confusion here.  The d1 mini is really just a smaller version of the nodemcu.  So instead of the nodemcu module, you could use a d1 mini module.  Same layout, and still using the lm2596 module and all other same components of your original  design.  No need to overcomplicate this . It was just a suggestiong. That's all.

PipeDeveloper, 2020-12-24:
@Dilbert66 Ohhh now i understand! hahaha thats easy and faster, hold on!

Dilbert66, 2020-12-24:
Yes, exactly! Nice.

Dilbert66, 2020-12-24:
One caution you need to add in your documentation though, is to make sure that the lm2595 needs to have the voltage adjusted to 5v or 3.3 BEFORE the 5v/3.3 volt jumper is installed, or the module could be fried on power up.

wkchick, 2020-12-24:
@PipeDeveloper Thanks for your information. This is my setup based on Wemos D1 mini.
[image]

Several resistors (1/4W MF) and transistor 2N3904 are soldered under the D1 mini. I am now using an external 5V AC adapter to power it but planned to use [this](https://www.canadarobotix.com/products/2227) 5V Step-up/Step-down regulator to power from 12V of PC1832. The regulator should be small enough to fit on the prototype shield. The 3 pin terminal post is reserved for additional I/O for garage door sensors input (D5, D6) and strobe/siren output relay (D7). Despite size of D1 mini is small, it is very powerful. PCB designed will definitely bigger in size than the D1 mini to accommodate the terminal post but resistors can still be placed under it.

In the photo, behind the D1 mini is an Arduino UNO + Ethernet shield that runs the dscKeybusInterface library by taligentx.

Dilbert66, 2020-12-24:
I would highly recommend powering on the 3.3 volt input instead of the 5 volts input for the d1.  The 5volt regulator on many clones is underpowered and will fry on it's own and take the esp chip with it. I've seen that twice myself.   Use a 12v -> 3.3 volt power source instead.

wkchick, 2020-12-24:
In such case, the LM2596 is probably the best choice. It's even much less expensive than the one I planned to use.

Dilbert66, 2020-12-24:
The mp1584en modules are smaller ,cheap and work well also.

PipeDeveloper, 2020-12-24:
I think is done.
@Dilbert66 please check de documentation and the files before i make the pl

https://github.com/PipeDeveloper/esphome-dsckeybus/tree/master/PCB%20Layouts/WemosD1%20Mini/SMD

Dilbert66, 2020-12-24:
There's a couple typos, etc. you might want to check into.  My documentation always has some typos also as English is not my first language.

PipeDeveloper, 2020-12-24:
Hahaha i know, english also is not my first language. Let me check and try to simplify some things, i want to be the most clear possible, but tomorrow.
When I feel that everything looks good i'll make PL.
Mary xmas!

PipeDeveloper, 2020-12-27:
Done. If i got some time i'll make the through hole version.