Dilbert66/esphome-dsckeybus #11: ESPHome 1.15.3

slipx06, 2020-11-13:
Getting a compile error with the new ESPHome version.  Any ideas

```
Linking /data/alarm/.pioenvs/alarm/firmware.elf
/root/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/alarm/.pioenvs/alarm/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [/data/alarm/.pioenvs/alarm/firmware.elf] Error 1
```

[image]

Dilbert66, 2020-11-13:
Yes, I see that. Tks.  I had not upgraded yet. .  Looks like some of the updated libraries used by esphome are using a lot more ram 
 than they were before. I have to see how to trim ram use.

Dilbert66, 2020-11-13:
You can downgrade but you then need to delete the temporary compilation directory first or it will try and use the previous compile

Dilbert66, 2020-11-14:
I've pushed a new version to the "dev" branch that compiles fine on the esphome 1.15.3.   A few changes needed to reduce memory useage. The problem is that esphome and the various libraries included, use a lot of ram which does not leave much for other use.  A lot of functions use ISR's which require the functions to live in RAM to function correctly so some memory size tricks and optimizations are needed.

slipx06, 2020-11-14:
Thank you. Confirmed working int the new version

closed 2021-01-21 by Dilbert66