Dilbert66/esphome-dsckeybus #12: dsckeybus Fault

thunderwood19, 2020-12-16:
After hooking everything up, I am currently getting a keybus fault error on the main alarm panel and am unable to use the Alarm system.

Dilbert66, 2020-12-17:
This would be a problem with your wiring.  Bad ground, etc.

thunderwood19, 2020-12-17:
I believe that to be the issue aswell, just can't seem to diagnose the point of failure. Grounds are all tied, wiring diagram was followed to the t. 
[image]

Dilbert66, 2020-12-17:
transistor pinout correct? Correct values for resistors? etc.  Something is shorting one of the lines or a bad ground, those are the only possibilities as far as I can see.

thunderwood19, 2020-12-17:
Just got into wiring, but resistors are simple enough. I think I got the transistor correct : 
[image]

Left: to green wire
Center: to 1k resistor and then do
Right: to ground 
Transistor is a 2222a

Dilbert66, 2020-12-17:
e b c.  E=ground,b=1k, c=green wire. I think you have e and c backwards.

thunderwood19, 2020-12-17:
Ok, so that was a problem. Switched the ends around and now it seems im a little further along. The issue now is that the alarm system only works when the black wire is not connected to the nodemcu

thunderwood19, 2020-12-17:
Re did wiring on a new board:
[image]
Left to nodemcu, riright to alarm panel. 

Connected to panel:
[image]

Wires to ground bundled and then connected to other grounds on panel. 

Panel is currently spazzing out and sensors aren't responding.  Nodemcu was flashed with latest and zones were configured prior in the yaml.

Dilbert66, 2020-12-17:
You still have a wiring issue somewhere.

thunderwood19, 2020-12-17:
Resistors are all showing as their proper values (10kohms,  33kohms, and 1kohms). The following are the Resistors im using. 
[image]

Dilbert66, 2020-12-17:
Well, I don't have any other suggestion. It's still a wiring issue somewhere.

Dilbert66, 2020-12-17:
One part that does not make sense in your wiring is this resistor?  The nodemcu ground should be direct to the panel ground. No resistor in between.
[image]

Dilbert66, 2020-12-17:
I don't see the 1k resistor between the nodemcu and the transistor base.

thunderwood19, 2020-12-17:
The misplaced 1k resistor was it. I really appreciate your help with troubleshooting this as I learned alot from this experience. Keep doing amazing work!

closed 2020-12-17 by thunderwood19

Dilbert66, 2020-12-17:
Good stuff! Have fun!