Dilbert66/esphome-dsckeybus #13: Zone "open" Fault

thunderwood19, 2020-12-23:
Approx every 24 hours (Between 1700-1900), the panel starts beeping throwing an error for an open zone.  The Zone is always the same (Zone 14). The kicker is that this zone is not in use.  Only zones 9-13 are.  While this error is thrown, I am unable to interact with the system via HA.  The only way to temporarily resolve the issue and allow arming/disarming is to power cycle the system.  

Logs:

[18:48:27][D][Debug22:247]: Panel command data: A5,00,20,32,D2,C0,00,00,89,00
[18:48:27][D][Debug33:254]: Partition data 00: 90,03,00,00,00,00,00,00,00,00,00,00,00
[18:48:28][D][Debug22:247]: Panel command data: 05,00,90,07,10,C7,10,C7,10,C7
[18:48:28][D][Debug33:254]: Partition data 00: 90,07,00,00,00,00,00,00,00,00,00,00,00
[18:48:28][D][text_sensor:015]: 'dscalarm Partition 1 Msg ': Sending state '07: Failed to arm'
[18:48:31][D][Debug22:247]: Panel command data: 05,00,90,03,10,C7,10,C7,10,C7
[18:48:31][D][Debug33:254]: Partition data 00: 90,03,00,00,00,00,00,00,00,00,00,00,00
[18:48:31][D][text_sensor:015]: 'dscalarm Partition 1 Msg ': Sending state '03: Zones open'
[18:52:27][D][Debug22:247]: Panel command data: A5,00,20,32,D2,D0,00,00,99,00
[18:52:27][D][Debug33:254]: Partition data 00: 90,03,00,00,00,00,00,00,00,00,00,00,00

Config:
#for documentation see project at https://github.com/Dilbert66/esphome-dsckeybus
substitutions:
  systemName: "dscalarm"
  accessCode: "1595"
  cmdWaitTime: "0" # milliseconds.  set how long to delay before accepting a new 05 cmd as valid to filter out intermittent short duration bogus commands that some panels send. default = 0

esphome:
  name: $systemName
  platform: ESP8266
  board: d1_mini
  
  
  includes:
    - dscKeybusInterface/

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: ".iot.locallan.digitalmetaverse.com"
  manual_ip:
    static_ip: 192.168.100.58
    gateway: 192.168.100.2
    subnet: 255.255.255.0
logger:
  baud_rate: 0
  level: debug

api:
   password: !secret api_password  
      
   
ota:
   password: !secret ota_password
   safe_mode: True
   
status_led:
  pin:
    number: D4
    inverted: yes

custom_component:
- lambda: |-
    auto DSCkeybus = new DSCkeybushome();
    
    DSCkeybus->accessCode="$accessCode";
    DSCkeybus->cmdWaitTime=$cmdWaitTime;
    DSCkeybus->debug=2; // 0 = off, 1 = status change data, 2 = + 05/27 packets, 3 = + all packets received
    
    DSCkeybus->onSystemStatusChange([&](std::string statusCode) {
       id(system_status).publish_state(statusCode);
    });
    DSCkeybus->onPartitionStatusChange([&](uint8_t partition,std::string statusCode) {
       switch(partition) {
         case 1: id(p1).publish_state(statusCode); break;
         case 2: id(p2).publish_state(statusCode); break;
      }
    });
    DSCkeybus->onPartitionMsgChange([&](uint8_t partition,std::string msg) {
       switch(partition) {
         case 1: id(m1).publish_state(msg); break;
         case 2: id(m2).publish_state(msg); break;
      }
    });
    DSCkeybus->onTroubleStatusChange([&](troubleStatus ts,bool open) {
        switch(ts) {
            case trStatus: id(tr).publish_state(open);break;
            case batStatus: id(bat).publish_state(open);break;
            case acStatus: id(ac).publish_state(open);break;
            case panicStatus: id(panic).publish_state(open);break;
            default: break;
        }
    });
    DSCkeybus->onFireStatusChange([&](uint8_t partition, bool open) {
      switch (partition) {
          case 1: id(f1).publish_state(open); break;
      }
    });
    DSCkeybus->onZoneStatusChange([&](uint8_t zone, bool open) {
      switch (zone) {
        case 9: id(z9).publish_state(open); break;
        case 10: id(z10).publish_state(open); break;
        case 11: id(z11).publish_state(open); break;
        case 12: id(z12).publish_state(open); break;
        case 13: id(z13).publish_state(open); break;
      }
    });
    DSCkeybus->onZoneAlarmChange([&](uint8_t zone, bool open) {
      switch (zone) {
        case 9: id(za9).publish_state(open); break;
        case 10: id(za10).publish_state(open); break;
        case 11: id(za11).publish_state(open); break;
        case 12: id(za12).publish_state(open); break;
        case 13: id(za13).publish_state(open); break;
      }
    });
    return {DSCkeybus};

binary_sensor:
    #zone status
  - platform: template
    id: z9
    name: "$systemName Garage Entry"
    device_class: door
  - platform: template
    id: z10
    name: "$systemName Main Entrance"
    device_class: door
  - platform: template
    id: z11
    name: "$systemName South West Deck"
    device_class: door
  - platform: template
    id: z12
    name: "$systemName South Patio"
    device_class: door
  - platform: template
    id: z13
    name: "$systemName Living Room Motion"
    device_class: motion
    
    #zone alarm status
    #device_class: safety
  - platform: template
    id: za9
    name: "$systemName Garage Entry alarm"
    #device_class: safety
  - platform: template
    id: za10
    name: "$systemName Main Entrance alarm"
    #device_class: safety
  - platform: template
    id: za11
    name: "$systemName South West Deck alarm"
    #device_class: safety
  - platform: template
    id: za12
    name: "$systemName South Patio alarm"
    #device_class: safety
  - platform: template
    id: za13
    name: "$systemName Living Room Motion alarm"
    #device_class: safety
  
  - platform: template
    id: tr
    name: "$systemName Trouble Status"
    device_class: problem
  - platform: template
    id: bat
    name: "$systemName Battery Status"
    device_class: problem

  - platform: template
    id: ac
    name: "$systemName AC Status"
    device_class: plug

  - platform: template
    id: panic
    name: "$systemName Panic Status"
    device_class: safety
    
  - platform: template
    id: f1
    device_class: safety
    name: "$systemName Fire Status"
  
text_sensor:
  - platform: template
    id: system_status
    name: "$systemName System Status"
    icon: "mdi:shield"
  - platform: template
    id: p1
    name: "$systemName Partition 1 Status "
    icon: "mdi:shield"
  - platform: template
    id: p2
    name: "$systemName Partition 2 Status "
    icon: "mdi:shield"  
  - platform: template
    id: m1
    name: "$systemName Partition 1 Msg "
    icon: "mdi:alert-box"
  - platform: template
    id: m2
    name: "$systemName Partition 2 Msg "
    icon: "mdi:alert-box"
     
switch:
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();
  - platform: restart
    id: restart_switch

Dilbert66, 2020-12-24:
Sorry, I have no idea.  Does this false trigger still occur with the esp board disconnected from the system? I can't see how the esp would cause this but anything is possible.  Is zone 14 enabled in your system at all?

thunderwood19, 2020-12-24:
It wasn't before the esp was connected to it, and no, zone 14 isn't used at all. I tried just bypassing it via the keypad, but no go. Definitely wierd.

thunderwood19, 2020-12-24:
Going to replace the alarm system backup battery and see if that resolves it. Apparently dsc 1832's are known to check back up status every 24hrs at the same time frame.

Dilbert66, 2020-12-24:
Are you using the latest dev version of the code? My dev system is a dsc1832 and I havent seen the issue.  Then again, I also don't have a battery on it so I always have a battery warning

thunderwood19, 2020-12-25:
> Are you using the latest dev version of the code? My dev system is a dsc1832 and I havent seen the issue. Then again, I also don't have a battery on it so I always have a battery warning

I do not know, good idea to check when I get home from the holidays on sunday!

thunderwood19, 2021-01-06:
New battery and deletion of zone 14 fixed this issue.

closed 2021-01-06 by thunderwood19