Dilbert66/esphome-dsckeybus #25: Error Connecting

neohusky, 2021-05-21:
I'm trying to use your dev branch to have a play with zone expanders.  After compiling and flashing the board OTA I get the following issue in the logs.

`and "/data/dscalarm/.pioenvs/dscalarm/firmware.elf"
======================== [SUCCESS] Took 120.23 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of dscalarm.local
INFO  -> 192.168.0.124
INFO Uploading /data/dscalarm/.pioenvs/dscalarm/firmware.bin (446096 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from dscalarm.local using esphome API
INFO Connecting to dscalarm.local:6053 (192.168.0.124)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.0.124: timed out). Re-Trying in 1 seconds
INFO Connecting to dscalarm.local:6053 (192.168.0.124)`

Any ideas why I can't connect anymore?  I checked my router and my device is connected to my wifi.

Dilbert66, 2021-05-21:
First thing I suspect is a software bug causing the ESP to reboot.  The only way to know for sure is for you to monitor the serial output (connect the arduino serial monitor)  The logs will tell you what's happening.

speedster-pta, 2021-05-22:
I think I have a very similar issue. My ESP8266 module (nodemcu) was working perfectly since I installed it last week and then all of a sudden (with no changes from my side) it stopped connecting yesterday. I've taken the module down, reflashed and confirmed that it has wifi (router page shows it is connected) but it just keeps flashing a blue LED. HA says it is disconnected and the alarm connection switch is greyed out

speedster-pta, 2021-05-22:
@neohusky what version of ESPHome are you on? All that changed on my side recently was the ESPHome update - I wonder if that might be a factor?

Dilbert66, 2021-05-22:
I've updated to the latest esphome and seeing the same issue. I will advise when I find the problem and correct.

Dilbert66, 2021-05-22:
There definitively is an issue with the version on dev branch and the latest esphome. You can try the code in the "new" branch for nowwhich has all the expander support and works with the latest ESPHome.  I'm currently in the process of adding full LCD panel emulation to that version so it will grow.  It also uses the latest dev version of the dsckeybusinterface .   

I will see if I can identify the culprit code in the dev branch and advise when it is resolved.  Please note that the "new" branch will become the default "dev" branch.

Dilbert66, 2021-05-22:
Ok, I think I found the issue.  Try doing a clean build files from espome home before doing an upload. It will force a full rebuild.
Alternatively, you can delete the compiler created directory "dscalarm" where the build files are created.

[image]

w1tw0lf, 2021-05-22:
Can confirm that clean build and no issue here. Do it everytime new version comes out for esphome.

Dilbert66, 2021-05-22:
Absolutely, that's the right way to do it.  I forget to do it sometimes since i'm always testing and recompiling various code iterations that I get in a hurry to compile.

neohusky, 2021-05-27:
Hi Guys

Thanks all for the feedback and suggestions.  Updating ESPHome to version 1.18.0 and updating the the latest code changes has done the trick. Happy days!  So far so good everything is working nicely.

closed 2021-05-27 by neohusky