Dilbert66/esphome-dsckeybus #27: You need to define a kpd_line1 on android app

HenJou, 2021-05-26:
Hi thanks for great job on the app
I got the the alarm running with the new ha_card on home assistant 2021.5.3 and the windows pc show the alarm-keypad-card with no problem, but on the android home assistant app  i get the following message on the phone.  Any idea where to look?

[image]

Dilbert66, 2021-05-26:
Try to refresh the app. Kill it and reload it. That might happen . I also have to fix that error message.

Dilbert66, 2021-05-26:
> Hi thanks for great job on the app
> I got the the alarm running with the new ha_card on home assistant 2021.5.3 and the windows pc show the alarm-keypad-card with no problem, but on the android home assistant app i get the following message on the phone. Any idea where to look?
> 
> [image]

You have an old version showing in the app since the kpd_line message is not the current version.  Refresh it.

slipx06, 2021-05-26:
You need to log out of the android app and back in.

HenJou, 2021-05-27:
I did close all the apps but also with no luck then i reboot my phone and also with no luck.  I solve the problem with the following.  On the Phone setting go to Apps and click on Home assistant, then click on storage and clear the cache. 
Happiness now

closed 2021-05-27 by HenJou

realthk, 2022-10-09:
@HenJou, 
you probably did the same mistake as myself, installing the js from https://github.com/GalaxyGateway/HA-Cards first, rather than the one here, in the ha_cards folder...
I did, and had the same "You need to define a kpd_line1" message in my desktop browser. 

Realising it is the wrong js, replaced it with the one from here, but the other was also cached by the browser, so I forced the reload with adding ver=2 to the include line of Lovelace, so it is now loaded like this
`/local/alarm-keypad-card.js?ver=2`
and it finally works fine.