Dilbert66/esphome-dsckeybus #31: Feature Request: User name and Set Time

Glazzanz, 2021-07-23:
Hi there

Couple of things for your consideration...  I have been using a serial interface to a DSC panel, using NodeRed I have an interface in to the alarm, however your solution looks 'tidier'

A couple of features I currently have are:

**Set Time/Date**

"010" I set the time of the panel, every hour to the LOCAL time, which is sync'd with HA.  This ensures the panels always display the correct time and date, including DST changes.

**User Management**
I keep an array of alarm users (number) and names - so I can output to HA who disabled the alarm...

Would either or these be possible within your app?

Cheers

motolav, 2021-08-01:
On the question of User Management, Home Assistant should track when someone uses the disarm entity that this config creates so all users on your HA will be named if they arm or disarm the system.

Seamus-Mccartin, 2021-08-16:
I know the time/date is possible using the send key service. I wrote a automation in home assistant that sets the time every Monday at 10am. 

[time.txt](https://github.com/Dilbert66/esphome-dsckeybus/files/6989851/time.txt)

Dilbert66, 2022-05-19:
FYI, i've added new features in testing to the "new" branch.  A couple of them is event publishing. This includes arming and disarming with which access codes.  You can have HA capture the event text_field and extract the user code used and then use a lookup table in HA.  If you are asking about having a lookup table in the esphome.yaml, I suppose that's doable but anytime you change or add a user, you would need to modify the yaml and recompile/upload.

Also available is a set_panel_time() function.  To use this , all you need to call the command.  The reason this works is that my code uses the esphome time component which automatically syncs with time servers.   Calling this command pushes that time to the panel from esphome.    There are a slew of other changes I've added to the code base such as multi partition support, virtual LCD keypad emulation, etc.  It's still in alpha mode so still in experimentation.

I still need to document all the changes.

Glazzanz, 2022-05-22:
All appears to be working well.

I did try and add this in the yaml file:
```

interval:
  - interval: 3600s
    then:
      - lambda: |-
          set_panel_time();
```
however, it failed with the following...

```
/config/esphome/alarm.yaml:396:22: error: 'set_panel_time' was not declared in this scope
           set_panel_time();
                      ^
*** [/data/alarm/.pioenvs/alarm/src/main.cpp.o] Error 1
```

Dilbert66, 2022-05-22:
The service is designed to be run from homeassistant. Not from the yaml

Dilbert66, 2022-05-23:
But, having said that, I never needed to do it so I messed around and found a solution  for it.  It normally involves a complex constructor but  I've added a macro at the top of dscalarm.h to simplify the call.  I also added an ID to the custom_component so we can reference it.   To do the call you need to change your function to this:
```
interval:
  - interval: 3600s
    then:
      - lambda: |-
          get_dsckeybus(id(dsckeybus))->set_panel_time();  
```

Have a look at the updated dscalarm.yaml for the needed changes.

Glazzanz, 2022-05-23:
Awesome, thank you.

I guess my thinking here is it's something I just want to happen on  a daily basis, so if I can skip the HA part, it made sense to me.

Thanks for making it happen for me.

closed 2022-09-16 by Dilbert66