Dilbert66/esphome-dsckeybus #37: Build fails on compiling

heytcass, 2021-09-30:
Hello,

I'm trying to get this going on my device, and while the YAML checks out, every time I try to build through ESPHome I get the following errors:

```
Processing dscalarm (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <noise-c> 0.1.1
Compiling /data/dscalarm/.pioenvs/dscalarm/src/main.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
In file included from src/main.cpp:52:0:
src/dscAlarm.h:8:1: error: 'dscKeybusInterface' does not name a type
 dscKeybusInterface dsc(dscClockPin, dscReadPin, dscWritePin);
 ^
src/dscAlarm.h: In function 'void disconnectKeybus()':
src/dscAlarm.h:12:3: error: 'dsc' was not declared in this scope
   dsc.stop();
   ^
In file included from src/main.cpp:52:0:
src/dscAlarm.h: At global scope:
src/dscAlarm.h:63:18: error: 'dscPartitions' was not declared in this scope
  byte lastStatus[dscPartitions];
                  ^
src/dscAlarm.h: In member function 'virtual void DSCkeybushome::setup()':
src/dscAlarm.h:77:2: error: 'dsc' was not declared in this scope
  dsc.resetStatus();
  ^
src/dscAlarm.h: In member function 'void DSCkeybushome::alarm_keypress(std::string)':
src/dscAlarm.h:121:5: error: 'dsc' was not declared in this scope
     dsc.write(keys);
     ^
src/dscAlarm.h: In member function 'void DSCkeybushome::set_alarm_state(int, std::string, std::string)':
src/dscAlarm.h:139:37: error: 'dsc' was not declared in this scope
     if (state.compare("S") == 0 && !dsc.armed[partition] && !dsc.exitDelay[partition]) {
                                     ^
src/dscAlarm.h: In member function 'virtual void DSCkeybushome::loop()':
src/dscAlarm.h:179:27: error: 'dsc' was not declared in this scope
  if (!forceDisconnect  && dsc.loop())  { 
                           ^
In file included from src/main.cpp:52:0:
src/dscAlarm.h:187:10: error: 'dsc' was not declared in this scope
     if ( dsc.statusChanged ) {   // Processes data only when a valid Keybus command has been read
          ^
src/dscAlarm.h:235:40: error: 'dscPartitions' was not declared in this scope
   for (byte partition = 0; partition < dscPartitions; partition++) {
                                        ^
src/dscAlarm.h:241:8: error: 'lastStatus' was not declared in this scope
    if (lastStatus[partition] != dsc.status[partition]  ) {
        ^
src/dscAlarm.h:298:41: error: 'dscZones' was not declared in this scope
    for (byte zoneGroup = 0; zoneGroup < dscZones; zoneGroup++) {
                                         ^
src/dscAlarm.h:319:41: error: 'dscZones' was not declared in this scope
    for (byte zoneGroup = 0; zoneGroup < dscZones; zoneGroup++) {
                                         ^
/config/esphome/dsc-alarm-interface.yaml: In lambda function:
/config/esphome/dsc-alarm-interface.yaml:252:14: error: 'dsc' was not declared in this scope
       return dsc.keybusConnected;
              ^
/config/esphome/dsc-alarm-interface.yaml: In lambda function:
/config/esphome/dsc-alarm-interface.yaml:44:18: error: 'class DSCkeybushome' has no member named 'cmdWaitTime'
     DSCkeybus->cmdWaitTime=$cmdWaitTime;
                  ^
/config/esphome/dsc-alarm-interface.yaml:62:44: error: 'troubleStatus' has not been declared
     DSCkeybus->onTroubleStatusChange([&](troubleStatus ts,bool open) {
                                            ^
/config/esphome/dsc-alarm-interface.yaml: In lambda function:
/config/esphome/dsc-alarm-interface.yaml:64:20: error: 'trStatus' was not declared in this scope
             case trStatus: id(tr).publish_state(open);break;
                    ^
/config/esphome/dsc-alarm-interface.yaml:65:20: error: 'batStatus' was not declared in this scope
             case batStatus: id(bat).publish_state(open);break;
                    ^
/config/esphome/dsc-alarm-interface.yaml:66:20: error: 'acStatus' was not declared in this scope
             case acStatus: id(ac).publish_state(open);break;
                    ^
/config/esphome/dsc-alarm-interface.yaml:67:20: error: 'panicStatus' was not declared in this scope
             case panicStatus: id(panic).publish_state(open);break;
                    ^
/config/esphome/dsc-alarm-interface.yaml: In lambda function:
/config/esphome/dsc-alarm-interface.yaml:70:8: error: no matching function for call to 'DSCkeybushome::onTroubleStatusChange(setup()::__lambda15::__lambda19)'
     });
        ^
/config/esphome/dsc-alarm-interface.yaml:70:8: note: candidate is:
In file included from src/main.cpp:52:0:
src/dscAlarm.h:53:8: note: void DSCkeybushome::onTroubleStatusChange(std::function<void(bool)>)
   void onTroubleStatusChange(std::function<void (bool isOpen)> callback) { troubleStatusChangeCallback = callback; }
        ^
src/dscAlarm.h:53:8: note:   no known conversion for argument 1 from 'setup()::__lambda15::__lambda19' to 'std::function<void(bool)>'
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/WiFiServer.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling /data/dscalarm/.pioenvs/dscalarm/lib4d9/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
/config/esphome/dsc-alarm-interface.yaml: In lambda function:
/config/esphome/dsc-alarm-interface.yaml:253:3: warning: control reaches end of non-void function [-Wreturn-type]
     icon: "mdi:shield-link-variant"
   ^
*** [/data/dscalarm/.pioenvs/dscalarm/src/main.cpp.o] Error 1
========================== [FAILED] Took 7.68 seconds ==========================
```
I copied the YAML exactly to my ESPHome file, and placed dscAlarm.h in the dscKeybusInterface directory in the esphome parent directory, as said in the documentation.

Dilbert66, 2021-09-30:
you are missing the dsckeybus*.cpp files from the dsckeybusinterface directory

heytcass, 2021-10-01:
Ah ha. I see. I will have to issue a pull request at the dscKeybusInterface repo, they only have the dscAlarm.h file in the directory, which also seems to be out of date. Thanks!

closed 2021-10-01 by heytcass

Dilbert66, 2021-10-01:
No, don't do that.  The dsckeybusinterface repo has their own directory structure for the main library and the examples only hold the extra files.  It's assumed that you will copy the library to the correct location.  On my repo, I have everything together since I structure it to suit my project.