Dilbert66/esphome-dsckeybus #43: Latest new branch causes module to go offline

incogizmo, 2022-03-14:
I just updated to the latest new branch and found I lost with the esp module immediately after flashing. (I am using the D1 Mini SMT pcb) I made no config changes to my esphome yaml, just git pulled and flashed.

Did some further troubleshooting. Reset (powered down and up) the alarm and the system came online and could see status of the alarm, but using the lovelace keypad I could not interact with the alarm.

As soon as I alarmed and unarmed the alarm to clear the date/time fault from the alarm keypad the esphome entities status switched to unknown and I could no longer view the esphome logs or ota. Ping remained working for ~10 mins.
I reflashed just incase but no success.

I have since reverted back to commit a65bfc51724b87968623bc27652623035ed9fa56 and all functionality has been restored.

Thanks for all your work on this project btw Ive found it incredibly useful to see the status and get notifications of the alarm status remotely and use room motion sensors for other automations in HA. Let me know if there is anything I can try and any assistance I can provide to help troubleshoot if you need.

Dilbert66, 2022-03-14:
Thanks for the input.  I suspect this is related to my removal of the Icache_ram_attr use for the esp8266 since on newer version of the esp compiler this option is deprecated.  Unfortunately this will break older compiler versions and I did not add a version check on my code.  I've reverted the option back so hopefully that will cure the issue you had.  You should not need to update since there were no real bugfixes in the last few commits.  It was mostly adding defines to allow removing all expander related code for those that don't have enough iram available on the 8266 and don't need the virtual expander functionality.

incogizmo, 2022-05-09:
Just an update on this after your fix I rebuilt and uploaded and there was no issue.
I will now close this issue.

closed 2022-05-09 by incogizmo