Dilbert66/esphome-dsckeybus #46: Integration fails to update suddenly

Waylonf, 2022-05-19:
I am having a challenge getting my alarm interface to work again. Everything was working 100% for a while and then suddenly stopped working or updating any sensors in Home Assistant. 

At the time the issue first occurred I had made no changes to the firmware on my board, but there may have been a Home Assistant update that has caused this.

Have there been any broken changes in HA that could affect the DSC integration? If I could get confirmation on this it would allow me to  focus my efforts on other possible causes.

Any insights will be greatly appreciated and should I need to provide logs or any other info that may help shed light on my issue, please let me know. I relied quite a bit on this integration, any insights at all would be amazing :-)

Dilbert66, 2022-05-19:
I have not experienced or heard of these issues but then again I'm not running the latest HA version. I do use the latest epshome 2022.5.0. HA does have issues sometimes and there are two things can you do. Go to Configuration,Devices & Services. Find your device in the ESPHome integration, click on the three vertical dots to open a menu. ther eyou can try first to do a Reload. If that does not work, then the last option is to Delete it and then do a "+ Add integration" to add it again.
Before you do that though, have a look at your ESPHome logs and ensure that you are seeing traffic and updates to the sensors.  If you dont' see that then the issue is with your device and not HA.  If your device is acting up, try resetting it.

Glazzanz, 2022-05-21:
@Dilbert66 what board do you recommend to go with the PCBs that are already designed.  All the D1 Mini Pros that I can find are 8266 powered...  

I currently have a heap of the NodeMCU boards....  but the pin out doesnt work with an esp32 dev board :(

Dilbert66, 2022-05-21:
I would recommend a d1 mini pro esp32. It's basically an esp32 that can fit in the same footprint as a d1 mini so works fine. You just use the inside rows of pins.
https://www.aliexpress.com/item/32858054775.html?spm=a2g0o.productlist.0.0.5d5f283dBskdXk&algo_pvid=f077df51-490c-4a83-a3ff-3ef3c48153a1&algo_exp_id=f077df51-490c-4a83-a3ff-3ef3c48153a1-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000021656141438%22%7D&pdp_npi=2%40dis%21CAD%21%214.96%21%21%211.36%21%21%40210318bb16531016307901961e0e6f%2112000021656141438%21sea

Glazzanz, 2022-05-21:
Thanks - having a look at that, it looks like I would end up using pin IO22, IO21 and IO5, if installing on this board?

https://github.com/Dilbert66/esphome-dsckeybus/tree/master/PCB%20Layouts/WemosD1%20Mini/SMD

No issues with that?

Dilbert66, 2022-05-21:
No issues at all. I have one like that. Just set dscWritePin to 5 , dsClockpin to 22 and dscReadPin to 21.

closed 2022-09-16 by Dilbert66