Dilbert66/esphome-dsckeybus #48: Wemos D1 R2 with THT board - no data yet

jshridha, 2022-05-24:
I built the THT version of the PCB with a wemos d1 R2 and so far I am not receiving any data using the latest commit on the new branch. I get all 0's for the panelData output and am seeing that the alarm panel is listed as "offline". Do you have any suggestions on where I can start troubleshooting? I have checked the board connections and they seem okay, but I should probably check them again. Are there any configuration changes that I need to make in the yaml file for this specific mcu/pcb combination? Below are the logs that I'm seeing so far. Any help would be greatly appreciated.

```
[08:57:25][D][text_sensor:067]: 'dscalarm event': Sending state ''
[08:57:33][D][text_sensor:067]: 'dscalarm event': Sending state ''
[08:57:38][D][text_sensor:067]: 'dscalarm beeps': Sending state '0'
[08:57:40][D][text_sensor:067]: 'dscalarm System Status': Sending state 'offline'
[08:57:41][D][text_sensor:067]: 'dscalarm partition 2 beeps': Sending state '0'
[08:57:44][D][text_sensor:067]: 'dscalarm Trouble Msg ': Sending state 'OK'
[08:57:44][D][text_sensor:067]: 'dscalarm zone status ': Sending state 'OK'
[08:57:55][I][Paneldata: :898]: 00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[08:57:55][D][text_sensor:067]: 'dscalarm System Status': Sending state 'offline'
[08:57:55][D][text_sensor:067]: 'dscalarm zone status ': Sending state 'OK'
[08:57:55][D][text_sensor:067]: 'dscalarm Trouble Msg ': Sending state 'OK'
[08:57:55][D][text_sensor:067]: 'dscalarm event': Sending state ''
```

Dilbert66, 2022-05-24:
Well, hard to say. Obviously you have a wiring or configuration issue somewhere.  The default pins for the esp8266 should work for that board.   Did you uncomment the correct pins for the esp8266 in the yaml? Should be 5, 4 and 15

jshridha, 2022-05-24:
Ok yeah I'll check the wiring again. It's certain possible I messed up something. Yes I commented out those lines so it should be using the esp8266 pin config. I'm new to the DSC panel. Is there any configuration that needs to be done on the panel side? I assume not since I believe this project is just reading the same output that is being sent to the tablet control panels.

Dilbert66, 2022-05-24:
Which DSC panel are you using? This will not work with the DSC NEO versions.

jshridha, 2022-05-24:
I have the hs2032 which is a dsc neo panel. I should have read the fine print better. Thanks for the help.

closed 2022-05-24 by jshridha