Dilbert66/esphome-dsckeybus #56: WT32-ETH01 IO pins?

Rickerdo, 2022-09-15:
I have a WT32-ETH01 running the new branch. The logs seems fine and HA can connect to the module, but I cannot figure out which pins should be used for the keybus. The DscAlarm.yaml file suggests:
```
  #ESP32 Pins
  dscClockPin: "22"
  dscReadPin: "21"
  dscWritePin: "18"
```
But it seems the WT32-ETH01 uses GPIOxx naming which doesn't seem to directly map. Any suggestions? The datasheets I found are not in English.

ESP Home page for the WT32-ETH01 - https://esphome.io/components/ethernet.html?highlight=wt32+eth01#configuration-for-wireless-tag-wt32-eth01

Dilbert66, 2022-09-15:
You will need to use a different pin for the dscwritepin instead of 18 as that is taken by the wt32-eth01 for the mdio_pin.    You can use gpio5 or gpio19 instead of gpio18 as it defaults to.

Dilbert66, 2022-09-15:
If you are using a long range radio (IP or GSM) interface  and monitored by a central monitoring station , then set it to false.  If you don't have this interface, then set it to true and the firmware will emulate one.  It enables extra status messages sent from the panel to be captured by the firmware for display.

Rickerdo, 2022-09-15:
I now have an Ethernet connected WT32-ETH01 connected to my DSC PC1864 through ESP Home. Here's how I "made it work":
```
  #ESP32 Pins
  # IO39 - input only
  # IO36 - input only
  # IO35 - input only
  dscClockPin: "15"
  dscReadPin: "14"
  dscWritePin: "12"
```
I translated the datasheet from Chinese to English and found out that pins 35, 36, and 39 are input only. I suppose dscReadPin could have been mapped to one of those pins, but hey, it works as listed above.

Also, it seems the pins within code are as labeled on the PCB. In other words, pin 15 within the YAML file is actually IO15 on the PCB. Go figure.

For posterity's sake:
 - The WT32-ETH01 that I purchased for $17.99 - https://www.amazon.com/dp/B0B7JGXYLZ
 - A CH340G based USB to TTL converter was recommended in the below post. This is the one that I purchased for $6.99 - https://www.amazon.com/dp/B07DJ3ZQM4
 - How to wire the TTL to USB converter to the WT32-ETH01 - https://community.home-assistant.io/t/how-i-installed-esphome-on-the-wt32-eth01/359027
 - If using the CH340G converter on Linux Mint (possibly other distros as well?), disable brltty (braille USB device) per - https://github.com/juliagoda/CH341SER/issues/18 If this step is skipped, the /dev/ttyUSB{x} serial port will immediately be stolen and converted to a brltty device. Rather frustrating, but easy to fix.

I suppose a better (cleaner?) write up is in order, but I wanted to do a brain dump before I forgot the details.

closed 2022-09-15 by Rickerdo

zsamiatt, 2022-10-20:
Hello, Where can I find esp32 Pins? I can not see in yaml file.

twistedsanity, 2024-11-28:
FYI: using the WT32-ETH01 pcb layout supplied in this repo I needed to use the following pins to work with the pcb:
  dscclockpin: "15"
  dscreadpin: "14"
  dscwritepin: "4"