Dilbert66/esphome-dsckeybus #60: Arm / Disarm Buttons not working

PipeDeveloper, 2022-09-22:
Hello @Dilbert66 

Im having some issues with the new branch using ESP32 and isolated version.
The interaction works like a charm, however using the template card for basic service like a button to arm/disarm they are not working.

I have copied the template directly on my Home Assistant config.yaml but after reboot i have noted that those entities doesnt appears on the list. I have manually set the entities and buttons seems not working (also there is no log messages on the ESPhome console after pressing the button, Arm and Dissarm). I leave you some screenshots

[image]

[image]

[image]

[image]

Before the update, those buttons works perfectly, also when you press the button disarm, the keypad beeps, but now nothing happens.

Dilbert66, 2022-09-22:
Since the "new" version adds partition support so you need to add "partition" as an additional parameter so the system will know what partition to disarm.

Dilbert66, 2022-09-22:
These are the registered services and parameters: 
```
      register_service( & vistaECPHome::alarm_keypress, "alarm_keypress", {
        "keys"
      });
      register_service( & vistaECPHome::alarm_keypress_partition, "alarm_keypress_partition", {
        "keys","partition"
      });      
     // register_service( & vistaECPHome::set_keypad_address, "set_keypad_address", {
      //  "addr"
     // });
      register_service( & vistaECPHome::alarm_disarm, "alarm_disarm", {
        "code","partition"
      });
      register_service( & vistaECPHome::alarm_arm_home, "alarm_arm_home",{"partition"});
      register_service( & vistaECPHome::alarm_arm_night, "alarm_arm_night",{"partition"});
      register_service( & vistaECPHome::alarm_arm_away, "alarm_arm_away",{"partition"});
      register_service( & vistaECPHome::alarm_trigger_panic, "alarm_trigger_panic", {
        "code","partition"
      });
      register_service( & vistaECPHome::alarm_trigger_fire, "alarm_trigger_fire", {
        "code","partition"
      });
      register_service( & vistaECPHome::set_zone_fault, "set_zone_fault", {
        "zone",
        "fault"
      });
```

PipeDeveloper, 2022-09-22:
> Since the "new" version adds partition support you now need to add "partition" as an additional parameter so the system will know what partition to disarm. Same thing for the "arm" services.

I use patition 1 as default (only one partition), however the template to call the register service doesnt appears as an valid entity. Maybe a HA update?
Or maybe some steps to follow for some people without experience using ESPhome.

Dilbert66, 2022-09-22:
I have not see the issue on my end but when I do changes, I delete the alarm integration and re-add it. 
alternatively, you can go to the esphome integrations config and do a reload and see what happens:
[image]

PipeDeveloper, 2022-09-22:
Nothing Happen. All the same. I have deleted the integration, same as you 56 entities, 1 device.
The entitie "disarm" if i search on all entities on my browser doesnt appear but it is added on the HA config.yaml

Dilbert66, 2022-09-22:
Well, I'm not sure what to say. I just deleted and re-added my test system and all works fine.  I assume you are using the "dev" branch correct?

PipeDeveloper, 2022-09-22:
No, im using the "new" branch

Dilbert66, 2022-09-22:
Whups I was confusing it with the vistaecp  project.   let me test

PipeDeveloper, 2022-09-22:
Can you leave an example of it? i really apreciate. Also update de readme.md if correspond

Dilbert66, 2022-09-23:
Well, I pushed a later version. Not much has changed except that I'm adding the option to allow using  the esphome mqtt api for those that want to use it.  Otherwise it should be the same. Anyhow, give the last "new" branch a try.  Otherwise , i'm not sure what the issue you are having is.   Your config looks fine as is.  You don't need to add the partition parameter as I indicated was for another system.

Dilbert66, 2022-09-23:
As to using the set_keypress_command. The readme already has an example. Here's an excerpt:
```
"alarm_keypress", Parameter: "keys" where keys can be any sequence of cmds from the list below. For example to arm partition 2 in night mode you set keys to be "/2Nxxxx" where xxxx is your access code. Alternatively you can use the * code for the same function and send /2*9xxxx.
```

Dilbert66, 2022-09-23:
Ok, I see the issue with the disarm now. I will correct it.

PipeDeveloper, 2022-09-23:
Actually i tried to manually call some services but nothing happens. Also i check the console log in the ESPhome web UI and after sending the command and i dont see any message.

[image]

I discard a hardware issue, double check connection with multimeter and visual inspection (also is a PCB well builded)

[image]

Dilbert66, 2022-09-23:
Are you sure you have the correct write pin set in the configuration of the yaml?

PipeDeveloper, 2022-09-23:
Yep, pretty sure.

[image]

[image]

[image]

Dilbert66, 2022-09-23:
Your circuit board looks great!  Interesting issue. I really don't see what else it could be though since the service is recognized by Homeassistant.   Is the "defaultPartition" set to 1 also in your yaml?  

I'll keep looking but I just can't duplicate the issue you are having.

PipeDeveloper, 2022-09-23:
This is my full yaml

```
#for documentation see project at https://github.com/Dilbert66/esphome-dsckeybus

#you can enter a list of user codes mapped to their names for display in armed/disarmed event messages
globals: 
  - id: userCodes
    type: std::string
    restore_value: no
    initial_value: '"1:John,2:Sally,3:Bob"' 
    
substitutions:
  # change the name to suit your needs. This is what esphome will use as the friendly name for your component.
  # also used as a prefix for all fields
  systemName: "alarma" 
  
  #Only comes into effect if a password prompt occurs when arming eg. night mode
  accessCode: "My code" #Edited
  
  #used to select the default partition associated with the alarm panel messages
  defaultPartition: "1" 

  #zone expander addresses:
  # 9  - zones 9-16
  # 10 - zones 17-24
  # 11 - zones 25-32
  # 12 - zones 33-40 (for systems with 64 zone support)
  # 13 - zones 41-48 (for systems with 64 zone support)
  # 14 - zones 49-56 (for systems with 64 zone support)
  # 16 - zones 57-64 (for systems with 64 zone support)  
  expanderAddr1: "0" # 1st zone expander emulator address to use . Set to 0 to disable. 
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 
  
  #ESP32 Pins
  dscClockPin: "22"
  dscReadPin: "21"
  dscWritePin: "18"
  
  maxZones: "32" # maximum amount of zones your system supports
  
esphome:
  name: $systemName
  platform: ESP32
  board: nodemcu-32s

  includes:
    # subdirectory path where custom component *.h and *.cpp files are located
    - dscKeybusInterface/

wifi:
  ssid: "" #Edited
  password: "" #Edited

  ap:
    ssid: "$systemName"
    password: "bmWBpRMsKr5g"

logger:
  baud_rate: 115200
  level: DEBUG

##API communications - ensure the mqtt: section is commented if using this protocol
api:
   password: My Code #Edited
   #encryption:         
     #key: !secret encryption_key
     
##MQTT communications - ensure the api: section is commented if using this protocol
#mqtt:
[... 458 lines cut ...]
    name: "$systemName partition 2 beeps"
    icon: "mdi:alert-box"   
    
  - platform: template
    id: tr_msg
    name: "$systemName Trouble Msg " # uncomment to show in home assistant
    icon: "mdi:alert-box"
        
switch:
    #shows status of connection status to panel.  You can disconnect before upload using the switch.
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();
  - platform: restart
    id: restart_switch

```

Dilbert66, 2022-09-23:
The only thing that would affect your writes is the dscwritepin and the defaultpartion really.  Otherwise, I would have said hardware but in this case, I'm at a loss to the actual reason. I'm using an esp32 for my main dev platform and it's fine here.

PipeDeveloper, 2022-09-24:
I could use the dev branch in stead, but i have read the yaml and there is no line to define write pin.
Im also using ESP32

Dilbert66, 2022-09-24:
The dev is older. On that one the config is done in dscalarm.h with defines.

Dilbert66, 2022-09-25:
I've added some debug messages to the "new" branch for the write commands to see if they are indeed being processed by the esp.  It should help pinpoint the issue.

PipeDeveloper, 2022-09-25:
Well the console doesn't lie

[image]

Something related with the circuit diagram then (?).
I detect is some diferences in the resistor values on the new branch (1K and 220ohm) and the main branch (4.7k and 180ohm)

Im using 4.7k on the read pins and 180ohm on the write pin. I dont know if that can affect. But actually read pins works perfect

Dilbert66, 2022-09-25:
I used smaller resisters to allow for larger current to drive the leds in order to have a better ouptput pulse.  If 4.7k works with your panel for the read lines then stick with that.  I went with a smaller resistor since I was not getting a consistent read at all times due to the output pulse being too small in amplitude.  Same goes with the write pin. I went smaller to allow a better output.  It all depends on the optocoupler chips you are using. Same have better gain then others so won't require as large a resistor.

PipeDeveloper, 2022-09-25:
On that im using 4N35 (SMD version) [Datasheet](https://pdf1.alldatasheet.com/datasheet-pdf/view/2848/MOTOROLA/4N35.html)

I also leave the schematics if you can detect something weird.
[image]

Dilbert66, 2022-09-25:
Your schematic looks fine to me.  Can you try using the alarm_keypress service and try to arm that way as well. I want to see what the logs say for that one.

Dilbert66, 2022-09-25:
I see in the log above that your panel had an open zone so of course it won't arm.  Can you bypass the zone and so your panel goes ready and try arming again?

PipeDeveloper, 2022-09-26:
All zones are closed, i tried via buttons, the console seems to works but physically the keypad beeps when you arm or disarm (even if not armed, disarm buttom always beeps the keypad)

[image]

Tried with the alarm_keypress_service and some commands
[image]

Dilbert66, 2022-09-26:
It definitively looks like the command is not being sent to the panel at all since I don't see any responses from the panel or following 05 moduledata command.   I've made a small update on the code to test something in the "new" branch.  Try it and see what it does

PipeDeveloper, 2022-09-28:
Sorry for the late response.
With all zones closed (ready to arm) I have test some commands.

[image]

[image]

PipeDeveloper, 2022-09-28:
I did some test. The service emulates the remote controller, i compared some log messages from the pannel when i press the controller vs when i press the buttons via HA

```
[18:15:32][I][Paneldata: :976]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[18:15:32][I][Moduledata::976]: 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 
[18:15:43][D][debug:913]: Setting Alarm state: A to partition 1
[18:15:43][D][debug:929]: Arming away

*Here i press the button on HA, nothing happens*

[18:15:47][I][Paneldata: :976]: E6: E6 00 2C 08 00 00 00 00 1A 00 00 00 00 00 00 00 
[18:15:56][D][text_sensor:067]: 'alarma event': Sending state ''
[18:15:57][D][text_sensor:067]: 'alarma Partition 1 Msg ': Sending state '01: Ready'
[18:15:59][D][text_sensor:067]: 'alarma Trouble Msg ': Sending state ''
[18:16:01][D][text_sensor:067]: 'alarma beeps': Sending state '0'
[18:16:02][D][text_sensor:067]: 'alarma zone status ': Sending state 'OK'
[18:16:02][I][Paneldata: :976]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[18:16:02][I][Moduledata::976]: 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 
[18:16:07][D][text_sensor:067]: 'alarma line1': Sending state 'System is Ready'
[18:16:07][I][Moduledata::976]: 05: FF 01 B1 FF FF FF FF FF FF FF 01 00 00 00 00 00 
[18:16:07][I][Paneldata: :976]: 64: 64 00 06 6A 00 00 00 00 00 00 00 00 00 00 00 00 

*Here i press the remote controller, It started to beep the keypad*

[18:16:07][D][text_sensor:067]: 'alarma beeps': Sending state '3'
[18:16:07][D][info:1778]: status 01, last status 01,line2status 70,selection 01,partition=1,skip=0
[18:16:07][D][text_sensor:067]: 'alarma line1': Sending state 'System is Ready'
[18:16:07][D][text_sensor:067]: 'alarma line2': Sending state 'Ready to Arm <>'
[18:16:07][I][Paneldata: :976]: 5D: 5D 00 00 00 00 00 00 5D 00 00 00 00 00 00 00 00 
[18:16:07][I][Paneldata: :976]: E6: E6 00 18 01 00 00 00 00 00 FF 00 00 00 00 00 00 
[18:16:07][I][Paneldata: :976]: BB: BB 00 00 00 BB 00 00 00 00 00 00 00 00 00 00 00 
[18:16:07][I][Paneldata: :976]: E6: E6 00 1A 40 00 00 01 00 00 00 41 00 00 00 00 00 
[18:16:07][I][Paneldata: :976]: 75: 75 00 11 86 00 00 00 00 00 00 00 00 00 00 00 00 
[18:16:07][I][Paneldata: :976]: CE: CE 00 01 80 00 00 00 4F 00 00 00 00 00 00 00 00 
[18:16:07][I][Paneldata: :976]: CE: CE 00 40 FF FF FF FF 0A 00 00 00 00 00 00 00 00 
[18:16:08][I][Paneldata: :976]: 05: 05 00 83 08 00 C7 00 C7 00 C7 01 00 00 00 00 00 
[18:16:08][D][info:1778]: status 08, last status 01,line2status 70,selection 01,partition=1,skip=0
[18:16:08][D][text_sensor:067]: 'alarma line1': Sending state 'Exit delay   '
[18:16:08][D][text_sensor:067]: 'alarma line2': Sending state 'in progress     '
[18:16:08][D][text_sensor:067]: 'alarma Partition 1 Msg ': Sending state '08: Exit delay'
[18:16:08][D][text_sensor:067]: 'alarma Partition 1 Status ': Sending state 'pending'
[18:16:09][D][text_sensor:067]: 'alarma beeps': Sending state '0'
[18:16:12][D][text_sensor:067]: 'alarma event': Sending state ''
[18:16:13][D][text_sensor:067]: 'alarma Partition 1 Status ': Sending state 'pending'
[18:16:14][D][text_sensor:067]: 'alarma partition 2 beeps': Sending state '0'
[18:16:17][I][Paneldata: :976]: 3E: 3E 00 83 08 00 C7 00 90 00 00 00 00 00 00 00 00 
[18:16:18][D][text_sensor:067]: 'alarma line2': Sending state 'in progress     '
[18:16:21][I][Paneldata: :976]: 75: 75 00 31 A6 00 00 00 00 00 00 00 00 00 00 00 00 
[18:16:24][D][text_sensor:067]: 'alarma System Status': Sending state 'online'
[18:16:25][I][Paneldata: :976]: 75: 75 00 31 A6 00 00 00 00 00 00 00 00 00 00 00 00 
[18:16:25][D][info:1778]: status 08, last status 08,line2status 70,selection 01,partition=1,skip=0
[18:16:25][D][text_sensor:067]: 'alarma line1': Sending state 'Exit delay   '
[18:16:25][D][text_sensor:067]: 'alarma line2': Sending state 'in progress     '
[18:16:25][D][text_sensor:067]: 'alarma Partition 1 Status ': Sending state 'pending'
[18:16:25][D][text_sensor:067]: 'alarma zone status ': Sending state 'OK'
[18:16:26][D][text_sensor:067]: 'alarma event': Sending state ''
[18:16:32][I][Paneldata: :976]: 27: 27 00 82 05 00 C7 00 75 00 00 00 00 00 00 00 00 
[18:16:32][D][info:1778]: status 08, last status 08,line2status 70,selection 01,partition=1,skip=0
[18:16:32][D][text_sensor:067]: 'alarma line1': Sending state 'Exit delay   '
[18:16:32][D][text_sensor:067]: 'alarma line2': Sending state 'in progress     '
[18:16:32][D][binary_sensor:036]: 'alarma Partition 1 Ready': Sending state OFF
[18:16:32][I][Paneldata: :976]: 16: 16 00 0E 42 F2 58 00 00 00 00 00 00 00 00 00 00 
[18:16:32][I][Paneldata: :976]: A5: A5 00 22 49 F1 4C BF 00 0C 00 00 00 00 00 00 00 
[18:16:32][D][text_sensor:067]: 'alarma event': Sending state '2022.02.15 17:19 P:1 Armed: Special'
[18:16:32][I][Paneldata: :976]: 75: 75 00 00 75 00 00 00 00 00 00 00 00 00 00 00 00 
[18:16:32][I][Paneldata: :976]: CE: CE 00 01 FF FF FF 7F 4B 00 00 00 00 00 00 00 00 
[18:16:32][I][Paneldata: :976]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[18:16:32][I][Moduledata::976]: 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 
[18:16:33][I][Paneldata: :976]: CE: CE 00 40 FF FF FF FF 0A 00 00 00 00 00 00 00 00 
[18:16:33][I][Moduledata::976]: 05: FF 01 FF FF FF FB FF F7 FF FF 01 00 00 00 00 00 
[... 80 lines cut ...]
[18:17:24][I][Paneldata: :976]: 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 
[18:17:24][D][info:1778]: status 01, last status 3E,line2status 70,selection 01,partition=1,skip=0
[... cut ...]
[18:17:52][D][text_sensor:067]: 'alarma event': Sending state ''
[18:17:57][D][text_sensor:067]: 'alarma Partition 1 Msg ': Sending state '01: Ready'
[18:17:59][D][text_sensor:067]: 'alarma Trouble Msg ': Sending state ''
[18:18:01][D][text_sensor:067]: 'alarma beeps': Sending state '0'
[18:18:02][D][text_sensor:067]: 'alarma zone status ': Sending state 'OK'
[18:18:02][I][Paneldata: :976]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[18:18:02][I][Moduledata::976]: 11: FF 01 3F FF FF F3 FF FF FF 03 00 00 00 00 00 00 
```

Dilbert66, 2022-09-28:
Does your board work with the older code from the "master" branch?  Everything points to a hardware issue with the writeline.

PipeDeveloper, 2022-09-28:
My last board works "fine" with the ESP8266, there was too much data on the bus that it cannot handle for 31 zones and thats why i move with the ESP32. That issue was solved but the it cames this.

Something that i noted is that the ESP32 module, on pin 18 it doesnt send any signal.
I measured with a multimeter pin18 and GND, doing 0v all time. I sent some commands if i can see some voltage value (meaning data) but 0v on the screen all time.
Note that i have this module
[image]

Dilbert66, 2022-09-28:
The fact that d18 is not showing activity would be related to a coding issue or a bad pin.  I can't see it being the code since I use the same module you have for testing and it works fine.   I did have a module that had a bad pin before, so used a different write pin.   
To eliminate a code issue, try an old version , or the master branch.   For hardware try a different pin for the dscwrite.  Since your board is hardwired, you will need to use a jumper though to do that.

PipeDeveloper, 2022-09-29:
Done, D18 working, it sends data as expected.
Actually with the alarm_keypress service i see data trafic on D18. But still with the issue, pannel not responding...

PipeDeveloper, 2022-09-29:
Could be something related with the logic? because the read pins do positive voltage pulses to ground (BLK).

For write it cannot do pulses from the ESP32... the schematic is wrong on that case
[image]

Then we should fix the schematic and use red cable to pull up the green line like this
[image]

Dilbert66, 2022-09-29:
The logic is correct. It needs to be driven low to send a pulse.  Try using a 180ohm resistor instead of 220. It's possible the line is not being driven low enough.

PipeDeveloper, 2022-09-29:
Ok, fortunally i have that value, tried with 1K and 220ohm.
Last i can do is to use a jumper (0ohm) but i dont know if i could burn the octocoupler if i attach it directly to the pin.

Dilbert66, 2022-09-29:
Use a another 220 in parallel to the existing 220.  It will give you an equivalent resistance of 110 ohm. Smaller value, means large puls

PipeDeveloper, 2022-09-29:
Working with 180 ohm!!
That was very weird because already i used that value before

[image]

Dilbert66, 2022-09-29:
I found the optocoupler circuit version more problematic than the simple design due to the fact that to get max output gain from the optocoupler you need a high gain optocoupler because we are limited by the currrent drive limits of the esp.  I could have made it better by putting another transistor stage in front of the optocouple r to boost the current but that complicates the circuit.   

Also on the panel side, you have to use a high resistance to drive the led due to the fact that using too low a value,  loads the data lines and impacts communications.  I settled on 1kohm, so that it would not load up too much but still drive the led enough to give a good pulse on the receiver side.  It's a compromise and 1kohm still loads the lines a bit.

Believe it or not, the simple circuit with passive high value resistors and a transistor has the least impact on the data lines but it provides no protection to the esp or panel in case of any voltage spikes so it's all a compromise.

PipeDeveloper, 2022-09-29:
Totally agree, by the way to protect the ESP actually we should include a transistor between data (base), and maybe 5v (collector) directly from the external power supply.
Maybe for this kind of profesional boards we could develop something that we can warranty the security and the circuit protection. But for most of people the simplified version also works fine, the "complicated" version lets leave it just for enthusiast.

closed 2022-09-29 by PipeDeveloper

Dilbert66, 2022-09-29:
I went back and looked at the esp32 specs again for max current and the esp32 pins can handle 40ma instead of the approx 12ma for the esp8266. When I designed that circuit I was using the esp8266 limits.   When using an esp32 you can safely use a 100ohm resitor instead of the 180 for better signal output and still be well within the esp32 max current capability.

PipeDeveloper, 2022-09-30:
Thats a very important point and good to know.
Maybe for some people could be a little confusing wich resistor should use depending on wich ESP they choose to build it. Obviusly ESP32 is highly recomended. But in summary... on the WritePin
for ESP32 use 100ohm
for ESP8266 use 180ohm

For isolated version, On the case of ESP side, pull up resistor could be any value between 1k and 10k, it doesnt matter at all but just in case (experience with arduino). On the DSC green and yellow i dont know if that applies at all, at least 4,7k works fine.

Dilbert66, 2022-09-30:
I will update the schematic soon to reflect the resistor choices depending on the platform.  Actually for the esp32, you could even go down as low as 80ohm for the write led side.   I used 4.7k just to be consistent.  You are correct, it can be any value from 1 to 10k.  On the panel side though, the value was important as far as the led drive, I found 4.7k was the max I could go to have good output pulse but keep bus loading to minimal.   As a pullup, it's not important.

gudinoemiliano, 2024-12-24:
I'm working around a similar issue: all the read functionality works fine, but I can't control my DSC1832 panel from the ESP32.

I tried using 100-ohm and 180-ohm resistors at R1 (D8 pin - WR), but the result is the same. I captured some images with the oscilloscope, and it seems like the ESP32 is writing correctly, but it is not taking control of the DSC bus. All captures were taken at the same time with the same trigger while the panel was sending the "*" button signal.

This is the clock signal, captured from the Yellow BUS cable:
[image]

This is the data signal, captured from the Green BUS cable:
[image]

This is the D8 pin on the ESP32, captured directly from the ESP32 pin:
[image]

You can see that these parts of the signals present on the bus do not match the signal generated by the ESP32:
[image]

My installation contains two (2) physical keyboards working on the bus, along with the ESP32.

Does anyone else have the same issue and know how to solve it? It seems to be more of a software issue rather than a hardware one or some incompatibility with the DSC bus.

Dilbert66, 2024-12-24:
Not software. You have some issue with your wiring or connection.