Dilbert66/esphome-dsckeybus #65: Problem with setting time

chreggy, 2022-10-09:
Hello,

I have a problem with sending time, the month and time are wrong. 

```
time:
 - platform: sntp
   id: sntp_time
   timezone: "Europe/Paris"
   servers:
     - 134.214.100.6
```

`[20:11:22][D][text_sensor:067]: 'dscalarm event': Sending state '2022.00.09 18:11 P:1 Disarmed: Master code:40'`

Is there a problem in my config ?

Thanks.

Dilbert66, 2022-10-09:
No, it's a bug in the panel events date date decoding. I have to  look into it.

Dilbert66, 2022-10-10:
Ok, issue resolved and new version pushed to branch "new"

PipeDeveloper, 2022-11-02:
I noted it like a few hours ago on the DLS5. Not tested yet with the changes.
Just a question. The decoder updates automatically the date and the hour to the pannel, isn't it?.

When i look the hour on the keypad it shows a different hour and minutes than the ESP32 on the console log. (assuming that the ESP32 is syncronized with an NTP server)

Dilbert66, 2023-02-07:
@pipedeveloper, sorry I just noticed your comment now. The times should match.  I've also added some fixes on the system to correctly update the time on bootup.  If you look at the latest yaml, you will see that it adds a delay then updates it.  It will then update it to the time schecule as set in the yaml.

closed 2023-11-02 by Dilbert66