Dilbert66/esphome-dsckeybus #73: GPIO output?

abishur, 2022-12-29:
I was wondering if there is a way to have a GPIO on the ESP32 toggle on whenever a zone has a fault.

I had to replace my windows recently and they refused to reattach any of the physical wires because they didn't want to drill through the new window frames, something I considered to be a fair concern, so no big loss.

But now I have 5 unused physical inputs on my panel, and since I can't override their status, I was thinking I could trigger a relay from my esp32 if I could have it turn on a GPIO when a zone faulted and the relay would in turn be connected to my original panel to trigger a zone fault.

Is that possible or did I miss an obvious way to override the state of a physical input on the panel?

Dilbert66, 2022-12-29:
I'm not 100% sure what it is you want to do here.  If your panel zones are not connectedd where are the zone faults you talk about coming from??   Sure, you can certainly trigger a gpio from any sensor input but depending on where the source of the zone data comes from, will determine the changes required.   Basically you would just need to setup a gpio switch and then trigger it from the onzonestatus change function using a turn_on or turn_off action.   

The way it reads right now, you have a zone fault from the panel then you want to trigger another zone on the same panel? Doesnt make sense.

abishur, 2022-12-29:
Yeah, I guess I did word it strangely.  My panel is the PC1616, so I can have 16 hard wired zones and 16 wireless zones (If I read the documentation correctly)

I'm using some Aqara Zigbee door and sensor contacts to see if my window is open or shut and I was wondering I could send a message from home assistant to the ESP to turn on a GPIO as an output.  That way if I start to get tight on zones I can use those physical inputs.

As you pointed out if all I'm doing is setting another zone's fault to turn on the GPIO to trigger one of the built-in zone's fault in the panel that would make no sense.

I'm curious if it's possible to just tell the ESP32 to turn on/off the GPIO independent of setting any zone faults on or off.

Dilbert66, 2022-12-29:
Oh I see.. Well, this has nothing to do with the DSC esphome library.  What you are attempting is done just with the basic esphome api capabilities.  Just use a user defined service that you can call from home assistant to turn on the gpio.  Read up here: https://esphome.io/components/api.html#api-services

Dilbert66, 2022-12-29:
Do you have a 5108 zone expander on your panel? As it comes with only 6 hardwired in by default if I read correctly.

abishur, 2022-12-30:
No, I only have the PC1616 it can do 8 inputs I think, but on my board two were left unpopulated.  I could soldering some leads there and use them if push came to shove.

So I'm able to use the DSC esphome library to simulate any zone above 8, anything below that and the panel overwrites (like it should) my attempt to set the fault for that zone.

I'm newish to esphome so I didn't realize I could mix and match functions, though it seems obvious now in hindsight.  Thanks for pointing me straight, I really like this library!

Dilbert66, 2022-12-30:
Great! I"m glad it's working out for you! Yes, you can use any esphome function in addition to the custom ones added by my library.  As you noted, unfortunately you can't set by software the zone status of the hardwired zones unfortunately. (That I know of anyhow)

closed 2022-12-30 by abishur