Dilbert66/esphome-dsckeybus #74: Zones Not Setting/Clearing Faults

abishur, 2022-12-30:
I'm using ESPHome v2022.12.3 and HA 2022.12.8 and did a pull of the "new" esphome-dsckeybus this morning.

I just finished setting up my system tonight and noticed that most my zones using the emulated expander weren't responding.  I could get responses on zones 9, 10, 17, and 18, but nothing from 11-16.  I thought maybe zones 11-16 were reserved by my pc1616 board so I moved zones 11-16 to 19-24, but now none of my zones above 10 respond to the set zone fault command.

I've made sure to enable all the zones from 1-24 using *8-[Installer_code]-202/203/204 and making sure all 8 options were enabled in each area.  Is there a compatibility issue with the latest version of ESPHome?

I've attached my current code, maybe I entered a setting wrong.

```
#for documentation see project at https://github.com/Dilbert66/esphome-dsckeybus

#Zones:
  # Zones built into security panel (there are 8 possible, but only 6 physically present)
    # Zone 1: Front Door
    # Zone 2: Garage Door
    # Zone 3: Fire Alarm (Planned)
    # Zone 4: Unused
    # Zone 5: Unused
    # Zone 6: Motion Detector
    # Zone 7: Not Populated
    # Zone 8: Not Populated
  # Virtual Zones
    # Zone 9: Rear Door
    # Zone 10: Front Room Windows
    # Zone 11: Kitchen Windows
    # Zone 12: Living Room Windows
    # Zone 13: Master Bedroom Windows
    # Zone 14: Master Bathroom Window
    # Zone 15: Office Window
    # Zone 16: Red Room Windows
    # Zone 17: Eden's Windows
    # Zone 18: Boys' Room Window

substitutions:
  # change the name to suit your needs. This is what esphome will use as the friendly name for your component.
  # also used as a prefix for all fields
  systemName: "dscalarm" 
  
  #you can enter a list of user codes mapped to their names for display in armed/disarmed event messages
  userCodes: "1:Matthew,2:Amy,33:Hostage" 
  
  #Only comes into effect if a password prompt occurs when arming eg. night mode
  accessCode: !secret access_code 
  
  #used to select the default partition associated with the alarm panel messages
  defaultPartition: "1" 

  #zone expander addresses:
  # 9  - zones 9-16
  # 10 - zones 17-24
  # 11 - zones 25-32
  # 12 - zones 33-40 (for systems with 64 zone support)
  # 13 - zones 41-48 (for systems with 64 zone support)
  # 14 - zones 49-56 (for systems with 64 zone support)
  # 16 - zones 57-64 (for systems with 64 zone support)  
  expanderAddr1: "9" # 1st zone expander emulator address to use . Set to 0 to disable. 
  expanderAddr2: "10" # 2nd expander emulator address to use . Set to 0 to disable. 
  
  #ESP32 Pins
  dscClockPin: "22"
  dscReadPin: "21"
  dscWritePin: "18"

  #ESP8266 Pins
  #dscClockPin: "5"
  #dscReadPin: "4"
  #dscWritePin: "15"  
  
  maxZones: "32" # maximum amount of zones your system supports
  
[... 408 lines cut ...]
    name: "$systemName partition 2 beeps"
    icon: "mdi:alert-box"   
    
  - platform: template
    id: tr_msg
    name: "$systemName Trouble Msg " # uncomment to show in home assistant
    icon: "mdi:alert-box"
        
switch:
    #shows status of connection status to panel.  You can disconnect before upload using the switch.
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();
  - platform: restart
    id: restart_switch

```

Dilbert66, 2022-12-30:
You need to configure this section for you system.  You'll need to hadd your zones 11,12,13,14,15,16, etc.  I just had a samples there configure to my own system. 
```
DSCkeybus->onZoneStatusChange([&](uint8_t zone, bool open) {
      switch (zone) {
        case 1: id(z1).publish_state(open); break;
        case 2: id(z2).publish_state(open); break;
        case 3: id(z3).publish_state(open); break;
        case 6: id(z6).publish_state(open); break;
        case 9: id(z9).publish_state(open); break;
        case 10: id(z10).publish_state(open); break;
        case 17: id(z17).publish_state(open); break;
        case 18: id(z18).publish_state(open); break;
        case 19: id(z19).publish_state(open); break;
        case 20: id(z20).publish_state(open); break;
        case 21: id(z21).publish_state(open); break;
        case 22: id(z22).publish_state(open); break;
        case 23: id(z23).publish_state(open); break;
        case 24: id(z24).publish_state(open); break;
      } // add or delete any zone you need above. Add the coresponding sensor id in the binary_sensor section
    });
```

abishur, 2022-12-30:
Okay, I've added in all the missing case sections, so I now have entries for case 1 (z1) through case 24 (z24).  And I've added them both in my programming section 202-205 (active zones in Partition 1) and set up the delay types in section 001 and 002 for my zone definitions, but I'm getting the same behavior.

Sometimes I can set it, but then it won't clear and everything gets stuck.  For instance, right now I'm getting this message:
`[10:03:08][D][text_sensor:067]: 'dscalarm zone status ': Sending state 'OP:10,OP:11,OP:12,OP:18,OP:19,OP:20'`

And if I try to clear zone 19 I'll see this in the logs:

```
[10:03:54][D][Debug:496]: Setting Zone Fault: 19,0
[10:03:54][I][Moduledata::971]: 05: FF 01 FF FF DF FF FF FF FF FF 01 00 00 00 00 00 
```

But it will still say it's open a moment later

`[10:04:08][D][text_sensor:067]: 'dscalarm zone status ': Sending state 'OP:10,OP:11,OP:12,OP:18,OP:19,OP:20'`

Dilbert66, 2022-12-30:
Sorry, I have no idea what the issue can be.  All works fine on my system.  Hard to guess.

Dilbert66, 2022-12-30:
One thing I know though is that the pc16 can only support 1 zone expander.  You had two enabled in your config above.  Only use the 09 one.  Set the other to 0

abishur, 2022-12-30:
Okay so I've set it to only use one zone expander, I had misunderstood the documentation, but I'm on the same page now.

I've set my max zones to 16, expanderaddr1 to 9, expanderaddr2 to 0.  I went through and disabled the zones above 16 in the dsc device itself through programming sections 204/205 and section 202 by removing their zone definitions.

When I uploaded I could set and unset zone 9, 10, 15, and 16 as much as I wanted UNTIL I tried to  mess with any zone 11-14.  Once I did that all the zones in the emulated expander locked up, but the hard wired zones (zones 1-6) continued to work with no issue.  I can see in the logs it's receiving my command to set and unset zones, it just doesn't do it.

Here's the bit from the log immediately following my command to set the fault for zone 11

```
[11:36:54][D][text_sensor:067]: 'dscalarm line1': Sending state 'System is Ready'
[11:36:54][D][text_sensor:067]: 'dscalarm line2': Sending state 'Ready to Arm <>'
[11:36:54][D][text_sensor:067]: 'dscalarm Partition 1 Msg ': Sending state '02: Stay zones open'
[11:36:54][D][Debug:496]: Setting Zone Fault: 11,1
[11:36:54][I][Moduledata::971]: 05: FF 01 FF FF BF FF FF FF FF FF 01 00 00 00 00 00 
[11:36:54][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:54][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:54][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:54][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:36:55][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:37:00][I][Paneldata: :971]: 27: 27 00 D5 02 10 C7 00 D5 00 00 00 00 00 00 00 00 
[11:37:00][D][info:1773]: status 02, last status 02,line2status 00,selection 01,partition=1,skip=0
[11:37:00][D][text_sensor:067]: 'dscalarm line1': Sending state 'System is Ready'
[11:37:00][D][text_sensor:067]: 'dscalarm line2': Sending state 'Ready to Arm <>'
[11:37:00][D][binary_sensor:036]: 'Motion Detector (z6)': Sending state OFF
[11:37:00][D][text_sensor:067]: 'dscalarm zone status ': Sending state 'OK'
[11:37:00][I][Paneldata: :971]: 05: 05 00 D5 01 10 C7 10 C7 10 C7 01 00 00 00 00 00 
```

This happens every time I try to set one of the zones from 11-14.

Here's the messages I get right after setting zone 10 (A Zone that Works):

```
[12:05:27][D][Debug:496]: Setting Zone Fault: 10,1
[12:05:27][I][Moduledata::971]: 05: FF 01 FF FF BF FF FF FF FF FF 01 00 00 00 00 00 
[12:05:27][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[12:05:27][I][Moduledata::971]: 28: FF 01 5D 00 FF FF 2F 00 00 00 00 00 00 00 00 00 
[12:05:27][I][Paneldata: :971]: 2D: 2D 00 D5 02 10 C7 02 DD 00 00 00 00 00 00 00 00 
[12:05:27][D][info:1773]: status 02, last status 02,line2status 00,selection 01,partition=1,skip=0
[12:05:27][D][text_sensor:067]: 'dscalarm line1': Sending state 'System is Ready'
[12:05:27][D][text_sensor:067]: 'dscalarm line2': Sending state 'Ready to Arm <>'
[12:05:27][D][binary_sensor:036]: 'Front Room Windows (z10)': Sending state ON
```
And Zone 11:
```
[11:47:08][D][Debug:496]: Setting Zone Fault: 11,0
[11:47:08][I][Moduledata::971]: 05: FF 01 FF FF BF FF FF FF FF FF 01 00 00 00 00 00 
[11:47:09][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:47:09][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:47:09][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:47:09][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:47:09][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:47:09][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[11:47:09][I][Paneldata: :971]: 28: 28 00 FF FF FF FF FF 00 00 00 00 00 00 00 00 00 
[11:47:09][I][Moduledata::971]: 28: FF 01 FF FF 55 55 FF 00 00 00 00 00 00 00 00 00 
[... 474 lines cut ...]
    id: beeps_2
    name: "$systemName partition 2 beeps"
    icon: "mdi:alert-box"   
    
  - platform: template
    id: tr_msg
    name: "$systemName Trouble Msg " # uncomment to show in home assistant
    icon: "mdi:alert-box"
        
switch:
    #shows status of connection status to panel.  You can disconnect before upload using the switch.
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();
  - platform: restart
    id: restart_switch
```

Dilbert66, 2022-12-30:
ahhh. ok I see the issue.  The reset zone fault logic doesnt work right on your panel.   The emulated 0 response is not being accepted by the panel and it's being rejected, hence the multitude of 28 cmds and responses.   What is your zone termination set to? single end of line or double end of line?

Dilbert66, 2022-12-30:
yes, it's a bug. I'll correct and push

abishur, 2022-12-30:
Cool!  Glad to help (also if it makes any difference the answer to your initial question is it looks like my system was set up for normally closed loops with single end of line resistors)

Dilbert66, 2022-12-30:
ok, I did find the issue. Chksum was not being calculated correctly for the response. I've pushed an update.

abishur, 2022-12-31:
Thanks!  That did it and now I can trigger all of the emulated zones.  It's even working for zones on the second expander address.

Dilbert66, 2022-12-31:
Awesome! Glad it's doing the job,!

closed 2023-01-16 by abishur