Dilbert66/esphome-dsckeybus #75: Question/Comment - Log levels (*new* branch)

vincentdh, 2023-01-08:
Hi,

First of all, I really like your library and I think that you are doing a great job! 

There is one thing I've notice and is kind of annoying. There is a lot of "non relevant" data when the level is set to INFO and it makes it difficult to find the info we are looking for.

All my other devices uses the INFO level and the logs only shows data that is relevant for a "standard human" like, in example, a sensor state changed or a service is called.

At the moment, Moduledata and Paneldata are showed in the log at INFO level. In my opinion, as these informations are for devs or people who understand the keybus format, those lines are relevant in DEBUG or event VERBOSE levels.

That being said, I would like to know why those are in INFO and if you are open to change those to, at least, DEBUG?

Thanks!

Dilbert66, 2023-01-08:
It's in dev at this point.  Useful for testing and users reporting issues.  I'll switch to lesser logs when I merge to master.

vincentdh, 2023-01-09:
Thanks for the quick response!

From what I can see, it looks like mostly all packets on the bus are logged. Otherwise, everything **seems** to be working properly.

My module is currently disconnected, but I will post the logs later today.  Maybe there is something wrong...

Glazzanz, 2023-02-05:
Ive got no doubt I have a similar issue - I'm picking there is something else going on - but the keypad is not beeping, nor does it have any errors on it.

`[17:03:20][D][text_sensor:067]: 'Alarm event': Sending state ''
[17:03:28][I][Paneldata: :978]: A5: A5 00 23 08 B1 10 00 00 91 00 00 00 00 00 00 00 
[17:03:28][D][info:1785]: status 01, last status 01,line2status 3F,selection 01,partition=1,skip=0
[17:03:28][D][text_sensor:067]: 'Alarm line1': Sending state 'System is Ready'
[17:03:28][D][text_sensor:067]: 'Alarm line2': Sending state 'Ready to Arm <>'
[17:03:28][D][text_sensor:067]: 'Alarm Trouble Msg ': Sending state ''
[17:03:28][D][text_sensor:067]: 'Alarm Partition 1 Msg ': Sending state '01: Ready'
[17:03:29][I][Paneldata: :978]: C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 
[17:03:37][D][text_sensor:067]: 'Alarm line2': Sending state 'Ready to Arm <>'
[17:03:37][D][text_sensor:067]: 'Alarm System Status': Sending state 'online'
[17:03:41][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:03:42][D][text_sensor:067]: 'Alarm zone status ': Sending state 'OK'
[17:03:42][I][Paneldata: :978]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[17:03:42][I][Moduledata::978]: 11: FF 01 FF F3 3F FF FF FF FF 03 00 00 00 00 00 00 
[17:03:43][D][text_sensor:067]: 'Alarm Partition 1 Status ': Sending state 'disarmed'
[17:03:43][D][text_sensor:067]: 'Alarm Trouble Msg ': Sending state ''
[17:03:46][D][text_sensor:067]: 'Alarm line1': Sending state 'System is Ready'
[17:03:50][D][text_sensor:067]: 'Alarm event': Sending state ''
[17:03:50][D][text_sensor:067]: 'Alarm event': Sending state ''
[17:03:58][I][Paneldata: :978]: E6: E6 00 2C 01 00 00 00 00 13 00 00 00 00 00 00 00 
[17:03:58][I][Paneldata: :978]: B1: B1 00 FF FF 00 00 00 00 00 00 AF 00 00 00 00 00 
[17:03:59][I][Paneldata: :978]: C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 
[17:04:10][D][info:1785]: status 01, last status 01,line2status 3F,selection 01,partition=1,skip=0
[17:04:10][D][text_sensor:067]: 'Alarm line1': Sending state 'System is Ready'
[17:04:10][D][text_sensor:067]: 'Alarm line2': Sending state 'Ready to Arm <>'
[17:04:10][D][text_sensor:067]: 'Alarm System Status': Sending state 'online'
[17:04:10][D][text_sensor:067]: 'Alarm Partition 1 Status ': Sending state 'disarmed'
[17:04:10][D][text_sensor:067]: 'Alarm zone status ': Sending state 'OK'
[17:04:10][D][text_sensor:067]: 'Alarm Trouble Msg ': Sending state ''
[17:04:10][I][Moduledata::978]: 05: FF 01 28 FF FF FF FF FF FF FF 01 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: 5D: 5D 00 00 00 00 00 00 5D 00 00 00 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: E6: E6 00 18 01 00 00 00 00 00 FF 00 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: 0A: 0A 00 80 9E 00 00 00 00 00 28 00 00 00 00 00 00 
[17:04:10][I][Moduledata::978]: 0A: FF 01 0A FF FF FF FF FF FF FF 01 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: 0A: 0A 00 80 A1 00 00 00 00 00 2B 00 00 00 00 00 00 
[17:04:10][I][Moduledata::978]: 0A: FF 01 05 FF FF FF FF FF FF FF 01 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: 5D: 5D 00 10 00 00 00 00 6D 00 00 00 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: E6: E6 00 18 01 10 00 00 00 00 0F 00 00 00 00 00 00 
[17:04:10][I][Paneldata: :978]: 64: 64 00 04 68 00 00 00 00 00 00 00 00 00 00 00 00 
[17:04:10][D][text_sensor:067]: 'Alarm beeps': Sending state '2'
[17:04:10][D][info:1785]: status 01, last status 01,line2status 3F,selection 01,partition=1,skip=0
[17:04:10][D][text_sensor:067]: 'Alarm line1': Sending state 'System is Ready'
[17:04:10][D][text_sensor:067]: 'Alarm line2': Sending state 'Ready to Arm <>'
[17:04:10][D][text_sensor:067]: 'Alarm Trouble Msg ': Sending state ''
[17:04:11][I][Paneldata: :978]: 0A: 0A 00 80 C8 00 00 00 00 00 52 00 00 00 00 00 00 
[17:04:11][I][Moduledata::978]: 0A: FF 01 2D FF FF FF FF FF FF FF 01 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: 0A: 0A 00 80 A1 00 00 00 00 00 2B 00 00 00 00 00 00 
[17:04:11][I][Moduledata::978]: 0A: FF 01 1C FF FF FF FF FF FF FF 01 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: 5D: 5D 00 10 00 00 00 00 6D 00 00 00 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: E6: E6 00 18 01 10 00 00 00 00 0F 00 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: 64: 64 00 02 66 00 00 00 00 00 00 00 00 00 00 00 00 
[17:04:11][D][text_sensor:067]: 'Alarm beeps': Sending state '1'
[17:04:11][D][info:1785]: status 01, last status 01,line2status 3F,selection 01,partition=1,skip=0
[17:04:11][D][text_sensor:067]: 'Alarm line1': Sending state 'System is Ready'
[17:04:11][D][text_sensor:067]: 'Alarm line2': Sending state 'Ready to Arm <>'
[17:04:11][D][text_sensor:067]: 'Alarm Trouble Msg ': Sending state ''
[17:04:11][I][Paneldata: :978]: 0A: 0A 00 80 BA 00 00 00 00 00 44 00 00 00 00 00 00 
[17:04:11][I][Moduledata::978]: 0A: FF 01 2D FF FF FF FF FF FF FF 01 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: 0A: 0A 00 80 A1 00 00 00 00 00 2B 00 00 00 00 00 00 
[17:04:11][I][Moduledata::978]: 0A: FF 01 2D FF FF FF FF FF FF FF 01 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: 5D: 5D 00 10 00 00 00 00 6D 00 00 00 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: E6: E6 00 18 01 10 00 00 00 00 0F 00 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: 5D: 5D 00 00 00 00 00 00 5D 00 00 00 00 00 00 00 00 
[17:04:11][I][Paneldata: :978]: E6: E6 00 18 01 00 00 00 00 00 FF 00 00 00 00 00 00 
[17:04:12][I][Paneldata: :978]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[17:04:12][I][Moduledata::978]: 11: FF 01 FF F3 3F FF FF FF FF 03 00 00 00 00 00 00 
[17:04:13][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:04:20][D][text_sensor:067]: 'Alarm event': Sending state ''`

Dilbert66, 2023-02-05:
You have a device on your bus that is reporting trouble intermittently.  Do a *2 to view the trouble list to see what's going on.  It's a very quick status change so that might be why your physical keypad is not beeping but the logs are showing the message from the panel.

Dilbert66, 2023-02-05:
Are you saying this is happening all the time??? This should not be unless your esp is always rebooting and that would be a different issue.

Glazzanz, 2023-02-06:
Hi, thats for your prompt answer.  Yes I am seeing this all the time.  From the keypad there is nothing in the trouble list.

Within Home Assistant, the log looks something like this:

[image]

I dont believe the ESP is rebooting, as I can remain connected to the esp log output...

Dilbert66, 2023-02-06:
Can you look at the file dscalarm.h and look for this line
 bool troubleFetch=false;

Make sure it's false.
Probably around line 294. 

That's the only thing that would keep sending that command. Otherwise there is something else going on your panel that causes a trouble indicator to keep showing up.

Dilbert66, 2023-02-06:
Can you indicate the time difference between each occurance?

Dilbert66, 2023-02-06:
ah.. Ok, I see the issue.  Use the Dev branch. This the new updated branch. I've merged the "new" branch to dev now and working on that. There is  a bug in the "new" branch that is causing this.

Dilbert66, 2023-02-06:
FYI, the issue is a missing condition on this line:

```
 if (!dsc.disabled[defaultPartition-1] && !partitionStatus[defaultPartition-1].locked) {
``
It should be:
```
            if (!dsc.disabled[defaultPartition-1] && !partitionStatus[defaultPartition-1].locked && troubleFetch) {
```

Glazzanz, 2023-02-06:
[17:37:12][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:37:13][D][text_sensor:067]: 'Alarm beeps': Sending state '2'
[17:37:13][D][text_sensor:067]: 'Alarm beeps': Sending state '1'
[17:37:15][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:38:12][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:38:13][D][text_sensor:067]: 'Alarm beeps': Sending state '2'
[17:38:13][D][text_sensor:067]: 'Alarm beeps': Sending state '1'
[17:38:15][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:39:12][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:39:13][D][text_sensor:067]: 'Alarm beeps': Sending state '2'
[17:39:13][D][text_sensor:067]: 'Alarm beeps': Sending state '1'
[17:39:15][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:40:12][D][text_sensor:067]: 'Alarm beeps': Sending state '0'
[17:40:13][D][text_sensor:067]: 'Alarm beeps': Sending state '2'
[17:40:13][D][text_sensor:067]: 'Alarm beeps': Sending state '1'
[17:40:15][D][text_sensor:067]: 'Alarm beeps': Sending state '0'

Glazzanz, 2023-02-06:
Thanks - will test very soon!

Glazzanz, 2023-02-06:
Thank you - all resolved :)

Dilbert66, 2023-02-06:
@vincentdh , seems there was a bug in the "new" branch that had an issue. I've removed that branch. Please use the "dev" branch .

Dilbert66, 2023-03-04:
I fixed another issue with the branch causing excessive bus traffic.  Might have creeped in during the merge of new to dev.  Anyhow, i've corrected it now.

vincentdh, 2023-09-09:
Hi!

It's been a long time since I've looked at my alarm project, sorry! 😅

However, the problem was finally on my side ... I've forgot that I had increase the debug level somewhere else, which explains why I had so much stuff in the console 🙄😅

I will close the issue as it was a user error!

Keep up the excellent work!

closed 2023-09-09 by vincentdh