Dilbert66/esphome-dsckeybus #77: Arm away

w1tw0lf, 2023-02-20:
Swicthed to the dev version from new.

Everything works as needed, except the arm away service. Below is from the logs when running the service. Disarm, night and stay service works correctly.

`
08:10:46 | [D] | [text_sensor:067] | 'alarm Trouble Msg ': Sending state ''
-- | -- | -- | --
08:10:47 | [D] | [text_sensor:067] | 'alarm event': Sending state ''
08:10:47 | [D] | [debug:927] | Setting Alarm state: W to partition 1
08:10:49 | [I] | [Paneldata: | 94]: B1: B1 00 FF FF 00 00 00 00 00 00 AF 00 00 00 00 00
08:10:49 | [I] | [Paneldata: | 94]: B1: B1 00 FF FF 00 00 00 00 00 00 AF 00 00 00 00 00
08:10:49 | [D] | [info:1853] | status 01, last status 01,line2status FB,selection 01,partition=1,skip=0,force=1
08:10:49 | [D] | [info:1853] | status 01, last status 01,line2status FB,selection 01,partition=1,skip=0,force=1
08:10:49 | [D] | [text_sensor:067] | 'alarm line1': Sending state 'System is Ready'
`

When pressing the button on the included card it arms without issue. Please advise if I might have missed something.

Dilbert66, 2023-02-20:
I switched to using "A" instead of W for away but I did modify the code to support both now

Dilbert66, 2023-02-20:
whups sorry. I modified the vista system... I worked on two different ones. hold on.

w1tw0lf, 2023-02-20:
:smiley: Eish.... no worries and thank you.

w1tw0lf, 2023-02-20:
All sorted again. Thank you very much.

closed 2023-02-20 by w1tw0lf

fede843, 2023-02-22:
had the same issue. All sorted now. Thanks!