Dilbert66/esphome-dsckeybus #83: Secret 'access_code' not defined

spplecxer, 2023-04-06:
I have a DSC Power832 (PC5016 v 1.13).
Copied the files.
Edited to work with an ESP32 dev board.
Tried to validate and got this error:

INFO Reading configuration /config/esphome/DscAlarm.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Secret 'access_code' not defined
  in "/config/esphome/DscAlarm.yaml", line 4, column 15:
      accessCode: !secret access_code #Only comes  ...

spplecxer, 2023-04-06:
Ok found it but leaving it here for others just in case.
You have to add your secrets in the secrets yaml file.

virus100b, 2023-10-26:
Hi, I have this running o nodemcu ESP 8266, what lines need to be modified in order to work with esp32 board? 
And another question, i cannot update the firmware over the air, with cable is working fine. It is compiling and then i et the following error.

RAM:   [=====     ]  47.2% (used 38660 bytes from 81920 bytes)
Flash: [=====     ]  50.0% (used 521805 bytes from 1044464 bytes)
Building .pioenvs/dscalarm/firmware.bin
esp8266_copy_factory_bin([".pioenvs/dscalarm/firmware.bin"], [".pioenvs/dscalarm/firmware.elf"])
======================== [SUCCESS] Took 711.31 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of dscalarm.local
INFO  -> 192.168.1.125
INFO Uploading /data/build/dscalarm/.pioenvs/dscalarm/firmware.bin (525952 bytes)
INFO Compressed to 364430 bytes
Uploading: [==                                                          ] 3% 
ERROR Error sending data: [Errno 104] Connection reset by peer

 What could be the issue?

closed 2024-11-01 by Dilbert66