Dilbert66/esphome-dsckeybus #86: alarm-keypad-card error in Home Asistant

jereitor, 2023-05-15:
Hello. Since version 2023.5.0 the card " alarm-keypad-card " I receive an error in the log and it does not show the card in the frontend.

Logger: frontend.js.latest.202305033
Source: components/system_log/__init__.py:257
First occurred: May 7, 2023, 20:01:19 (15 occurrences)
Last logged: 13:00:48

https://qfph5rffk0ysckutm42ub9rmdsqo9ies.ui.nabu.casa/local/alarm-keypad-card/alarm-keypad-card.js:3:31 Uncaught ReferenceError: Polymer is not defined
https://qfph5rffk0ysckutm42ub9rmdsqo9ies.ui.nabu.casa/local/alarm-keypad-card/alarm-keypad-card.js:3:38 ReferenceError: Can't find variable: Polymer.

All the best

Dilbert66, 2023-05-16:
Use the card version from dev branch

fede843, 2023-05-19:
Hello, I am in the same situation. I have just updated to the latest dev card, still getting 

```
Logger: frontend.js.latest.202305033
Source: components/system_log/__init__.py:257
First occurred: 13:59:24 (1 occurrences)
Last logged: 13:59:24

http://192.168.254.70:8123/local/alarm-keypad-card.js:3:31 Uncaught ReferenceError: Polymer is not defined
```

olympia, 2023-05-19:
Please note the NOTE in the instructions:

To configure the card, just place the main script alarm-panel-card.js and associated *.mp3 sound files into the /config/www directory of your homeassistant installation and add a new resource in your lovelace configuration pointing to /local/alarm-panel-card.js?id=1. You can then configure the card as shown below. Just substitute your service name to your application. This is an example for using it in a multi partition environment where you can have separate distinct cards for each partition just by changing the services used and partition numbers.

*** NOTE *** : Every time you update this file, you will need to increase the id=xx number in the url by 1 in order to force the browser to reload it.

fede843, 2023-05-19:
thanks for the quick reply. OK, did not know about the ID. Sorry about that, now that seems fixed.

I am confused about "Just substitute your service name to your application", what would that be?

I am still getting "Custom element doesn't exist: alarm-keypad-card." at the panel card.

fede843, 2023-05-19:
OK, in my case I left most as default, found the ESPHome: dscalarm_alarm_keypress_partition service. That is fine then. Still I am failing to get the card working

olympia, 2023-05-19:
Use 2 as Id

fede843, 2023-05-19:
yes, I did already, same result. Do the Id need to be part anywhere in the card?

jereitor, 2023-05-20:
The truth is that I'm a bit confused with this, it worked perfectly for me until after an HA update.

Dilbert66, 2023-05-21:
HA obsoleted the polymer web component library which the card used, I've migrated it to the supported Lit library a while a go on the dev version which you should now be using.  As to the issues you are having, what was recommended above is correct.  The id is only needed in the url so the browser sees it as different. You might need to close your browser or refresh it with F5 for it to reload correctly otherwise I have no idea what your issue would be.

fede843, 2023-05-22:
Thanks.
Still getting the same, I will try to debug a bit more. Also will try to go to a previous HA version to see if that makes a difference here. I am currently on 2023.5.3.

jereitor, 2023-05-22:
Hi.  Indeed this problem is due to the use of obsolete branch, I updated all the files with the new branch and everything is working fine.  Thank you all .

closed 2023-05-24 by jereitor