Dilbert66/esphome-dsckeybus #89: ESPHome Compiling Error

1gnwld, 2023-05-31:
Getting the following error when trying to compile in ESPHome. I have downloaded the files from your GitHub repo and uploaded directly into HA.

INFO ESPHome 2023.5.5
INFO Reading configuration /config/esphome/DscAlarm.yaml...
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing dscalarm (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
Compiling /data/dscalarm/.pioenvs/dscalarm/src/main.cpp.o
Archiving /data/dscalarm/.pioenvs/dscalarm/lib915/libESPmDNS.a
Compiling /data/dscalarm/.pioenvs/dscalarm/FrameworkArduino/Esp.cpp.o
In file included from src/main.cpp:59:
src/dscAlarm.h:21:71: error: expected class-name before '{' token
 class DSCkeybushome : public PollingComponent, public CustomAPIDevice {
                                                                       ^
In file included from src/main.cpp:59:
src/dscAlarm.h: In member function 'virtual void DSCkeybushome::setup()':
src/dscAlarm.h:71:5: error: 'register_service' was not declared in this scope
     register_service(&DSCkeybushome::set_alarm_state,"set_alarm_state", {"partition","state","code"});
     ^~~~~~~~~~~~~~~~
src/dscAlarm.h:71:5: note: suggested alternative: 'register_t'
     register_service(&DSCkeybushome::set_alarm_state,"set_alarm_state", {"partition","state","code"});
     ^~~~~~~~~~~~~~~~
     register_t
*** [/data/dscalarm/.pioenvs/dscalarm/src/main.cpp.o] Error 1
========================== [FAILED] Took 3.13 seconds ==========================

Dilbert66, 2023-06-05:
Ensure you have this section in your yaml:
```
time:
 - platform: sntp
   id: sntp_time 
```

1gnwld, 2023-06-05:
I double checked dscalarm.yaml and I do have that entry. 
I added my yaml file with the password removed. Am I missing something here?
[dscalarm.txt](https://github.com/Dilbert66/esphome-dsckeybus/files/11653493/dscalarm.txt)

Dilbert66, 2023-06-07:
You have commented the "api:" line. That is necessary. Either mqtt or api.

1gnwld, 2023-06-07:
DANG IT!!
You are right on the money! IT WORKS!!!
Thank you!!!

closed 2023-06-07 by 1gnwld