Dilbert66/esphome-dsckeybus #96: ESPhome 2023.7.0

w1tw0lf, 2023-07-19:
After upgrading to ESPhome 2023.7.0, the api is broken. Not connecting at all to esphome. Responds to ping and able to do ota uploads. Clean installed and still no luck.

w1tw0lf, 2023-07-19:
Managed to grab the logs via usb
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:262]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:469]: There have been 4 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][template.switch:046]:   Restored state OFF
[D][switch:016]: 'alarm Connection' Turning OFF.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400fab67  PS      : 0x00060430  A0      : 0x800d3da0  A1      : 0x3ffb2500  
A2      : 0x00000000  A3      : 0x3f401e04  A4      : 0x0000002e  A5      : 0x3f401df0  
A6      : 0x3f400e6c  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000004  
A10     : 0x3ffb88d4  A11     : 0x80000001  A12     : 0x800905d8  A13     : 0x3ffb2400  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x4008ab6d  LEND    : 0x4008ab7d  LCOUNT  : 0xfffffff0  

Backtrace:0x400fab64:0x3ffb25000x400d3d9d:0x3ffb2520 0x400e6635:0x3ffb2540 0x400e664f:0x3ffb2560 0x400e5a1d:0x3ffb2580 0x400e5873:0x3ffb25a0 0x401835d5:0x3ffb25c0 0x400dde75:0x3ffb25e0 0x400dd99f:0x3ffb2600 0x400dddfc:0x3ffb2620 0x400dde0a:0x3ffb2650 0x401839c5:0x3ffb2670 0x40183a75:0x3ffb2690 0x400e42d8:0x3ffb26b0 0x400eed9e:0x3ffb26e0 0x400fb4ee:0x3ffb2820 
```

Seems to repeat a few times, then this comes up
```
ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:262]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:469]: There have been 10 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[E][ota:476]: Boot loop detected. Proceeding to safe mode.
[I][app:029]: Running through setup()...
[C][wifi:038]: Setting up WiFi...
[C][wifi:048]: Starting WiFi...
[C][wifi:049]:   Local MAC: xx.xx.xx.xx.xx.xx
[I][wifi:274]: WiFi Connecting to 'xxxxxxxxxxx'...
[I][wifi:557]: WiFi Connected!
[C][wifi:379]:   Local MAC: xx.xx.xx.xx.xx.xx
[C][wifi:380]:   SSID: 'xxxxxxxxxxx'[redacted]
[C][wifi:381]:   IP Address: xxx.xxx.xx.x13
[C][wifi:383]:   BSSID: xx.xx.xx.xx.xx.xx[redacted]
[C][wifi:384]:   Hostname: 'alarm'
[C][wifi:386]:   Signal strength: -37 dB ▂▄▆█
[C][wifi:390]:   Channel: 1
[C][wifi:391]:   Subnet: 255.255.255.0
[C][wifi:392]:   Gateway: xxx.xxx.xx.x
[C][wifi:393]:   DNS1: xxx.xxx.xx.x
[C][wifi:394]:   DNS2: 0.0.0.0
[D][wifi:566]: Disabling AP...
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: xxx.xxx.xx.x13:3232
[C][ota:097]:   Using Password.
[W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[I][app:062]: setup() finished successfully!
[I][ota:489]: Waiting for OTA attempt.
[I][app:102]: ESPHome version 2023.7.0 compiled on Jul 19 2023, 14:12:10
[C][status_led:019]: Status LED:
[C][status_led:020]:   Pin: GPIO2
[C][wifi:543]: WiFi:
[C][wifi:379]:   Local MAC: xx.xx.xx.xx.xx.xx
[C][wifi:380]:   SSID: 'xxxxxxxxxxx'[redacted]
[C][wifi:381]:   IP Address: xxx.xxx.xx.x13
[C][wifi:383]:   BSSID: xx.xx.xx.xx.xx.xx[redacted]
[C][wifi:384]:   Hostname: 'alarm'
[C][wifi:386]:   Signal strength: -39 dB ▂▄▆█
[C][wifi:390]:   Channel: 1
[C][wifi:391]:   Subnet: 255.255.255.0
[C][wifi:392]:   Gateway: xxx.xxx.xx.x
[C][wifi:393]:   DNS1: xxx.xxx.xx.x
[C][wifi:394]:   DNS2: 0.0.0.0
[C][logger:301]: Logger:
[C][logger:302]:   Level: DEBUG
[C][logger:303]:   Log Baud Rate: 115200
[C][logger:305]:   Hardware UART: UART0
[C][mdns:112]: mDNS:
[C][mdns:113]:   Hostname: alarm
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: xxx.xxx.xx.x13:3232
[C][ota:097]:   Using Password.
[W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
Terminal disconnected: NetworkError: The device has been lost.
```

Dilbert66, 2023-07-19:
Tks.  Yes, the new esphome libs have caused an issue with timers used. Looking into a fix.

w1tw0lf, 2023-07-19:
All seem sorted. Thank you for the quick response and fix.

closed 2023-07-19 by w1tw0lf

coolstuff99, 2023-07-23:
Hi,

Looks like timer1 declaration is missing and unable to compile on my end:
[image]

Thank you,

coolstuff99, 2023-07-23:
Replacing timer1 by timer0 in "dscKeybusInterface.cpp" did the trick:
`  // Disables esp32 timer0
  #elif defined(ESP32)
  if (timer0 != NULL)  {  
    timerAlarmDisable(timer0);
    timerEnd(timer0);
  }  
`
But since not familiar enough with the code, would be great someone else to confirm.

Thanks,

Dilbert66, 2023-07-23:
There is someting  wrong with your setup.  Timer1 is assigned in dsckeybus.h and is the correct one to use.

coolstuff99, 2023-07-23:
Hi @Dilbert66,

First of all, thanks for your job done in this project. Been using it for 2 years without any issue.
About my problem, I was troubleshooting as per your last comment and I didn't found the declaration file "dsckeybus.h" in your project but I found this one in the other project https://github.com/taligentx/dscKeybusInterface/blob/master/src/dscKeybus.h
I don't know if from now on I should add this project too which never was required to work on my side.
Side note, I am using esp32 instead of esp8266. Maybe esp32 keeps using time0 one?

Thank you,

w1tw0lf, 2023-07-23:
Under correction, it should be dscKeybusInterface.h and  not dscKeybus.h. @Dilbert66 please confirm.

Dilbert66, 2023-07-23:
> Hi @Dilbert66,
> 
> First of all, thanks for your job done in this project. Been using it for 2 years without any issue. About my problem, I was troubleshooting as per your last comment and I didn't found the declaration file "dsckeybus.h" in your project but I found this one in the other project https://github.com/taligentx/dscKeybusInterface/blob/master/src/dscKeybus.h I don't know if from now on I should add this project too which never was required to work on my side.
> 
> Thank you,

Ok, I see the issue. You are are using master branch and that one is missing the file. i will correct that.  We were discussing dev branch here which is the latest recommended branch to use.

coolstuff99, 2023-07-23:
Yes, I always used Master. Thanks for confirmation.
I was not aware about Dev being recommend one so I will have a look to shift to this one.
Thanks again :)

Dilbert66, 2023-07-23:
I recommend it as it is where all new dev is done as far features and optimizations.  If master works for you, then you can stick with it. I try and backport any major issue fixes if I can.

PipeDeveloper, 2023-07-23:
After updating i see this error on console

[18:52:51][W][component:204]: Component <unknown> took a long time for an operation (0.07 s).
[18:52:51][W][component:205]: Components should block for at most 20-30ms.

Dilbert66, 2023-07-24:
Normal. It doesnt like the fact that the custom component is taking a while to execute.  This is a new thing with Esphome. It's not affecting anything at this point.