Dilbert66/esphome-dsckeybus #99: Partition Status Unavailable

meneerfrikkie, 2023-08-11:
Hi everyone, thank you to the Dilbert66 for the amazing work, has anyone come across the partition status where it is unavailable, could they please assisting, Many Thanks!
[image]

tiaanv, 2023-09-22:
I get the same.  Did you manage to figure it out?

Dilbert66, 2023-10-20:
It means that the partition is not ready. You can change it to be "Not Ready" in line 98 of dscalarm.h
```
const char STATUS_NOT_READY[] PROGMEM = "unavailable"; //ha alarm panel likes to see "unavailable" instead of not_ready when the system can't be armed
```

Dilbert66, 2023-10-22:
I've modified the code to show "not_ready" instead.

closed 2023-11-02 by Dilbert66