Dilbert66/esphome-dsckeybus #104: Wemos D1 mini ESPHome API error

feinadam, 2023-09-26:
Hello!
After compiling and flashing the wemos D1 mini I'm receiving the following warning:

WARNING Can't connect to ESPHome API for riaszto.local: Error connecting to ('192.168.1.110', 6053): [Errno 111] Connect call failed ('192.168.1.110', 6053) (SocketAPIError)

I can see the device in ESPhome and i'm also able to reflash(Ota) but there is no communication, no sensor data receiving in HA.

When I removed the dscKeybusInterface include the Warning message is gone and I'm able to see the logs.
I have also tried to give fix IP, to the device but same issue.

Any idea?

Daniel-Magg, 2023-10-06:
I had the same or similar issue. My alarm is a PC1565 so might not have all features.

Commenting some lines out in the following function fixed things for me. You might play around and comment more or less out...

`void dscKeybusInterface::resetStatus() {
  statusChanged = true;
  keybusChanged = true;
  troubleChanged = true;
  powerChanged = true;
  batteryChanged = true;
  for (byte partition = 0; partition < dscPartitions; partition++) {
    readyChanged[partition] = true;
    armedChanged[partition] = true;
    alarmChanged[partition] = true;
    //fireChanged[partition] = true;
    disabled[partition] = true;
  }
  openZonesStatusChanged = true;
  alarmZonesStatusChanged = true;
  for (byte zoneGroup = 0; zoneGroup < dscZones; zoneGroup++) {
    openZonesChanged[zoneGroup] = 0xFF;
    //alarmZonesChanged[zoneGroup] = 0xFF;
  }
  pgmOutputsChanged[0] = 0xFF;
  pgmOutputsChanged[1] = 0x3F;
}`

Glazzanz, 2023-10-20:
My feeling is there is something that is making it unhappy with newer versions of esphome.

feinadam, 2023-10-20:
> I had the same or similar issue. My alarm is a PC1565 so might not have all features.
> 
> Commenting some lines out in the following function (in dscKeybusProcessData.cpp) fixed things for me. You might play around and comment more or less out...
> 
> void dscKeybusInterface::resetStatus() { statusChanged = true; keybusChanged = true; troubleChanged = true; powerChanged = true; batteryChanged = true; for (byte partition = 0; partition < dscPartitions; partition++) { readyChanged[partition] = true; armedChanged[partition] = true; alarmChanged[partition] = true; **//fireChanged[partition] = true;** disabled[partition] = true; } openZonesStatusChanged = true; alarmZonesStatusChanged = true; for (byte zoneGroup = 0; zoneGroup < dscZones; zoneGroup++) { openZonesChanged[zoneGroup] = 0xFF; **//alarmZonesChanged[zoneGroup] = 0xFF;** } pgmOutputsChanged[0] = 0xFF; pgmOutputsChanged[1] = 0x3F; }
> 
> you also want to change this function, otherwise the esp freezes when the alarm is triggered:
> 
> void dscKeybusInterface::processAlarmZonesStatus(byte zonesByte, byte zoneCount, byte writeValue) { byte zonesRange = zonesByte; if (zonesRange >= 4) zonesRange -= 4;
> 
> bitWrite(alarmZones[zonesByte], (zoneCount - (zonesRange * 8)), writeValue);
> 
> if (bitRead(previousAlarmZones[zonesByte], (zoneCount - (zonesRange * 8))) != writeValue) { bitWrite(previousAlarmZones[zonesByte], (zoneCount - (zonesRange * 8)), writeValue); **//bitWrite(alarmZonesChanged[zonesByte], (zoneCount - (zonesRange * 8)), 1);**
> 
> ```
> alarmZonesStatusChanged = true;
> if (!pauseStatus) statusChanged = true;
> ```
> 
> } }

Thank you, for the hint I will try it out. In the mean time i have tried the implementation of taligentx. That is currently it is working via mqtt. But since the dsckeybus lib is the same, it should work.

Dilbert66, 2023-10-20:
Could be a dozen different things. You did not provide logs or indicate which branch you are using. It could be a wiring error or a memory error.  The dev version will most likely only work with an esp32 now due esphome needed more memory then previously and the increased size of the dev code.  Also be aware that this library is very different then the taligent version.  It is is highly modified as it includes other functionality not present in the original code such as emulated virtual expanders and lcd keypad emulation.

Daniel-Magg, 2023-11-10:
I cannot speak for feinadam, but I have tried both dev branch and master branch on ESP32 and ESP8266 and never got it to work with my PC1565. I get the same/similar error as feinadam describes in his initial post. It is not a wiring error since it works fine when using the taligentx mqtt version, plus I triple checked the connections and even build 2 different shields.

What I forgot to mention in my previous posts, the patches mentioned to make it work with esphome are for the taligentx v3.0 source code, not for this one, but I have a feeling this one could be make to work too...

Dilbert66, 2023-11-10:
The pc1565 is not supported on my branch . I only support the power series
as this is what I have to test with

Dilbert66, 2023-11-10:
I can consider adding support for it if there is enough demand but it difficult for me as I cant test and debug it. You folks would have to be the testers. Not very efficient though but possible.

feinadam, 2024-03-31:
With the latest version the issue disappeared.