Dilbert66/esphome-dsckeybus #108: DSCKeybus keypad control no longer working

feradelectronics, 2023-11-22:
Since the last dev updates after creating short yaml configuration the DSCKeybus keypad is not working reliably.  Cannot arm because system reports unknown.  I can manually arm system from DSC keypad.  DSCKeybus then reports system armed as expected.  I can then disarm system with DSCKeybus.  Seems that communication is ok.  Please let me know how to resolve this.

Dilbert66, 2023-11-22:
I need to see logs and config.  I can't guess as it works fine here.

feradelectronics, 2023-11-22:
Thanks in advance.

 

 ***@***.***> Scott Fritzler C.E.T.

 <https://ferad.com/> FERAD Electronics Inc.

B, 12736 St. Albert Trail NW

Edmonton, Alberta, T5L 4S5

Phone: 587-773-3333 or 780-628-3649

Toll Free: 1855-53-FERAD (1855-533-3723)

 

This email has been checked for viruses by AVG antivirus software.
www.avg.com
INFO ESPHome 2023.11.3
INFO Reading configuration /config/esphome/dscalarm.yaml...
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'America/Edmonton'
INFO Starting log output from dscalarm.local using esphome API
INFO Successfully connected to dscalarm in 0.344s
INFO Successful handshake with dscalarm in 0.121s
[13:49:04][I][app:102]: ESPHome version 2023.11.3 compiled on Nov 22 2023, 13:30:54
[13:49:04][C][status_led:019]: Status LED:
[13:49:04][C][status_led:020]:   Pin: GPIO2
[13:49:04][C][wifi:559]: WiFi:
[13:49:04][C][wifi:391]:   Local MAC: 9C:9C:1F:C7:83:80
[13:49:04][C][wifi:396]:   SSID: 'D105B70'[redacted]
[13:49:04][C][wifi:397]:   IP Address: 192.168.1.31
[13:49:04][C][wifi:399]:   BSSID: 54:A0:50:6D:6C:59[redacted]
[13:49:04][C][wifi:400]:   Hostname: 'dscalarm'
[13:49:04][C][wifi:402]:   Signal strength: -47 dB ▂▄▆█
[13:49:04][C][wifi:406]:   Channel: 11
[13:49:04][C][wifi:407]:   Subnet: 255.255.255.0
[13:49:04][C][wifi:408]:   Gateway: 192.168.1.1
[13:49:04][C][wifi:409]:   DNS1: 192.168.1.1
[13:49:04][C][wifi:410]:   DNS2: 0.0.0.0
[13:49:04][C][logger:416]: Logger:
[13:49:04][C][logger:417]:   Level: DEBUG
[13:49:04][C][logger:418]:   Log Baud Rate: 115200
[13:49:04][C][logger:420]:   Hardware UART: UART0
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Door (z1)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'door'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Smoke (z2)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'smoke'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Motion (z3)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'motion'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Zone 4 (z4)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'door'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Zone 5 (z5)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'door'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Zone 6 (z6)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'door'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Zone 7 (z7)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'motion'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Partition 1 Ready (rdy_1)'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Partition 1 Armed (arm_1)'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Trouble Status (tr)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'problem'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'Battery Status (bat)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'problem'
[13:49:04][C][template.binary_sensor:028]: Template Binary Sensor 'AC Status (ac)'
[13:49:04][C][template.binary_sensor:028]:   Device Class: 'plug'
[... 95 lines cut ...]
[13:49:40][W][component:215]: Components should block for at most 20-30ms.
[13:49:41][I][debug:1030]: Setting Alarm state: D to partition 1
[13:49:54][I][Paneldata: :1097]: B1: B1 00 FF FF FF FF 00 00 00 00 AD 00 00 00 00 00 
[13:50:03][I][Paneldata: :1097]: C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 
[13:50:09][I][Paneldata: :1097]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[13:50:09][I][Moduledata::1097]: 11: FF 01 FF FC FF FF FF FF FF 03 00 00 00 00 00 00 
[13:50:24][I][Paneldata: :1097]: E6: E6 00 2C 10 00 00 00 00 22 00 00 00 00 00 00 00 
[13:50:33][I][Paneldata: :1097]: C3: C3 00 00 FF C2 00 00 00 00 00 00 00 00 00 00 00 
[13:50:39][I][Paneldata: :1097]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[13:50:39][I][Moduledata::1097]: 11: FF 01 FF FC FF FF FF FF FF 03 00 00 00 00 00 00 
[13:50:40][D][dht:048]: Got Temperature=21.8°C Humidity=21.9%
[13:50:40][D][sensor:094]: 'System Temperature': Sending state 21.80000 °C with 1 decimals of accuracy
[13:50:40][D][sensor:094]: 'System Humidity': Sending state 21.90000 % with 0 decimals of accuracy
[13:50:40][I][Paneldata: :1097]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[13:50:40][I][Paneldata: :1097]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00 
[13:50:40][I][info:1929]: status 01, last status 01,line2status 40,selection 01,partition=1,skip=0,force=1
[13:50:40][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[13:50:40][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[13:50:41][D][text_sensor:064]: 'System Status (ss)': Sending state 'online'
[13:50:41][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'disarmed'
[13:50:41][D][text_sensor:064]: 'zone status (zs)': Sending state ''
[13:50:41][D][text_sensor:064]: 'Trouble Msg (tr_msg)': Sending state 'BELL SUP '
[13:50:41][W][component:214]: Component custom_component took a long time for an operation (0.07 s).
[13:50:41][W][component:215]: Components should block for at most 20-30ms.
[13:50:54][I][Paneldata: :1097]: 27: 27 00 91 01 10 C7 00 90 00 00 00 00 00 00 00 00

feradelectronics, 2023-11-22:
[logs_dscalarm_logs.zip](https://github.com/Dilbert66/esphome-dsckeybus/files/13444312/logs_dscalarm_logs.zip)
here are the files

Dilbert66, 2023-11-22:
The dev version will not work with the yaml you are using now.  Everthing has been converted to external components.  You need to convert your setup to the new yaml format and recompile. You do not need to download any code anymore. Just the yaml itself. As to why it's not arming, I can't see anything in the logs as the command is not even getting to the panel from what I see.  The disarm cmd is there though.

feradelectronics, 2023-11-22:
i will try and confirm. thanks.

Dilbert66, 2023-11-22:
The system unknown error indicates a possible config issue with the keypad card itself and not the esphome component.  Let me know how you send your arming cmd normally.

feradelectronics, 2023-11-22:
I reprogrammed with the new dscalarm.yaml and configured as needed.  Thanks,.
Regarding the HA Alarm panel, still displays the same as originally mentioned above.  This is the standard alarm panel with HA.  Do i need to use the DSCKeybus alarm panel instead?

Dilbert66, 2023-11-22:
You do not need to change anything on the HA panel itself. That works with all code.

closed 2023-11-22 by feradelectronics

Dilbert66, 2023-11-22:
Good point! I will do that. Thanks!