Dilbert66/esphome-dsckeybus #109: Setup Issues

feradelectronics, 2023-11-23:
I have dsckeybus working for the most part.  

Have issue with ESP32 sending wrong time zone to update time.  We also use daylight savings during summer months.  How can i set the correct zone.

Thanks in advance for your help

Dilbert66, 2023-11-23:
See here: 
https://esphome.io/components/time/index.html

feradelectronics, 2023-11-23:
Thanks.
Sorry for another issue.  Is this a wifi connection problem or something else?

log:
[15:09:37][W][component:214]: Component dsc_alarm_panel took a long time for an operation (0.08 s).
[15:09:37][W][component:215]: Components should block for at most 20-30ms.
[15:09:37][I][Paneldata: :1102]: 10: 10 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 
[15:09:37][I][Paneldata: :1102]: 52: 52 00 40 64 40 64 81 64 40 05 00 00 00 00 00 00 
WARNING dscalarm: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for dscalarm
WARNING Disconnected from API
INFO Successfully connected to dscalarm in 0.006s
INFO Successful handshake with dscalarm in 0.130s

Dilbert66, 2023-11-23:
Hard to say. I would need to see the serial logs from the device itself to know.  Could be a reboot or a connection drop but no way of knowing from the dashboard logs as those only show it disconnected.

feradelectronics, 2023-11-23:
i will try to test further.  Thanks for your response.

feradelectronics, 2023-11-24:
will be trying WT32-Eth01 for stability.  Also have a customer with 4 partitions cannot seem to arm/disarm partitions 3 and 4.  
[dsc Config.zip](https://github.com/Dilbert66/esphome-dsckeybus/files/13461402/dsc.Config.zip)

Thanks.

Dilbert66, 2023-11-24:
could be different things.  DSC panel config, code , etc.  I need to see logs.

feradelectronics, 2023-11-24:
ok i will check that next week.  Thanks!

Enjoy the weekend

Dilbert66, 2023-11-24:
I went and did some testing on the extended partitions and I did find an issue with partitons higher than 2. Not sure if it would affect your issue as it does not affect arming or disarming but who knows. I will post when I push a new fix.

Dilbert66, 2023-11-25:
Ok, I've pushed an updated fix. Hopefully this will fix the issue.

feradelectronics, 2023-11-25:
I tried recompiling and get the following error:
Compiling .pioenvs/dscalarm/src/main.cpp.o
In file included from src/esphome.h:21,
                 from src/main.cpp:3:
src/esphome/components/dsc_alarm_panel/dscAlarm.h: In member function 'void esphome::alarm_panel::DSCkeybushome::processStatus()':
src/esphome/components/dsc_alarm_panel/dscAlarm.h:2568:31: error: no matching function for call to 'esphome::alarm_panel::DSCkeybushome::processProgramZones(int)'
         processProgramZones(30);
                               ^
src/esphome/components/dsc_alarm_panel/dscAlarm.h:2748:8: note: candidate: 'void esphome::alarm_panel::DSCkeybushome::processProgramZones(byte, byte)'
   void processProgramZones(byte startByte,byte zoneStart) {
        ^~~~~~~~~~~~~~~~~~~
src/esphome/components/dsc_alarm_panel/dscAlarm.h:2748:8: note:   candidate expects 2 arguments, 1 provided
*** [.pioenvs/dscalarm/src/main.cpp.o] Error 1

Dilbert66, 2023-11-25:
Should be good now.

feradelectronics, 2023-11-25:
It compiles as expected now.  I did not make any yaml changes on my end.  Still cannot see partitions 3 and 4. 
I get this message when looking at the panel 2:

This entity ('alarm_control_panel.safe_alarm_panel_2') does not have a unique ID, therefore its settings cannot be managed from the UI. See the [documentation](https://www.home-assistant.io/faq/unique_id) for more detail.

Panels 3 and 4 do not exist.
alarm_control_panel.safe_alarm_panel_3
alarm_control_panel.safe_alarm_panel_4

Hope this helps.  Thanks again.

Dilbert66, 2023-11-25:
That's a problem with your home assistant config. Not my code. I don't use the alarm control panel from ha so really can't assist much with that one.

feradelectronics, 2023-11-25:
i will try your control panel

feradelectronics, 2023-11-28:
still having issues. Are the following entities supposed to be created by dsckeybus for each partition?  
sensor.dscalarm_partition_3_status_ps_3
sensor.dscalarm_partition_4_status_ps_4

In your configuration.yaml you have: everything in alarm_control_panel:

> alarm_control_panel:
>   - platform: template
>     panels:
>       safe_alarm_panel:
>         name: "Alarm Partition 2"
>         value_template: "{{states('sensor.dscalarm_partition_2_status_ps_2')}}"
>         code_arm_required: true #to identify who armed system
>         
>         arm_away:
>           - service: esphome.dscalarm_alarm_arm_away
>         disarm:
>           - service: esphome.dscalarm_alarm_disarm
>             data_template:
>               code: '{{code}}'                    
>  ... (for each other partition1,3,4 respectively)

feradelectronics, 2023-11-28:
fyi, I have successfully configured to work with ws32_Eth01.  Connection very stable now.

I keep getting these messages in logs with both wifi and ethernet connections.
    [13:29:01][W][component:214]: Component dsc_alarm_panel took a long time for an operation (0.07 s).
    [13:29:02][W][component:215]: Components should block for at most 20-30ms.

Should i keep ignoring these or is there something else that needs to be done.

Dilbert66, 2023-11-30:
Yes, ignore those.  They are merely esphome warnings.

closed 2024-01-29 by Dilbert66