Dilbert66/esphome-dsckeybus #113: Received invalid alarm panel state: not_ready for entity alarm_control_panel.safe_alarm_panel

virus100b, 2023-12-09:
Hello, 

I have installed the last version of DSC alarm yaml code and i keep receiving the error message below everytime one zone sensor is triggered. 
I have runned previous version on esp8266 and now i have upgraded to ESP32. I have cleaned build files and reinstall the code with no luck. Except the log error all is working, the sensors, setting arm or disarm...
I am a beginner in HA so i don't have any experience with YAML code.

Logger: homeassistant.components.template.alarm_control_panel
Source: components/template/alarm_control_panel.py:235
Integration: Template (documentation, issues)
First occurred: 16:38:41 (7 occurrences)
Last logged: 16:39:42

Received invalid alarm panel state: not_ready for entity alarm_control_panel.safe_alarm_panel. Expected: armed_away, armed_custom_bypass, armed_home, armed_night, armed_vacation, arming, disarmed, pending, triggered, unavailable

Dilbert66, 2023-12-09:
Yes , I know the issue.  The HA alarm control panel wants to see "unavailable" as a status when the panel has open zones and not ready to be armed.  Because of some users complaining that seeing "unavailble" on a status was confusing them into thinking that they were not getting a status from the panel where it actually was.   So i changed to to not_ready and of course, it causes the issue above.  i think I'll rename it back to "unavailable" as it least works with the HA panel .

virus100b, 2023-12-09:
Ok. So we will need wait for the update. 
When do you think will be release the new version?

Thank you for the nice work you've done!

Dilbert66, 2023-12-09:
It's updated on main branch.