Dilbert66/esphome-dsckeybus #116: Error when updating to EsHome 12.1 <unicode string>:419:20: error: 'disconnectKeybus' is not a member of 'esphome::alarm_panel

olddawgpowers, 2023-12-21:
Log when trying to update-

Updating /config/esphome/dscalarm32.yaml
------------------------------------------------------------

INFO ESPHome 2023.12.1
INFO Reading configuration /config/esphome/dscalarm32.yaml...
INFO Updating https://github.com/Dilbert66/esphome-components.git@dev
INFO Detected timezone 'America/Chicago'
INFO Generating C++ source...
INFO Compiling app...
Processing dscalarm32 (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
Compiling .pioenvs/dscalarm32/src/main.cpp.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/IPAddress.cpp.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/IPv6Address.cpp.o
<unicode string>: In lambda function:
**<unicode string>:419:20: error: 'disconnectKeybus' is not a member of 'esphome::alarm_panel'**
<unicode string>:419:20: note: suggested alternative:
In file included from src/esphome.h:18,
                 from src/main.cpp:3:
src/esphome/components/dsc_alarm_panel/dscAlarm.h:48:13: note:   'disconnectKeybus'
 extern void disconnectKeybus();
             ^~~~~~~~~~~~~~~~
*** [.pioenvs/dscalarm32/src/main.cpp.o] Error 1
========================= [FAILED] Took 16.29 seconds =========================
========= [ERROR] /config/esphome/dscalarm32.yaml =========

Dilbert66, 2023-12-21:
Ok, Fixed. I will push a new updated version later anyhow but this version is working now.

olddawgpowers, 2023-12-21:
Thanks- Will I need to update my yaml file when you push the new version?

Dilbert66, 2023-12-21:
No changes needed. You can recompile now.

olddawgpowers, 2023-12-21:
I just tried to recompile and got the same error and failed.

Updating /config/esphome/dscalarm32.yaml
------------------------------------------------------------

INFO ESPHome 2023.12.1
INFO Reading configuration /config/esphome/dscalarm32.yaml...
INFO Detected timezone 'America/Chicago'
INFO Generating C++ source...
INFO Compiling app...
Processing dscalarm32 (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
Compiling .pioenvs/dscalarm32/src/main.cpp.o
<unicode string>: In lambda function:
<unicode string>:419:20: error: 'disconnectKeybus' is not a member of 'esphome::alarm_panel'
<unicode string>:419:20: note: suggested alternative:
In file included from src/esphome.h:18,
                 from src/main.cpp:3:
src/esphome/components/dsc_alarm_panel/dscAlarm.h:48:13: note:   'disconnectKeybus'
 extern void disconnectKeybus();
             ^~~~~~~~~~~~~~~~
Compiling .pioenvs/dscalarm32/FrameworkArduino/base64.cpp.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/cbuf.cpp.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/esp32-hal-adc.c.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/esp32-hal-bt.c.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/esp32-hal-cpu.c.o
Compiling .pioenvs/dscalarm32/FrameworkArduino/esp32-hal-dac.c.o
*** [.pioenvs/dscalarm32/src/main.cpp.o] Error 1
========================= [FAILED] Took 15.91 seconds =========================
========= [ERROR] /config/esphome/dscalarm32.yaml =========

Dilbert66, 2023-12-21:
find the refresh: line under external_components and change it to 30sec.  This is how often it checks for updates. You can change to 1h later but for now to force it. You might have to try a compile a couple times.

```
external_components:
  - source: github://Dilbert66/esphome-components@main #uncomment to use github repository
    components: [dsc_alarm_panel,binary_sensor,text_sensor] 
    refresh:  30sec
```

Dilbert66, 2023-12-21:
Ah crap. You didnt tell me you were using dev branch. hold on. I'll check that. I correct main.

Dilbert66, 2023-12-21:
Ok, I see the issue.  The sample yaml had a prepended namespace.  Remove the alarm_panel:: from in front of the disconnectKeybus() in the yaml lambda.

olddawgpowers, 2023-12-21:
Thanks- That did fixed it.  For the future, should I be using Master instead of Dev.  I switched to Dev sometime back, but don't remember why?

Dilbert66, 2023-12-21:
Dev is perfect

coolstuff99, 2023-12-29:
May this change make the default alarm panel in home assistant not working properly? It shows status unknown and disappeared armed away and armed stay.

Dilbert66, 2023-12-29:
No

closed 2024-01-29 by Dilbert66