Dilbert66/esphome-dsckeybus #119: Feature Request - Zone Names

tiaanv, 2024-01-24:
Awesome work.  Been using your component for a long time, and following all the updates, and improvements.

At the moment, if you look at the two main text sensors (that are used to populate the virtual keypad), then the one slight difference is that it is of course not able to show the zone names when looking at open zones etc.  
[image]

I do understand this.

I took a brief look at the code, and could not find a clear place where this text is generated.  I have a limited understanding at the moment of how it all hangs together.  I was wondering if it would be possible to retrieve the actual zone names from the yaml config of zones, and use the name, as configured in yaml

[image]

I am happy to look into this, and potentially contribute the code after some development effort, but would appreciate some initial guidance.

Dilbert66, 2024-01-24:
Good idea, I will add the feature to the dev branch.

tiaanv, 2024-01-24:
If I am not mistaken, then the actual change will be in the https://github.com/Dilbert66/esphome-components repo?

Dilbert66, 2024-01-24:
Yes, the change will be in the components repo in dev branch.  Just use this source line under the "external_components" section

- source: github://Dilbert66/esphome-components@dev

tiaanv, 2024-01-24:
Schweet.... FAST man...
Will give it a dry run and let you know.

tiaanv, 2024-01-24:
[image]
Happy dance!

tiaanv, 2024-01-24:
I haven't checked what impact this change (if any) will have on if you try to use the Bypass zone menus etc. 
From a front en perspective it works.  
[image]

I will verify on the actual alarm panel and see

Dilbert66, 2024-01-24:
Ya, the interface needs tweeking for a more pleasing display depending on the zone name lenghts.  Won't impact operation.

tiaanv, 2024-01-24:
I think this 100% satisfies my request.  I will look out for merge into main, then change back eventually.  Sidenote.  Let me know if you need any experimental stuff tested.

Dilbert66, 2024-01-24:
Good stuff! Sure, I'll merge to main soon enough. Tks for the offer on testing! Much appreciated.
I'm working on a few other components unrelated to the panel right now.  One is a telegram bot running totally on the esp  which includes notification and responding to telegram cmds.

tiaanv, 2024-01-24:
Interesting.  Curious as the application of running Telegram bot on ESP vs just implementing it on HA.

Dilbert66, 2024-01-24:
The concept is for using the ESPhome device completely standalone without a controlling server such as HA.  Let's say you want to give friends, parents access to having a virtual keypad, monitoring of their alarm system without having to also install a whole home monitoring system.  With the ESP having a web interface with the virtual keypad, the bot for notification of events, etc, it provides most of the needed features.  A telegram bot with remote command capability is handy for managing the alarm outside the home network.  I have something similar runngin with an arduino implementation but the code was cumbersome, so trying it out with the ESPhome platform

tiaanv, 2024-01-24:
Makes sense. Happy to play around with it and give you some feedback

closed 2024-01-29 by tiaanv