Dilbert66/esphome-dsckeybus #130: dsc_alarm_panel took a long time for an operation (113 ms).

DjSpidey, 2024-06-08:
Hi @Dilbert66 

I hope you can help me please.
I started this project a year ago and eventually got everything setup and working, with one issue i cant figure out.
I am not sure if this issue is related to signal quality, my Signal strength is -64 dB
So when i arm the alarm through HA it shows pending for about 30sec to 1 min, then only it shows Armed, so it's not instantly armed.

This below is the error i get in the logs:

          [W][component:237]: Component dsc_alarm_panel took a long time for an operation (113 ms).
          [W][component:238]: Components should block for at most 30ms.

Dilbert66, 2024-06-09:
That warning message is normal and unrelated to your issue. The delay between arming and showing armed is due to the programmed exit delay in your system. The system does not arm immediately as it allows time for entry/exit first and thus shows pending for that duration.  The exit delay is programmed in your dsc panel.

DjSpidey, 2024-06-09:
Hi, @Dilbert66 

I guess i have nothing to worry about then, I was just curious as i got a friend with the same setup, but his activates instantly.
I am posting my logs, if you can just have a look at them to see is everything is fine please.

INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/dsc-alarm.yaml...
INFO Updating https://github.com/Dilbert66/esphome-components.git@main
INFO Detected timezone 'Africa/Johannesburg'
INFO Starting log output from 10.0.0.111 using esphome API
INFO Successfully connected to dscalarm @ 10.0.0.111 in 0.073s
INFO Successful handshake with dscalarm @ 10.0.0.111 in 0.089s
[23:39:45][I][app:100]: ESPHome version 2024.5.5 compiled on Jun  5 2024, 20:03:53
[23:39:45][C][wifi:580]: WiFi:
[23:39:45][C][wifi:408]:   Local MAC: C4:DD:57:98:C2:7C
[23:39:45][C][wifi:413]:   SSID: 'DjSpidey 2.4Ghz'[redacted]
[23:39:45][C][wifi:416]:   IP Address: 10.0.0.111
[23:39:45][C][wifi:420]:   BSSID: 68:FF:7B:54:3A:B1[redacted]
[23:39:45][C][wifi:421]:   Hostname: 'dscalarm'
[23:39:45][C][wifi:423]:   Signal strength: -66 dB ▂▄▆█
[23:39:45][C][wifi:427]:   Channel: 10
[23:39:45][C][wifi:428]:   Subnet: 255.255.255.0
[23:39:45][C][wifi:429]:   Gateway: 10.0.0.2
[23:39:45][C][wifi:430]:   DNS1: 10.0.0.2
[23:39:45][C][wifi:431]:   DNS2: 10.0.0.2
[23:39:45][C][logger:185]: Logger:
[23:39:45][C][logger:186]:   Level: DEBUG
[23:39:45][C][logger:188]:   Log Baud Rate: 115200
[23:39:45][C][logger:189]:   Hardware UART: UART0
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'Front door sensor (z1)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'door'
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'Dining room motion (z2)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'motion'
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'Living room motion (z3)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'motion'
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'Back door sensor (z4)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'door'
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'PC Room Sliding door motion (z5)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'motion'
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'PC Room Sliding door sensor (z6)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'door'
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'Family room window RF (z7)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'window'
[23:39:45][C][template.binary_sensor:028]: Template Binary Sensor 'Main Bedroom motion (z8)'
[23:39:45][C][template.binary_sensor:028]:   Device Class: 'motion'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Main Bedroom Sliding door (z9)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'door'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Basement motion (z10)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'motion'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Master bedroom door motion (z11)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'motion'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Test sensor (z17)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'motion'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Partition 1 Ready (rdy_1)'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Partition 2 Ready (rdy_2)'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Partition 1 Armed (arm_1)'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Partition 2 Armed (arm_2)'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Trouble Status (tr)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'problem'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Battery Status (bat)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'problem'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'AC Status (ac)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'plug'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Fire partition 1 Status (fa_1)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'smoke'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'Fire partition 2 Status (fa_2)'
[23:39:46][C][template.binary_sensor:028]:   Device Class: 'smoke'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'partition 1 Alarm Status (al_1)'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'partition 2 Alarm Status (al_2)'
[23:39:46][C][template.binary_sensor:028]: Template Binary Sensor 'PGM 1 (r1)'
[... 646 lines cut ...]
[00:04:49][I][Paneldata: :1130]: 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 
[00:04:49][I][Paneldata: :1130]: 05: 05 00 81 01 00 C7 00 C7 00 C7 01 00 00 00 00 00 
[00:04:49][I][info:1955]: status 01, last status 02,line2status 3E,selection 01,partition=1,skip=0,force=0
[00:04:49][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[00:04:49][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[00:04:49][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state '01: Ready'
[... cut ...]
[00:05:25][D][text_sensor:064]: 'System Status (ss)': Sending state 'online'
[00:05:25][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'disarmed'
[00:05:25][D][text_sensor:064]: 'zone status (zs)': Sending state ''
[00:05:25][D][text_sensor:064]: 'Trouble Msg (tr_msg)': Sending state ''
[00:05:25][W][component:237]: Component dsc_alarm_panel took a long time for an operation (103 ms).
[00:05:25][W][component:238]: Components should block for at most 30 ms.
[00:05:37][I][Paneldata: :1130]: 16: 16 00 0E 42 F6 5C 00 00 00 00 00 00 00 00 00 00

DjSpidey, 2024-06-09:
its also showing pending even if i arm "HOME"

Dilbert66, 2024-06-09:
Well, those logs are not useful in the fact that they do not show the sequence of the arming process on your end. They are just normal open zone logs.  Please post a log where you are actually arming the system so we can see the data from that.

DjSpidey, 2024-06-09:
Do you know where i can find those logs?

Dilbert66, 2024-06-09:
Same type of logs as you show above but during the time that you are arming the panel.

Dilbert66, 2024-06-11:
I see no issues here.   You armed the system in stay mode, it went into the Exit delay for 30 seconds then armed as it should.  If you find Home assistant is not reporting correctly with the same time frame then it's an issue with your connection with HA and not this application. 
```
Arming started, system in exit delay so partitions status goes pending as it's fully armed yet.

[23:15:22][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Exit delay '
[23:15:22][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'in progress '
[23:15:22][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state '08: Exit delay'
[23:15:22][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'pending
```

```
After 30 seconds, system goes to armed. 
[23:15:52][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Armed: '
[23:15:52][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Stay '
[23:15:52][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state '04: Armed stay'
[23:15:52][D][binary_sensor:036]: 'Partition 1 Armed (arm_1)': Sending state ON
[23:15:52][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'armed_home'
```

Dilbert66, 2024-06-11:
If your friend's activates immediately then he might not have an exit delay or he is arming in Instant mode.

DjSpidey, 2024-06-23:
> I see no issues here. You armed the system in stay mode, it went into the Exit delay for 30 seconds then armed as it should. If you find Home assistant is not reporting correctly with the same time frame then it's an issue with your connection with HA and not this application.
> 
> ```
> Arming started, system in exit delay so partitions status goes pending as it's not fully armed yet.
> 
> [23:15:22][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Exit delay '
> [23:15:22][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'in progress '
> [23:15:22][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state '08: Exit delay'
> [23:15:22][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'pending
> ```
> 
> ```
> After 30 seconds, system goes to armed. 
> [23:15:52][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Armed: '
> [23:15:52][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Stay '
> [23:15:52][D][text_sensor:064]: 'Partition 1 Msg (msg_1)': Sending state '04: Armed stay'
> [23:15:52][D][binary_sensor:036]: 'Partition 1 Armed (arm_1)': Sending state ON
> [23:15:52][D][text_sensor:064]: 'Partition 1 Status (ps_1)': Sending state 'armed_home'
> ```

Thanks for the feedback @Dilbert66 

Confirmed with my friend he does not have an exit delay.
I just thought if i arm stay it would arm immediately, Thats why i thought someting is wrong.

Thanks for all your help @Dilbert66 it's appreciated.

closed 2024-11-01 by Dilbert66