Dilbert66/esphome-dsckeybus #131: Open gate through alarm wiring

kebasaa, 2024-06-13:
Hi
Thanks a lot for this ESPhome code. It seems that my gate is wired through the alarm system (I can currently use the same remote to open the gate and arm/disarm my alarm), and so is my intercom/bell. Do you have any idea how I can add 2 buttons for HA in the ESPhome YAML code, so I can:

1. pick up the door bell and 
2. open the gate 

Thanks

Dilbert66, 2024-06-13:
Sounds like your panel is using PGM outputs for those devices.   PGM devices and be activated by sending the cmds "*7x" where x is 1 to 4 . You can try those on the virtual keypad or send them from HA and see what happens.   The easiest way to find out for sure is to capture display log from the esphome device during the time you activate those controls with your remote then we can see what devices are activated. As far as buttons, you have choices. You can create a cmd on HA or if you do want to create a button on ESPHOME you can do that , I have to look up the best way though for that.

kebasaa, 2024-06-13:
Thanks. I'd actually really like to have it in ESPhome, simple because it will allow me to migrate the ESP to a new HA more easily. Let's figure it out, and maybe I can help add it to your yaml as an example.

Dilbert66, 2024-06-13:
Here is an example of a button in ESPHOME that sends a cmd to activate PGM1 on partition 1. You can send any keypad cmd.
```
button:
    platform: template
    name: Gate Button
    id: gate1
    icon: "mdi:emoticon-outline"    
    on_press:
      - lambda: |-
           $panelId->alarm_keypress_partition("*71",1);
```

kebasaa, 2024-06-13:
Thanks. I am actually struggling to install the virtual keypad. I used the file editor to upload the .js and .mp3 files, but there is no folder "config/www", and even after I add it and add the resource pointing to it, I can't add the card because it says that the alarm-keypad-card can't be found

Dilbert66, 2024-06-13:
make sure it's in /config/www.   The www directory goes in same directory where your homeassistant config files are.
Also whenever you update the js file make sure you update the ?id=xx parameter to force the browser to see it as a new file. It's a kludge to force the browser to reload it.   Also it's important to refresh your browser to reload it.

kebasaa, 2024-06-13:
Great, that worked. But I'm still getting messages like `Service esphome.dscalarm_alarm_keypress_partition not found.`

Dilbert66, 2024-06-13:
What did you call your esphome instance? The default in the yaml is "dscalrm" but if you called it something else, you will need to modify the service name to match yours.  You can search for the service name in the Developer tools , services  menu

kebasaa, 2024-06-13:
I did call it dscalarm:
`substitutions:
  name: "dscalarm"
  friendly_name: "Alarm System"`

kebasaa, 2024-06-13:
Home Assistant needed to be restarted to pick it all up. Works now, thank you

kebasaa, 2024-06-13:
Oddly, none of the codes worked. I tried to check the logs (nothing gets recorded on opening/closing the gate), and the *7x codes only give a long beep but the gate doesn't open

Dilbert66, 2024-06-13:
Sometimes the esp component needs to be refreshed from the ESPHOME integration in settings/devices and services.

What you can do is capture the esphome device log while you use your remote to open the gate and we can see if it will capture the commands sent.

kebasaa, 2024-06-13:
It appears to be wired into the panel directly. The same remote triggers the gate. There is a relay connected to the alarm panel that activates when I push the button, labelled ev-relay v.g. It seems to be wired to an expansion board, terminals Z4 and COM

Dilbert66, 2024-06-13:
What DSC model do you have? You might need to go into program sections 009-010 to see how your relay are programmed. I assume you have the installer code for your system. It looks like you have pc5204 board with external relays maybe.  I don't know why the actions don't show up in the log though..

Dilbert66, 2024-07-13:
Tks for the pics but more importantly I do need to see esphome logs of the activity from your panel when you send the gate open command using the remote.  This way I can see what data is being sent.  
As to your other questions, I would also need to see the logs for item 1 as well as the yaml config for that zone itself. as I can't guess as to what the issue would be.
For 2, 3 , sure that's all possible. You would just send the keystroke commands to the esphome alarm_keypress_partition service. 
For item 4, I would also need to see the logs when you send that command the panel. I'm not familiar with the long press sequence you describe so I prefer to see what is actually sent to the panel as that tells everything. 
For item 5, the firmware can only get the low battery statuses from the panel (trouble menu). It automatically gets thatb you can also request it from the keypad service by sending *21#7## and it will fetch it.

Dilbert66, 2024-07-13:
As to your gate motor. You indicate that it is connected to the z4 input of the 5108.  Since thats' a zone expander, that is an input so I suspect that is setup as a sensor to detect either the motor direction or gate open sensor?  There might be another connector to actually turn on the motor and I would think that would be either a relay output or a wireless switch somewhere.

Dilbert66, 2024-07-13:
Ok, I'm re-reading one of your comments. This one
```
t appears to be wired into the panel directly. The same remote triggers the gate. There is a relay connected to the alarm panel that activates when I push the button, labelled ev-relay v.g. It seems to be wired to an expansion board, terminals Z4 and COM
```

Can you clarify that this is a totally external relay , and not on the panel board? So reading what you are saying, is that this relay when active would potentially trigger z4 . Hmm.  I don't know why it would be done that way? Maybe using z4 as a trigger for another action?

Dilbert66, 2024-07-13:
Ok, perhaps I'm making the wrong assumption. I'm assuming that you are using a DSC keyfob what interacts with a RF5132 module which talks to the panel and can be seen on the logs.  There is also the possibility of using an external non DSC receiver that can be used to do the same without being interfaced to the panel bus.  I assume then it would use a zone input for sending on/off status back to the panel. Perhaps you have the secondary implementation.  For your panel to control such a device , you would need to wire up a PGM1 output to the external relay itself then you could control it with a *71 or *72 command. That's getting more complex but I think doable. Of course you would need to wire it in such a way as it would not interfere with your remote function such as using a diode in between.

kebasaa, 2024-07-15:
Thanks! I'll start with the esphome code. I just took your example file and modified it, but I wasn't sure what I could delete and what I shouldn't. I'll upload the log outputs later
[alarm.txt](https://github.com/user-attachments/files/16231163/alarm.txt)

kebasaa, 2024-07-19:
Sorry this took so long, I had to update everything to the new ESPhome version.
Here is the output when I open & close the gate with my remote. I am noticing this "[Moduledata::1130]: 11: " message, but it seems to happen 1-2s after I press the button, so may not be connected to the gate:

`
[17:35:03][D][text_sensor:064]: 'System Status': Sending state 'online'
[17:35:03][D][text_sensor:064]: 'Partition 1 Status': Sending state 'disarmed'
[17:35:04][D][text_sensor:064]: 'zone status': Sending state ''
[17:35:04][D][text_sensor:064]: 'Trouble Msg': Sending state ''
[17:35:04][W][component:237]: Component dsc_alarm_panel took a long time for an operation (106 ms).
[17:35:04][W][component:238]: Components should block for at most 30 ms.
[17:35:05][I][Paneldata: :1130]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00
[17:35:05][I][Moduledata::1130]: 11: FF 01 3F FF 3C F3 FF FF FF 03 00 00 00 00 00 00
[17:35:20][I][Paneldata: :1130]: E6: E6 00 09 00 EF 00 00 00 00 00 00 00 00 00 00 00
[17:35:35][I][Paneldata: :1130]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00
[17:35:35][I][Moduledata::1130]: 11: FF 01 3F FF 3C F3 FF FF FF 03 00 00 00 00 00 00
[17:35:50][I][Paneldata: :1130]: A5: A5 00 24 1E 71 90 00 00 E8 00 00 00 00 00 00 00
[17:35:50][I][Paneldata: :1130]: A5: A5 00 24 1E 71 90 00 00 E8 00 00 00 00 00 00 00
[17:36:03][I][Paneldata: :1130]: A5: A5 00 24 1E 71 90 00 00 E8 00 00 00 00 00 00 00
[17:36:03][I][Paneldata: :1130]: A5: A5 00 24 1E 71 90 00 00 E8 00 00 00 00 00 00 00
[17:36:03][I][info:1955]: status 01, last status 01,line2status 00,selection 01,partition=1,skip=0,force=1
[17:36:03][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[17:36:03][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[17:36:03][I][info:1955]: status 03, last status 03,line2status 00,selection 01,partition=2,skip=0,force=1
[17:36:03][D][text_sensor:064]: 'System Status': Sending state 'online'
[17:36:03][D][text_sensor:064]: 'Partition 1 Status': Sending state 'disarmed'
[17:36:04][D][text_sensor:064]: 'zone status': Sending state ''
[17:36:04][D][text_sensor:064]: 'Trouble Msg': Sending state ''
[17:36:04][W][component:237]: Component dsc_alarm_panel took a long time for an operation (104 ms).
[17:36:04][W][component:238]: Components should block for at most 30 ms.
[17:36:05][I][Paneldata: :1130]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00
[17:36:05][I][Moduledata::1130]: 11: FF 01 3F FF 3C F3 FF FF FF 03 00 00 00 00 00 00
[17:36:20][I][Paneldata: :1130]: E6: E6 00 2C 40 00 00 00 00 52 00 00 00 00 00 00 00
[17:36:35][I][Paneldata: :1130]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00
[17:36:35][I][Moduledata::1130]: 11: FF 01 3F FF 3C F3 FF FF FF 03 00 00 00 00 00 00
[17:36:50][I][Paneldata: :1130]: 16: 16 00 0E 32 B6 0C 00 00 00 00 00 00 00 00 00 00
[17:37:03][I][Paneldata: :1130]: 16: 16 00 0E 32 B6 0C 00 00 00 00 00 00 00 00 00 00
[17:37:03][I][Paneldata: :1130]: 16: 16 00 0E 32 B6 0C 00 00 00 00 00 00 00 00 00 00
[17:37:03][I][info:1955]: status 01, last status 01,line2status 00,selection 01,partition=1,skip=0,force=1
[17:37:03][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[17:37:03][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[17:37:03][I][info:1955]: status 03, last status 03,line2status 00,selection 01,partition=2,skip=0,force=1
[17:37:03][D][text_sensor:064]: 'System Status': Sending state 'online'
[17:37:03][D][text_sensor:064]: 'Partition 1 Status': Sending state 'disarmed'
[17:37:04][D][text_sensor:064]: 'zone status': Sending state ''
[17:37:04][D][text_sensor:064]: 'Trouble Msg': Sending state ''
[17:37:04][W][component:237]: Component dsc_alarm_panel took a long time for an operation (105 ms).
[17:37:04][W][component:238]: Components should block for at most 30 ms.
[17:37:05][I][Paneldata: :1130]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00
[17:37:05][I][Moduledata::1130]: 11: FF 01 3F FF 3C F3 FF FF FF 03 00 00 00 00 00 00
[17:37:20][I][Paneldata: :1130]: E6: E6 00 2C 80 00 00 00 00 92 00 00 00 00 00 00 00
[17:37:35][I][Paneldata: :1130]: 11: 11 00 AA AA AA AA AA AA AA 02 00 00 00 00 00 00
[17:37:35][I][Moduledata::1130]: 11: FF 01 3F FF 3C F3 FF FF FF 03 00 00 00 00 00 00
`

Dilbert66, 2024-07-20:
I don't see anything that could represent any RF or relay activity from the log.  The  module data response you indicate is a normal supervision status update.  It seems your remote is not a DSC component as I should see some status in the log.  I'm not that familiar with all the possible configurations for this type of setup so I can't really say for sure how it would work. I suppose, you could control your gate with the dsc system using one of the relays, but you'd have to wire it up and program the output config then you could send a *71 to open/close.  I don't know if there is a sensor back but that could be wired back to your panel using a zone, but again, not my area of expertise :)

closed 2025-01-12 by Dilbert66