Dilbert66/esphome-dsckeybus #132: Failed config

DjSpidey, 2024-06-23:
@Dilbert66  Wonder if you could help with this issue, only doing this after an update of ESPHome was released recently.

Alarm still works fine, but i cant update the ESP32 device.

INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/dsc-alarm.yaml...
Failed config

ota.unknown: [source /config/esphome/dsc-alarm.yaml:115]
  
  'ota' requires a 'platform' key but it was not specified.
  password: 61e5f17f2ee0703be384ff42f3d8d572
  safe_mode: True
  on_begin: 
    switch.turn_off: connection_status_switch

Dilbert66, 2024-06-23:
Add this line to your ota section and put safe mode: on it's own separately from ota. Read the esphome docs about the new changes.
   platform: esphome

brianj74, 2024-06-25:
safe_mode:
  disabled : False

DjSpidey, 2024-06-25:
Thanks @Dilbert66  and @brianj74 

It's up and running again

closed 2024-11-01 by Dilbert66