Dilbert66/esphome-dsckeybus #136: Trouble Messages not Displaying

dotslashsuperstar, 2024-08-08:
I just connected my old alarm system and I think I may have wired my isolated version up wrong because I don't think I am able to write to the system.  I have an ESP32, PC1555, PC5508 and PC5132.  I used a LD1177 3.3 Voltage Regulator and tied the Black to ground since usually I tie the grounds together.  Why are the grounds separated?  Is there a difference between the Isolated version and Alternative ground isolated version?  They look the same.  I also used a 180 ohm resistor for R1, could that be why?  The battery status is also not displaying correctly, it displays 'ok' but has a trouble condition.  All my key presses seem to have no effect.

Here's a log...
```
[08:38:18][I][Debug:997]: Writing keys: * to partition 1, partition disabled: 0 , partition locked: 0
[08:38:18][I][info:1955]: status 03, last status 03,line2status 04,selection 01,partition=1,skip=0,force=1
[08:38:18][D][text_sensor:064]: 'Line 1 Partition 1': Sending state 'Secure System'
[08:38:18][D][text_sensor:064]: 'Line 2 Partition 1': Sending state 'Before Arming <>'
[08:38:19][I][Debug:997]: Writing keys: 2 to partition 1, partition disabled: 0 , partition locked: 0
[08:38:19][I][info:1955]: status 03, last status 03,line2status 04,selection 01,partition=1,skip=0,force=1
[08:38:19][D][text_sensor:064]: 'Line 1 Partition 1': Sending state 'Secure System'
[08:38:19][D][text_sensor:064]: 'Line 2 Partition 1': Sending state 'Before Arming <>'
[08:38:25][I][Paneldata: :1130]: 64: 64 00 04 68 00 00 00 00 00 00 00 00 00 00 00 00 
[08:38:25][D][text_sensor:064]: 'beeps': Sending state '2'
[08:38:25][I][Paneldata: :1130]: 64: 64 00 04 68 00 00 00 00 00 00 00 00 00 00 00 00 
[08:38:25][I][info:1955]: status 03, last status 03,line2status 04,selection 01,partition=1,skip=0,force=0
[08:38:25][D][text_sensor:064]: 'Line 1 Partition 1': Sending state 'Secure System'
[08:38:25][D][text_sensor:064]: 'Line 2 Partition 1': Sending state 'Before Arming <>'
[08:38:27][D][text_sensor:064]: 'beeps': Sending state '0'
[08:38:29][I][Paneldata: :1130]: 11: 11 00 AA AA AA AA AA 2A 00 00 00 00 00 00 00 00 
[08:38:29][I][Moduledata::1130]: 11: FF 01 3F FF FF F3 FF 3F 00 00 00 00 00 00 00 00 
[08:38:35][I][Paneldata: :1130]: 64: 64 00 04 68 00 00 00 00 00 00 00 00 00 00 00 00 
[08:38:35][D][text_sensor:064]: 'beeps': Sending state '2'
[08:38:35][I][Paneldata: :1130]: 64: 64 00 04 68 00 00 00 00 00 00 00 00 00 00 00 00 
[08:38:35][I][info:1955]: status 03, last status 03,line2status 04,selection 01,partition=1,skip=0,force=0
[08:38:35][D][text_sensor:064]: 'Line 1 Partition 1': Sending state 'Secure System'
[08:38:35][D][text_sensor:064]: 'Line 2 Partition 1': Sending state 'Before Arming <>'
[08:38:35][W][component:237]: Component dsc_alarm_panel took a long time for an operation (52 ms).
[08:38:35][W][component:238]: Components should block for at most 30 ms.
[08:38:37][D][text_sensor:064]: 'beeps': Sending state '0'
```
Overall everything appears to be working except writing to the panel.

I also had to add height: auto to keypad_display in the js because my themes font is big and line2 was displaying behind the keypad.

Dilbert66, 2024-08-08:
The master readme is incorrect. There is only the one ground isolated version. You might want to use the "dev" branch version instead as it shows some other curcuit alternatives and updated code.   The grounds are separated for the "isolated" version as some users preferred that type of circuit when connecting to the panel with the perception that it will be safer to the panel due to it's physical signal line isolation so I provided it. In my experience it does not perform as well as the other versions. 

For your battery sensor issue, that might be a problem with the "master" branch code. I suggest again that you use the "dev" branch version and see if that solves your issue. I will be migrating the "dev" branch to "master" in the near future.

As to your send problem, well that could be a number of things. Usually it's a wiring problem. Also make sure the pin you used for the output is setup correctly in your yaml. 180ohm is fine for r1.

closed 2024-08-08 by dotslashsuperstar

Dilbert66, 2024-08-08:
The reason you are not seeing the Bat and other trouble statuses is because your esp can't send the cmds to the panel requesting those statuses.  If you go on your physical panel and goto into the service menu, you will see the status update on HA.  Once you fix the send issue, that should take care of that.