Dilbert66/esphome-dsckeybus #138: Virtual Keypad not workin on PC858H

Postnasal8, 2024-08-26:
I created a prototype on a breadboard, It can read any data regarding alarm status (door sensors, motion detectors, keypad button..etc).
However after everything I tried, virtual keypad is not working.

[image]
Every component on the board is just as the schematic. R2 and R4 on the picture is 10K, but I also tried with 36K (I didn't had a 33K on-hand, also it should be fine). Also I'm using a KC637 not a PN2222A tranzistor, but afaik it should be good.

[image]

Can someone help figure out what could be the issue?

Dilbert66, 2024-08-26:
It could be many things. Bad wiring, bad config, etc. Without knowing your config and it's impossible to know but looking at your wiring though, I can see that you have the transistor collector going to the esp 3.3 volt side.  It should be going to the panel 12 volt side of the resistor.

Postnasal8, 2024-08-26:
Oh my days...good catch. Tested it right away, virtual keypad working just great.

Dilbert66, 2024-08-26:
👍

closed 2024-11-01 by Dilbert66