Dilbert66/esphome-dsckeybus #139: Unavailable "Arm Home" option

mgiamberardino, 2024-09-12:
Hi @Dilbert66 

I built the board a few weeks ago. I'm having all the sensors and can manage the alarm from HA. Great work!

But I'm having an small issue that I'm kind of solving but not entirely and I'd like to get it right.

If I use the alarm_control_panel created when adding the esphome device as the source of my alarm_control_panel card in my dashboard I'm only able to arm away but arm_home is not available to select as option.

https://github.com/user-attachments/assets/8c6b721c-26e4-40f5-b1fd-f2d2487a9315

"Armar ausente" is "Arm Away"
"Armar en casa" is "Arm Home"

The only way I can Arm Home is by creating a new alarm control panel in my configuration.yaml:
```yaml
alarm_control_panel:
  - platform: template
    panels:
      safe_alarm_panel:
        name: "Alarm Panel"
        unique_id: alarm_custom_1
        value_template: "{{ states('sensor.dsc_alarm_partition_1_status_ps_1') }}"
        code_arm_required: false
        
        arm_away:
          - service: esphome.dsc_alarm_alarm_arm_away
                  
        arm_home:
          - service: esphome.dsc_alarm_alarm_arm_home
          
        arm_night:
          - service: esphome.dsc_alarm_alarm_arm_night
            data_template:
              code: '{{code}}' #if you didnt set it in the yaml, then send the code here
          
        disarm:
          - service: esphome.dsc_alarm_alarm_disarm
            data_template:
              code: '{{code}}' 
```

This works and now I can arm home and arm away. And also I'm able to disarm. But I still use `dsc_alarm_alarm_panel` entities for sensor status.
The only problem is that when `ps_1` gets unavailable is messing around with the `safe_alarm_panel` status.

This is my esphome yaml config:
```yaml
esphome:
  name: dsc-alarm
  friendly_name: DSC Alarm

esp8266:
  board: esp01_1m

# Enable Home Assistant API
api:
  encryption:
    key: ---

ota:
  - platform: esphome
    password: ---

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Dsc-Alarm Fallback Hotspot"
    password: ---

[... 316 lines cut ...]
    name: "Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();

  - platform: restart
    id: restart_switch
    
  - platform: safe_mode
    name: "Safe Mode" 

alarm_control_panel:
  - platform: template
    name: "Alarm Panel"
    codes: 
      - ---
```

Thanks

Dilbert66, 2024-09-12:
Ok, the first issue is that you must have an older configuration for an alarm panel in your yaml since you cannot simply add an esp device to alarm panel which is why it works when you add a new correct one. You might want to make sure there isnt another there.  

As to the unavailable status for ps_1, I suggest you use the "dev" branch code instead of the main.   There are many enhancements there including the ps_1 status to show "disarmed" instead of unavalable when zones are open but not armed.

mgiamberardino, 2024-09-12:
> Ok, the first issue is that you must have an older configuration for an alarm panel in your yaml since you cannot simply add an esp device to an alarm panel.

Ohh. I see. Could you point me to some newer example of configuration?

> As to the unavailable status for ps_1, I suggest you use the "dev" branch code instead of the main. There are many enhancements there including the ps_1 status to show "disarmed" instead of unavalable when zones are open but not armed.

Oh. Ok. I'll give it a try.

Thanks for the quick response!

Dilbert66, 2024-09-12:
Sorry, I'm confused. you already have a working configuration since you indicated that when you added the displayed config in your yaml, it worked ok?

mgiamberardino, 2024-09-12:
Yes. The yaml I shared is working. I think I found the issue. I was adding an `alarm_control_panel` item in the esphome configuration and if I'm not wrong what you're suggesting is just add the the component in the `configuration.yaml` file, right?

Without adding the alarm_control_panel in the configuration.yaml and just using the one from esphome configuration is not allowing me to arm home.

BTW. Is the `dev` branch generating a bigger build? When switching to esphome_components@dev is failing to put it into the device memory.

Dilbert66, 2024-09-12:
Yes, do not use the "alarm_control_panel" in the esphome config. It will not work.  Also, yes, the dev branch is larger.  Try it without the "alarm_control_panel". If it still doesnt work, then you can go back to using the main branch or use an ESP32 board instead, which is much more powerful.

mgiamberardino, 2024-09-13:
Thanks man! Thanks for que quick response and kudos for the project.

Bought you a coffee. Enjoy it!

closed 2024-09-13 by mgiamberardino