Dilbert66/esphome-dsckeybus #142: Mosfet level converter circuit is wrong

go2net, 2024-10-01:
You should switch the drain and source connections for proper operation. The high-level (12V logic) pin should be connected to the drain (D) pin, and the low-level (3.3V logic) pin should be connected to the source (S) pin.

Dilbert66, 2024-10-01:
The mosfet convert drawing is correct. High level to Source and low level to Drain.
https://circuitdigest.com/tutorial/bi-directional-logic-level-controller-using-mosfet

Dilbert66, 2024-10-01:
Ok, I'm confused here.  What drawiing are you looking at? That's what I have.  High side to drain and low side to to source?

go2net, 2024-10-01:
I may have submitted to the wrong repository. After checking again, I couldn't find the MOSFET converter circuit. Let me verify if I need to submit it to another repository.

go2net, 2024-10-01:
The link (https://github.com/Dilbert66/esphome-dsckeybus/tree/dev) is still under your repository which have a wrong schematic, but it's on the dev branch, not the master branch. Google directed me to your dev branch.

go2net, 2024-10-01:
Not an issue with original circuit. it was just too small, and I misread it.

closed 2024-10-01 by go2net