Dilbert66/esphome-dsckeybus #147: Having "became unavailable" status occur which i think is throwing me some issues

evolv30101, 2024-11-22:
So i've been trying to get the user name from the evt data but noticed out of 8 state changes i only see 1 logged entry for the user who actually disarmed the alarm in the logbook.

While troubleshooting this i noticed i have quite a bit of "became unavailable" status showing up.  I havent fully correlated everything yet but it seems to be happening when changing partition states.  Ie from armed to disarmed.  

I see an entries:  
        arming ->2min later armed_away -> 21 secs later became unavailable -> 1 sec later armed_away

        pending ->disarmed -> became unavailable ->disarmed

Most of the state changes resemble this.  I only have 1 that did not, which is the one that actually logged the user correctly.  Im guessing when its going unavailable is when im "losing" the event data which is why i dont see them logged.  

Is there some kind of timing issue or something going on here?  

Not sure if its related but i dont recall seeing these unavailable messages on the main branch before i switched over to dev.  But i could be mis-correlating here.   Just thought id throw that tidbit there incase it helps think of anything.

I can grab logs from an arm / disarm when someone arrives there today if you would like.

Dilbert66, 2024-11-22:
Yes, post esphome logs so we can see what is happening with an arm/disarm.  That will show all that is needed to know otherwise, it's all a guess.

Dilbert66, 2024-11-22:
Not sure when was the last time you compiled the esp, but you might want to do a re-install on it.  A lot of fixes having gone into it in a while.  There was also a change of an partition status from "unavailable" to not_ready so if your version is older, that might be what you are seeing.

evolv30101, 2024-11-22:
Will do.  I think someone should be there in about an hour or so and have to disarm.  

I was reading thru some of the older issues and came across something somewhat similar and they did a clean build which seem to help some.  I just went ahead and did that because im sure i did not when i switched to dev.

evolv30101, 2024-11-22:
[DSC_Epic_log_112224.txt](https://github.com/user-attachments/files/17870504/DSC_Epic_log_112224.txt)

Here you go.  I tried to clean up the pre-event noise.  If you need that i can upload the full file.  

I see:  WARNING alarm-epic @ 172.16.8.82: Connection error occurred: [Errno 104] Connection reset by peer
Guessing this has something to do with.

Dilbert66, 2024-11-22:
Yes, that connection reset is suspicious.  Might be a restart on the esp. I'll do some testing on my end to see if I can duplicate it.  If I need more logs, I'll post back..

evolv30101, 2024-11-22:
Im updating esphome as well its a couple months behind.  

I see lots of people with this issue but it seems a split bag of networking and possibly some lambda statements (over my head).  The networking side seems to be people renewing dhcp too often which should not be the case for me.  Some had dhcp renewals at 1 min, lol.  I have set for a week.  For that its also strange that it would "magically" occur right in the middle of a disarm/arm.  

I'll keep digging see if i find anything else.

evolv30101, 2024-11-22:
I can also trigger the reset going into *5 menu apparently too from the virtual keypad card.
[DSC_Epic_log_Star5_112224.txt](https://github.com/user-attachments/files/17871127/DSC_Epic_log_Star5_112224.txt)

If i hit the <> i can still manage to get to the user codes and scroll thru and change.

Dilbert66, 2024-11-22:
Can you show me the values you have in the yaml for the usercodes: section.
NOTE: put x's in place of the digits so we dont' see your codes.

evolv30101, 2024-11-22:
Here is the abbreviation....i have like 30 people in there. 

I dont have the actual 4-digit codes, just the "slot numbers".  Is this wrong?  I thought this was the way it wanted it as it does show the correct person for my one working entry in the log book where it didnt reset. 

Most a full name, ie: "Person One" 
But a couple are only a single name...."Jacob"   (working with what i was given, lol)

wonder if the full vs short name has something to do with it?  Jacob was the only one that didnt reset it.  

dsc_alarm_panel:
  id: $panelId
  #you can enter a list of user codes mapped to their names for display in armed/disarmed event messages
  usercodes: "1:Larry Someone,2:Thom Person,3:Jacob"

evolv30101, 2024-11-22:
If it helps i can pull "jacob" (single name) code from the keypad and relay it to someone on site and have them arm/disarm and see what happens.  May be a couple hrs as my onsite "hands" is putting out emergencies there at the moment....leaking toilets, etc.. fun day :)

Dilbert66, 2024-11-22:
Yup, that's the problem. My code did not handle the "spaces" correctly and yes you are correct, it's just the id, not the code. My mistake. I'll fix the code and post an updated fix!

evolv30101, 2024-11-22:
Sweet.  Sorry...theres a lot of people on here.  Some are family too so makes it even more fun.

I guess i could maybe "first.lastname" if the space becomes an issue.

evolv30101, 2024-11-22:
Uploaded.

Whohoo!  i still need to arm/disarm but the *5 menu doesnt cause the reset at least.  I'll text my onsite guy and have him arm/disarm when he gets a break and let you know.

evolv30101, 2024-11-23:
Just to check back in....all looking great!

Setup a logbook card and my log file looks beautiful.  

No more random disconnects clicking thru virtual panel either

closed 2024-11-23 by evolv30101