Dilbert66/esphome-dsckeybus #155: Cannot connect to DSC PC1616

adydeac, 2024-12-23:
Hi,

I have used the example yaml from the repository as starting point.
I have made the isolated version interface.
Still, the module is not connecting. The partitions line 1 stays "ESP module start".
I've set the dscdebuglevel at 3, but nothing shows up on the debug console, except for when I press a button on the HomeAssistant card:
[13:54:59][I][Debug:997]: Writing keys: 1 to partition 1, partition disabled: 1 , partition locked: 0
[13:55:00][I][Debug:997]: Writing keys: 1 to partition 2, partition disabled: 1 , partition locked: 0

Also, the connection switch stays off.

The DSC alarm is brand new, with default configuration, nothing set. The physical keyboard is working fine (at least is showing the zone and system statuses).

I've added some debug lines in dscAlarm.h in the loop/update function (around line 1466) and it is triggering.
Still, I am unable to debug the dsc.loop() to check if it is triggering.

What am I missing?

Thank you,
Adrian

Dilbert66, 2024-12-23:
This tells me that your connection to the panel is not working. Check your wiring and circuit.

adydeac, 2024-12-23:
OK, I see. Will try to add some debug messages to point others in this direction and will send the diff.
Also started to do some code layout customization. :)

Dilbert66, 2024-12-23:
There is already debug messages in the code.  Your issue is not the code but your wiring to the panel. If you don't see traffic, it's your connection.

adydeac, 2024-12-23:
OK, there is no traffic, but there is also no message saying there is no traffic, the communication timed out, check your wiring, bla, bla, bla...

Dilbert66, 2024-12-23:
I would suggest you use the "dev" branch. The master branch will be replaced soon.

adydeac, 2024-12-23:
You were right, I had an pin solder joint short circuiting one of the lines. Module started just fine!
Since it is a new instalation, I will give the dev branch a go.

Thank you!

Keep up the good work!

Dilbert66, 2024-12-23:
FYI, i've added an error message on dev branch whenever the system detects that no data is available from the panel