Dilbert66/esphome-dsckeybus #157: No data from S2-Saola

Ssobarzo, 2025-01-19:
Hi Dilbert,
I made an install from de dev branch on ESPHome 2024.12.4 and despite the system is online and connected, there is no data flowing. I triple check the connections and there is no issues, even checking with an osciloscope directly on the dev board pins. The log shows only this over and over:

```
[23:47:44][I][Paneldata: :1130]: FF: FF 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 
[23:47:44][I][Paneldata: :1130]: FF: FF 01 FF FF FF FF FF FF FF FF 01 00 00 00 00 00 
[23:47:44][D][text_sensor:064]: 'System Status': Sending state 'online'
[23:47:44][D][text_sensor:064]: 'Trouble Message': Sending state ''
[23:47:44][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[23:47:44][D][esp32.preferences:114]: Saving 1 preferences to flash...
[23:47:44][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[23:48:30][D][sensor:094]: 'WiFi Signal dB': Sending state -62.00000 dBm with 0 decimals of accuracy
[23:48:40][D][sensor:094]: 'Uptime': Sending state 116.69500 s with 0 decimals of accuracy
```

I also tested it disconnected from the alarm and it says it's offline and disconnected. The last event is always ESP module start. ¿any ideas where else to look?

Dilbert66, 2025-01-19:
This could be for many reasons.  What alarm system is this connected to?  Your wiring  could be connected to the wrong pins or you have the wrong pins configured, or you are using the interface on a non supported alarm system.

Ssobarzo, 2025-01-20:
Hi again. Just to be sure of the pin numbers, I interchanged them and the system started to work. So, looking what was the difference, I found the GPIO 19 and 20, are used for USB. In ESPHome, the default logging ouput is sent to USB CDC, whichmust be disabled in order to use this pins.
Since I had the PCB ready, it was not easy to just change the pins. Logger components indicates you can disable the logging with:

`baud_rate: 0`

but it didn't work. What finally works was a reroute of the output with:

`logger:
  level: DEBUG
  hardware_uart: UART0`

At the end, pins must be carefully selected not just to avoid spurious signals, but to avoid conflict with some default or well known configuration using those pins. I think this could be added to the README to warn others.

Thanks a lot for your help and wonderful work.

closed 2025-09-07 by Dilbert66