Dilbert66/esphome-dsckeybus #158: Issue when arming

jokkling, 2025-01-27:
I get this error when arming in any mode:

> Action esphome.larmintegration_alarm_arm_home not found.

Does it have to do with the renaming frpm service to actions or something? Or is it a local error for me?

Dilbert66, 2025-01-27:
Looks like you are using the wrong service name. Find the correct one in the home asssistant developer tools section.

jokkling, 2025-01-28:
Thank you for answering. This error is from the Developer tool, Actions section. I will try delete and re-add the esp project in HA.

jet202020, 2025-05-21:
Hi. Dulbert,
following your instruction and codes, the dashboard DSC Panel is working partially.  The key are responding, but there is not message display or visual signs on it. May you have a suggestion or clue, what i am doing bad?

jokkling, 2025-08-03:
> Hi,
> 
> I am having same problem since latest updates from esphome. Unable to arm "Action esphome.larmintegration_alarm_arm_home not found."
> 
> Using dashboard default alarm panel and developer tools. [@jokkling](https://github.com/jokkling) did you solve the issue and / or remember how you did it?
> 
> Thank you,

If i remember correctly I just removed the device from home assistant and added it again.

coolstuff99, 2025-08-03:
Thanks @jokkling for reply but unfortunately did not solved the problem.
However, I took a radical approach and install new instance of home assistant and added the esphome device. To my surprise, the default "alarm panel" is no longer finding the dscalarm one anymore (although it can be added in my original home assistant, maybe because some unremoved data when deleted and added...).
This confirms the real behaviour of esphome action not found... bc indeed is not available.

[image]

[image]

I re-compared and checked my yaml code to the original from the repository https://github.com/Dilbert66/esphome-dsckeybus/blob/master/DscAlarm.yaml and looks good.

And while writing this reply I found the problem!
**I had to change in the configuration.yaml from "service:" to "action:"**

```
alarm_control_panel:
  - platform: template
    panels:
      safe_alarm_panel:
        name: "Alarm Panel"
        value_template: "{{states('sensor.dscalarm_partition_1_status')}}"
        code_arm_required: false
        
        arm_away:
          - service: esphome.dscalarm_alarm_arm_away
                  
        arm_home:
          - service: esphome.dscalarm_alarm_arm_home
          
        arm_night:
          - service: esphome.dscalarm_alarm_arm_night
            data_template:
              code: '{{code}}' #if you didnt set it in the yaml, then send the code here
          
        disarm:
          - service: esphome.dscalarm_alarm_disarm
            data_template:
              code: '{{code}}'                    

```

@Dilbert66 you may need to update the Readme to add this change that I guess now it is mandatory from now on on current versions of esphome.

Thank you!

Dilbert66, 2025-08-03:
@coolstuff99 great work finding that. I haven't had a chance to look at the new HA changes related to services. Yes, I'll amend the docs. Thanks!

Dilbert66, 2025-08-26:
Ensure you have the custom_services line in the api section.

api:
   encryption:
      key: !secret encryption_key
   custom_services: true

closed 2025-09-07 by Dilbert66