Dilbert66/esphome-dsckeybus #167: System Status rapidly oscillating between disarmed and not ready

lukemfarber, 2025-04-06:
I recently installed this on an ESP8266 NodeMCU. I used the recommended pins and wiring, but substituted 10K+22K resistors for the recommended 33K since I didn’t have one on hand. Any ideas as to why this might be occurring? I have tried reinstalling firmware but no dice. When I look at the logs from the ESP it is a constant stream of characters and codes, which I have included a bit of below. Any help would be appreciated. Thanks!

`08:08:41][I][Paneldata: :1130]: C5: C5 00 F9 E1 7F E7 10 FF 11 E7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:41][I][Paneldata: :1130]: 05: 05 01 F1 F1 10 C7 10 C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: 05 01 F1 F1 10 C7 10 C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: 05 00 91 7F 10 C7 10 C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: 05 00 91 7F 10 C7 10 C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 7F, last status F1,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state '127'
[08:08:42][D][text_sensor:064]: 'Message': Sending state '7F: Unknown'
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 03](tel:05 00 91 03) D0 FF 3F C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 03](tel:05 00 91 03) D0 FF 3F C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 03, last status 7F,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Secure System'
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Before Arming <>'
[08:08:42][D][text_sensor:064]: 'Message': Sending state '03: Zones open'
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 05](tel:05 00 91 05) 1F FF 11 FF FC C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 05](tel:05 00 91 05) 1F FF 11 FF FC C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 05, last status 03,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Armed:       '
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Away            '
[08:08:42][D][text_sensor:064]: 'Message': Sending state '05: Armed away'
[08:08:42][D][binary_sensor:036]: 'Partition 1 Armed': Sending state ON
[08:08:42][D][text_sensor:064]: 'Security Status': Sending state 'armed_away'
[08:08:42][W][component:237]: Component dsc_alarm_panel took a long time for an operation (53 ms).
[08:08:42][W][component:238]: Components should block for at most 30 ms.
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 01](tel:05 00 91 01) F0 CF 13 FF 1F C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 01](tel:05 00 91 01) F0 CF 13 FF 1F C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 01, last status 05,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[08:08:42][I][info:1955]: status CF, last status FF,line2status 00,selection 00,partition=2,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'Message': Sending state '01: Ready'
[08:08:42][D][binary_sensor:036]: 'Partition 1 Armed': Sending state OFF
[08:08:42][D][text_sensor:064]: 'Security Status': Sending state 'disarmed'
[08:08:42][D][binary_sensor:036]: 'Partition 1 Ready': Sending state ON
[08:08:42][W][component:237]: Component dsc_alarm_panel took a long time for an operation (74 ms).
[08:08:42][W][component:238]: Components should block for at most 30 ms.
[08:08:42][I][Paneldata: :1130]: 0F: 0F 00 91 F1 1F C7 FC C7 9F FF [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 7D: 7D 00 97 F9 F0 C7 10 C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 07 90](tel:05 00 91 07 90) C7 10 C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: [05 00 91 07 90](tel:05 00 91 07 90) C7 10 C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 07, last status 01,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'Failed       '
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'to arm          '
[08:08:42][D][text_sensor:064]: 'Message': Sending state '07: Failed to arm'
[08:08:42][D][text_sensor:064]: 'Security Status': Sending state 'not_ready'
[08:08:42][D][binary_sensor:036]: 'Partition 1 Ready': Sending state OFF
[08:08:42][W][component:237]: Component dsc_alarm_panel took a long time for an operation (54 ms).
[08:08:42][W][component:238]: Components should block for at most 30 ms.
[08:08:42][I][Paneldata: :1130]: 05: 05 00 9D 01 F0 DF D0 FF 1F C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: 05 00 9D 01 F0 DF D0 FF 1F C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 01, last status 07,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[08:08:42][I][info:1955]: status DF, last status CF,line2status 00,selection 00,partition=2,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'Message': Sending state '01: Ready'
[08:08:42][D][text_sensor:064]: 'Security Status': Sending state 'disarmed'
[08:08:42][D][binary_sensor:036]: 'Partition 1 Ready': Sending state ON
[08:08:42][W][component:237]: Component dsc_alarm_panel took a long time for an operation (62 ms).
[08:08:42][W][component:238]: Components should block for at most 30 ms.
[08:08:42][I][Paneldata: :1130]: 05: [05 00 97 81](tel:05 00 97 81) 7C DF D0 FF FC FF [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: [05 00 97 81](tel:05 00 97 81) 7C DF D0 FF FC FF [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 81, last status 01,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state '129'
[08:08:42][D][text_sensor:064]: 'Message': Sending state '81: Unknown'
[08:08:42][D][text_sensor:064]: 'Security Status': Sending state 'not_ready'
[08:08:42][D][binary_sensor:036]: 'Partition 1 Ready': Sending state OFF
[08:08:42][I][Paneldata: :1130]: FF: FF 01 91 F9 17 C7 7C C7 F0 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: 05 00 F9 01 10 C7 1F C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][Paneldata: :1130]: 05: 05 00 F9 01 10 C7 1F C7 10 C7 [01 00 00 00 00 00](tel:01 00 00 00 00 00) 
[08:08:42][I][info:1955]: status 01, last status 81,line2status 00,selection 01,partition=1,skip=0,force=0
[08:08:42][D][text_sensor:064]: 'line1 Partition 1 (ln1_1)': Sending state 'System is Ready'
[08:08:42][D][text_sensor:064]: 'line2 Partition 1 (ln2_1)': Sending state 'Ready to Arm <>'
[08:08:42][D][text_sensor:064]: 'Message': Sending state '01: Ready'
[08:08:42][D][text_sensor:064]: 'Security Status': Sending state 'disarmed'
[08:08:42][D][binary_sensor:036]: 'Partition 1 Ready': Sending state ON
[08:08:42][W][component:237]: Component dsc_alarm_panel took a long time for an operation (53 ms).
[08:08:42][W][component:238]: Components should block for at most 30 ms.`

Dilbert66, 2025-04-06:
Your log is bizare.  What is this section:  [05 00 91 03](tel:05 00 91 03)  . My code never sends this type of data.
Also, what panel is this installed on?

lukemfarber, 2025-04-06:
I’m not sure what that code is. It does seem irregular. I have it installed on a DSC PC1616 from ADT. Possibly could be caused by the excessive voltage you mentioned? This is my YAML config, I might have messed something up in there:

`#for documentation see project at https://github.com/Dilbert66/esphome-dsckeybus
substitutions:
  name: "alarmpanel" #unique network name
  friendlyName: "Alarm Panel" #friendly name for this device
  panelId: AlarmPanel  #used as the service variable name.
  
dsc_alarm_panel:
  id: $panelId
  #you can enter a list of user codes mapped to their names for display in armed/disarmed event messages
  #usercodes: "1:Someone" #
  
  #Only comes into effect if a password prompt occurs when arming eg. night mode
  accesscode: !secret access_code 
  
  #used to select the default partition associated with the alarm panel messages
  defaultpartition: "1" 
  
  maxzones: "6" # maximum amount of zones your system supports
  
  # dsc debug level: 0 = off, 1 = minimal, 2 = all packets shown on console  3 =  + serial port debugging
  dscdebuglevel: "2"  
   
  #zone expander addresses:
  # 9  - zones 9-16
  # 10 - zones 17-24
  # 11 - zones 25-32
  # 12 - zones 33-40 (for systems with 64 zone support)
  # 13 - zones 41-48 (for systems with 64 zone support)
  # 14 - zones 49-56 (for systems with 64 zone support)
  # 16 - zones 57-64 (for systems with 64 zone support)  
  expanderaddr1: "0" # 1st zone expander emulator address to use . Set to 0 to disable. 
  expanderaddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 

  ##ESP32 Pins
  #dscclockpin: "22"
  #dscreadpin: "21"
  #dscwritepin: "18"
  
  ##ESP8266 Pins
  dscclockpin: "5"
  dscreadpin: "4"
  dscwritepin: "15" 
  

  clean_build: "false" #default is false.  set to true if getting duplication errors in linking step.  

#esp32: 
  #board: nodemcu-32s
  #framework: 
    #type: arduino
    #version: recommended

esp8266:    
  board: nodemcuv2
  framework:
    version: recommended 

#location of alarm panel code.  You can use the github release version  or 
#copy the code to directory "my_components" in your main esphome directory   
# see here for more info: https://esphome.io/components/external_components       
external_components:
  - source: github://Dilbert66/esphome-components@main #uncomment to use github repository
  #- source:  my_components #uncomment to use local directory 
    components: [dsc_alarm_panel,binary_sensor,text_sensor] 
    refresh: 10min
  
esphome:
  name: $name
[... 310 lines cut ...]
 #     - lambda: |-
 #         dsc.setZoneFault(17,0);  #set zone 17 as closed
 

 # end of panel sensor setup - no need to edit anything below.
##########################################################################################
switch:
    #shows status of connection status to panel.  You can disconnect before upload using the switch.
  - platform: template
    name: "Connection"
    id: connection_status_switch
    lambda: |-
      return dsc.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectKeybus();

  - platform: restart
    id: restart_switch
    
  - platform: safe_mode
    name: "Safe Mode" `

Dilbert66, 2025-04-06:
I suspect you have a wiring issue as your log is just garbage data. Your config looks fine. Make sure you have the right wire going to the correct pin.  I don't think it's related to your resistor choices.

lukemfarber, 2025-04-19:
Hi, turns out it was a wiring issue, looked like I had some lines shorting causing random data on the logs. Thanks!

closed 2025-04-19 by lukemfarber