Dilbert66/esphome-dsckeybus #173: [ESPHome 2025.7.x] binary_sensor and text_sensor: ImportError: cannot import name 'setup_entity' + compatibility issues with recent ESPHome changes

blitzu, 2025-07-16:
Hello,

After upgrading to ESPHome 2025.7.0, I am unable to use the custom binary_sensor and text_sensor components from this repository. I receive the following error for both:

ImportError: cannot import name 'setup_entity' from 'esphome.cpp_helpers'

This happens for both /components/binary_sensor/__init__.py and /components/text_sensor/__init__.py. Removing these from my external_components list solves the build issue, but then I lose the custom features from this repo.

Additionally, ESPHome now gives me this warning:
WARNING The 'publish_initial_state' option has been replaced by 'trigger_on_initial_state' and will be removed in a future release

So it seems that all template sensors using publish_initial_state: will need to use trigger_on_initial_state: instead.

Also, the option id_code is no longer valid in template sensors. Only id is accepted for both binary_sensor and text_sensor using platform: template. The build fails with this message:
[id_code] is an invalid option for [binary_sensor.template]. Please check the indentation.

To be transparent:
I am not a programmer and all these checks and diagnostics were done with the help of ChatGPT. If you need my YAML file for testing or more info, I can provide it.

Thank you for your work!

Dilbert66, 2025-07-17:
I've pushed new updates to work with the updated version of Esphome and Arduino. Unfortunately you will need to update your yaml config to the new format. Please see the example yaml in the project page. Also note the addition of the new config parameter "custom_services: true" under the api section.

afarago, 2025-07-19:
Thank you, works nicely!

At first I missed changing the sensor platform from "template" to "dsc_alarm_panel". I followed up, not it works - thank you.
Might want to consider a validation for the proper ids in the .py files.