Dilbert66/esphome-dsckeybus #175: esphome-dsckeybus keeps system busy while "toruble light" flashes makes complete alarm system unavailable

bladika, 2025-08-10:
Hi,
first of all thanks for the grate work, it is awesome.
One issue I recognized that when there is a FLASHING! trouble light in the system (e.h.: both alarm and trouble are present, and alarm and trouble light flashes alternately) then esphome-dsckeybus keeps the system busy therefore not available.

Practical example:
- Alarm triggered by tamper
- System goes into alarm mode (siren is active)
- Then both "Alarm" and "Trouble" start flashing alternately.
- esphome-dsckeybus keeps system busy therefore user not even able to disable alarm from keypad (gets system busy message always)
- only solution is to disconnect esphome-dsckeybus and then system goes back to normal

My suspicion is that this function causes the issue:
`
#flag to have the esp check for trouble statuses for devices/zones when the trouble light goes on or off
#Note: It will prevent the physical panel from beeping to let you know as the check resets the beep.
#Sends *21 and *27 to the panel. Default: true

trouble_fetch: true
`
I believe the root cause is that due to both "alarm" and "trouble" present in the system, the "trouble" light not just goes ON, but flashing alternately with "alarm" with ~1sec period. Each and every time when "trouble" light goes on (w/ ~1sec period) esphome-dsckeybus reads troubles but because of the frequent period, keeps the bus busy therefore prevents everyone else on the bus to communicate (including keypads).

If the problem is really the one above, then one possible solution could be that it is limited that how often troubles are being read if trouble light is flashing. e.g.: if troubles has been read due to trouble light went ON then for a certain period of time (10sec? or parameter?) troubles will be not read even though trouble light went ON again.

As an intermittent solution I set "trouble_fetch: false" but would be good if I can keep this function running (e.g.: trigger automation as soon as any trouble appears not just reading with a certain period with (trouble_fetch_update_time")

My system:
- DSC PC1616
- 2x PK5507
- Signal PS128A-1 Siren
-  esphome-dsckeybus

Dilbert66, 2025-08-12:
It should be fixed already with the last commit. You will need to recompile

bladika, 2025-08-17:
Sorry for the amateur question, but "You will need to recompile" means if I recompiled in Home Assistant - ESPHome Device Builder, then it is enough or I need to do something more?

Anyhow. I did recompile in ESPHome device builder and it looks to be the issue ha been fixed.

Thanks for the fix!

closed 2025-08-20 by Dilbert66